srsran/srsRAN_Project #1: example config file doesn't exists

iman432, 2023-02-22:
Hi srsRAN,

Thanks a lot for this project, 
I cannot find configs/ folder in codes, can you please share or update source codes?

andrepuschmann, 2023-02-22:
Hey, we've just pushed an update that contains two example configs. From your build folder, you should be able to run with:
`./apps/gnb/gnb -c ../configs/gnb_rf_b200_tdd_n78_10mhz.yml`.
Make sure to adopt to your AMF params.

alexjiao2021, 2023-02-22:
Does srsgNB support LimeSDR? If yes, could you provide example files? Thanks!

pricalinat, 2023-02-23:
> 

I also want to ask, will it support bladerf? Thanks.

iman432, 2023-02-23:
> Hey, we've just pushed an update that contains two example configs. From your build folder, you should be able to run with: `./apps/gnb/gnb -c ../configs/gnb_rf_b200_tdd_n78_10mhz.yml`. Make sure to adopt to your AMF params.

Thank you, but seems there is an issue with "cell_cfg", got below error, i think should be "common_cell" :

by cell_cfg:

[image]

by common_cell is normal:

[image]

config:

[image]

andrepuschmann, 2023-02-23:
@iman432 Please do a git pull and update to `77be7d3391d45fb606aec4f0c16907390b49b0b0`. The name of that struct has been changed in one of the last commits.

> Does srsgNB support LimeSDR? If yes, could you provide example files? Thanks!
@alexjiao2021 We don't support it currently. Only UHD-based devices. But we might add support to it later on.

> I also want to ask, will it support bladerf? Thanks.
@pricalinat Same as above. We'll add support to it later.

closed 2023-02-23 by iman432

jiban-ray, 2023-02-25:
[image]

Getting this error.|
**INI was not able to parse common_cell.++**
Can anyone suggest solutions, please?