srsran/srsRAN_Project #2: zmq tutorial and examples

zyddnys, 2023-02-23:
Hi srsran,
It seems the entire project just underwent major changes and all old resources are now 404. I wonder what happened to srsRAN? And is there a renewed version on connecting srsUE with gNB via zmq?

andrepuschmann, 2023-02-23:
Hey @zyddnys 
> It seems the entire project just underwent major changes

That's correct. srsRAN Project is an entirely new codebase based on a new software architecture. Check the mailing list archive, there you'll find posts explaining the project and naming change in detail.

> all old resources are now 404

That shouldn't be the case. In fact, even if you checked out the repo like 5 years ago under srsLTE it should still work, as the repo links are aliases to the old names. The 4G code base is now called "srsRAN 4G"

> And is there a renewed version on connecting srsUE with gNB via zmq?

We might add an appnote explaining that at a later stage - for now we've focused on third-party UEs including COTS handsets.

Hope that helps.

suttonpd, 2023-04-14:
To be clear, the srsRAN_4G codebase is located at [github.com/srsran/srsRAN_4G](https://github.com/srsran/srsRAN_4G) and previous clone paths for srsLTE and srsRAN still direct to that repository.
The documentation is at [docs.srsran.com/projects/4g](https://docs.srsran.com/projects/4g) and is clearly linked from the top level [docs.srsran.com](https://docs.srsran.com) page.

closed 2023-04-14 by suttonpd