srsran/srsRAN_Project #3: Is there any specific gcc, g++ version requirement?

whitecloudy, 2023-02-23:
## Issue Description ##
Unable to build the project

## Setup Details ##
Ubuntu 18.04.6 LTS
cmake 3.25.2(I manually installed)
gcc 7.5.0
g++ 7.5.0

## Expected Behavior ##
Build Success

## Actual Behaviour ##
Build Fail
[image]

## Steps to reproduce the problem ##
Just tried build as you told me.

## Additional Information ##
Although I successfully build with manullay installed g++-11 and  gcc-11, it would be nice to have minimum version requirement.

andrepuschmann, 2023-02-23:
Yes, Ubuntu 18.04 is not supported anymore as the default compile is too old. You could upgrade them of course. Also see [here](https://github.com/srsran/srsRAN_Project/actions/runs/4246746296) for the pipeline with Ubuntu 20.04.

Closing this issue.

closed 2023-02-23 by andrepuschmann

whitecloudy, 2023-02-23:
@andrepuschmann Thanks. At least it would be nice to mention this problem in README or somewhere.