srsran/srsRAN_Project #4: About new config file

orion-belt, 2023-02-23:
Hi
* Can we have `Gnb ID` as config parameter since it should be unique per gnb instance ?
* Also is it possible to have two separate interface binding for `NGAP LOCAL ADDRESS` &  `GTPU LOCAL ADDRESS`  ?

andrepuschmann, 2023-02-24:
Hi,

> Is it possible to have two separate interface binding for NGAP LOCAL ADDRESS & GTPU LOCAL ADDRESS ?

The currently provided gNB config is a simplified version and doesn't contain all params, yet. But it's a good point to add.

Meanwhile, have a look at https://github.com/srsran/srsRAN_Project/blob/77be7d3391d45fb606aec4f0c16907390b49b0b0/apps/gnb/gnb.cpp#L495
and hard-code your IP there.

> Is there plan to network slicing support ?

Short answer: Yes! Long answer: What aspect of slicing are you interested in?

orion-belt, 2023-02-24:
> Hi,
>
>
> The currently provided gNB config is a simplified version and doesn't contain all params, yet. But it's a good point to add.
>
> Meanwhile, have a look at [...]

Okay thanks 

> > Is there plan to network slicing support ?
> 
> Short answer: Yes! Long answer: What aspect of slicing are you interested in?

* To remove hardcoded value (1, 0xFFFFFF), i.e. to be able to configure different slices (with sd value). It could be again hardcoded in nas (zmq UE) or ngap if complete PRB/DRB level sclicing not implemented/planned.

* It will be interesting to see multiple pdu sessions based on slices/dnn/qfi etc. 

Do you have plan to release ZMQ UE again in near future ?

andrepuschmann, 2023-02-24:
> To remove hardcoded value (1, 0xFFFFFF), i.e. to be able to configure different slices (with sd value). It could be again hardcoded in nas (zmq UE) or ngap if complete PRB/DRB level sclicing not implemented/planned.
It will be interesting to see multiple pdu sessions based on slices/dnn/qfi etc.

Yes, those values will be exposed in the upcoming code-drops for sure.

> Do you have plan to release ZMQ UE again in near future?

We will update the 5G SA UE code in "srsRAN 4G" to work with srsRAN Project. The main purpose of this is that people can continue using an srsRAN UE with our gNB. The UE will have the same feature limitations as before. As a result you need to use a gNB configuration that is  supported by srsUE, e.g. FDD, bandwidth of 10/20MHz and 15kHZ SCS.

orion-belt, 2023-02-24:
Noted. Thanks for the clarification.

closed 2023-02-24 by orion-belt