srsran/srsRAN_Project #5: stdout and log file empty until exit

desimetallica, 2023-02-24:
## Issue Description ##
Hello thanks for the work here! 

While running command:
`sudo gnb -c config.yml`

the problem is that there is no stdout and log files are empty until i quit, ctrl+c command.
Its look like also that usrp doesn't work properly (tx/rx leds are off).

Looks like also no connection is made with AMF (looking at the amf logs) but on in the log it says [GNB     ] [I] AMF connection established thats sounds strange. 

I'm little bit lost.

## Setup Details ##
make test runs ok

Os: Ubuntu 20.04.5 LTS
USRP: x310 (also tested with b210)
gnb: 0.1
UHD: 3.15.0.0-2build5
gnuradio: 3.8.1.0
something more needed?

## Expected Behavior ##
Log file and stdout output while running some feedback for errors and 

## Actual Behaviour ##
no output on the stdout or logfile while running

## Steps to reproduce the problem ##
config file:
[gnbconf.log](https://github.com/srsran/srsRAN_Project/files/10823074/gnbconf.log)

gnb.log:
[gnb.log](https://github.com/srsran/srsRAN_Project/files/10823067/gnb.log)

## Additional Information ##
The usrp and gnuradio stack seems to be functional cause I've tested it with the command: 
`./rx_ascii_art_dft --freq 98e6 --rate 5e6 --gain 20 --bw 5e6 --ref-lvl -30`

Thanks for any help

andrepuschmann, 2023-02-24:
Hey,

> Looks like also no connection is made with AMF (looking at the amf logs) but on in the log it says [GNB ] [I] AMF connection established thats sounds strange.

Do you a PCAP of the link?

> no output on the stdout or logfile while running

The log file output is cached. It will only flush at exit or occasionally during run. That is expected.

desimetallica, 2023-02-24:
Hello Andre I've just this pcap with one packet inside. The other one on gnb_mac.pcap is empty.

[gnb_ngap.zip](https://github.com/srsran/srsRAN_Project/files/10824866/gnb_ngap.zip)

andrepuschmann, 2023-03-03:
> But I have to investigate deeper. At the moment after the disconnection the gnb application just crash with:
srsGNB ERROR: Unable to create radio session.

But that's a different issue then that causes the AMF link to go down as well. So fix the radio issue first.

desimetallica, 2023-03-08:
Hi,many thanks for your suggestions! I will try also gr-fosphor.
Another thought, I'm on 1GE, is 10GE a strict requirement to connect USRP and PC running srsRAN?

mubk00, 2023-03-09:
> Hi,many thanks for your suggestions! I will try also gr-fosphor. Another thought, I'm on 1GE, is 10GE a strict requirement to connect USRP and PC running srsRAN?

I don't think there is a direct requirement using 10GE, but it can be recommended since you are using X310. With that said, you will get better results in terms of performance.

Since one of the main problems that you are facing is with the radio - I'd suggest you to test using a benchmark and see if the tests passes. Use another SDR and just try to see if you are getting any signal, which was mentioned. Gr-forsphor should work perfectly, or you can just simply setup up a simple flowgraph on GRC. If you need help with that, let me know. I'll be more than welcome to help you out.

Also, just curious you mentioned that you tested with X310 and B210. Can you please confirm if you are you getting the same exact problem that you mentioned?

desimetallica, 2023-03-09:
Hi, many thanks for your feedback, I've tested it with x310 and b210, I will try it again with the suggestion collected here and let you know the updates. I will change my setup in order to use gr-forsphor as a specturm analyzer on b210 and x310 as a gnb. 
Let me work on it I will update on this.

desimetallica, 2023-03-16:
Hi again some update from my side: 

- performed test with ./benchmark_rate --rx_rate 10e6 --tx_rate 10e6 without issues  

- try on osmocom_fft -F -W (looking good direct attaching with cables + attenuator) 

[image]

- still issues with AMF connection ( I confirm issue #21 ) basically the gnb register itself successfully with amf (open5gs) but after a while get disconnected.

`03/16 11:20:16.389: [amf] INFO: [Added] Number of gNBs is now 1 (../src/amf/context.c:973)
03/16 11:20:16.389: [amf] INFO: gNB-N2[10.244.0.0] max_num_of_ostreams : 30 (../src/amf/amf-sm.c:698)
03/16 11:20:20.174: [amf] INFO: gNB-N2[10.244.0.0] connection refused!!! (../src/amf/amf-sm.c:711)
03/16 11:20:20.174: [amf] INFO: [Removed] Number of gNBs is now 0 (../src/amf/context.c:1000)`

I've captured sctp traffic in the corresponding AMF service node with tcpdump.
I've changed the log to debug to get more information also. 

[gnbNGAP.log](https://github.com/srsran/srsRAN_Project/files/10990108/gnbNGAP.log)

[gnb.log](https://github.com/srsran/srsRAN_Project/files/10990127/gnb.log)

andrepuschmann, 2023-03-20:
Since issue #21 has been fixed and is now included in `main` I would think the issue is solved. Can you confirm @desimetallica ?

desimetallica, 2023-03-21:
@andrepuschmann I can confirm 21 is resolved.

andrepuschmann, 2023-03-21:
Excellent. thanks. closing ticket