srsran/srsRAN_Project #6: BladeRF Support

JoseSecmotic, 2023-02-24:
## Issue Description ##
After start researching on prototyping with bladeRF and srsRAN I came across to the new srsRAN_Project.
I decided to try it but after building it, it does not recognize my bladeRF (I have xA9 and xA5).

Is there support for bladeRF 2.0?

When I use the gnb, the following output is shown:

```
sudo ./gnb -c ../../../configs/gnb_rf_b200_tdd_n78_10mhz.yml 
Available radio types: .

--== srsRAN gNB (commit 77be7d339) ==--

Factory for radio type soapy not found. Make sure to select a valid type.
srsGNB ERROR: Unable to create radio session.
```

jao-tm, 2023-02-25:
Hi,
It seems it's not supported yet, only UHD:
https://github.com/srsran/srsRAN_Project/issues/1#issuecomment-1441343604

andrepuschmann, 2023-03-03:
Correct, it's not supported currently.

closed 2023-03-03 by andrepuschmann

JoseSecmotic, 2023-03-03:
Thanks @andrepuschmann 
Do you have any estimation on when will it be available? Or if it will be available at all?
Thanks in advance

ohault, 2024-11-28:
Same question about Soapy support for srsGNB