srsran/srsRAN_Project #9: srsue and gnb compatibility with ZMQ

iikercosta, 2023-02-27:
## Issue Description ##

UE does not attach with gNodeB using srsue. 

## Setup Details ##

srsue: 22.10
gnb: 0.1
RF front-end: libzmq.so.5 (ZMQ)
5G core: open5gs

## Expected Behavior ##

gNodeB attaches with open5gs core and the srsue UE too.

## Actual Behaviour ##

gNodeB attaches with open5gs but srsue UE doesn`t. 

[image]

## Steps to reproduce the problem ##

`$ srsue ue.conf`

[ue.conf.zip](https://github.com/srsran/srsRAN_Project/files/10836925/ue.conf.zip)

`$ gnb -c gnb_rf_zmq_tdd_n78_10mhz.yml`

[gnb_rf_zmq_tdd_n78_10mhz.yml.zip](https://github.com/srsran/srsRAN_Project/files/10836926/gnb_rf_zmq_tdd_n78_10mhz.yml.zip)

## Additional Information ##

Test pcap in the gNodeB host: 

[test-srs.pcapng.zip](https://github.com/srsran/srsRAN_Project/files/10836985/test-srs.pcapng.zip)

pricalinat, 2023-02-28:
Have you solved it? I also encountered the same problem.

iikercosta, 2023-02-28:
> Have you solved it? I also encountered the same problem.

Not for the moment

miniskar, 2023-03-02:
I am also facing same problem. Please let me know if anyone has solution.

asheeshtripathi, 2023-03-02:
I am trying with gNB on x310(band 78 TDD) and srsUE on N310. gNB is able to connect to open5gs core but UE is not able to detect the network. Please let me know if anyone is able to resolve it.

closed 2023-03-03 by iikercosta