srsran/srsRAN_Project #10: srsue with b210

eulenleber, 2023-02-28:
I'm trying to connect srsue with the gnB (each running a B210 sdr) but can not achieve that.
(for the core I basically followed https://open5gs.org/open5gs/docs/guide/01-quickstart/, but I guess that is irrelevant at this point since the connection between gnB and UE is not even built)

## gNB
```yaml
amf:
  addr: 10.10.0.5                 # The address or hostname of the AMF.
  bind_addr: 127.0.0.1              # A local IP that the gNB binds to for traffic from the AMF.
rf_driver:
  device_driver: uhd                # The RF driver name.
  device_args: type=b200            # Optionally pass arguments to the selected RF driver.
  clock:                            # Specify the clock source used by the RF.
  sync:                             # Specify the sync source used by the RF.
  srate: 11.52                      # RF sample rate might need to be adjusted according to selected bandwidth.
  tx_gain: 50                       # Transmit gain of the RF might need to adjusted to the given situation.
  rx_gain: 60                       # Receive gain of the RF might need to adjusted to the given situation.
cell_cfg:
  dl_arfcn: 632628                  # ARFCN of the downlink carrier (center frequency).
  band: 78                          # The NR band.
  channel_bandwidth_MHz: 10         # Bandwith in MHz. Number of PRBs will be automatically derived.
  common_scs: 30                    # Subcarrier spacing in kHz used for data.
  plmn: "00101"                     # PLMN broadcasted by the gNB.
  tac: 7                            # Tracking area code (needs to match the core configuration).
  pci: 1                            # Physical cell ID.
log:
  filename: /tmp/gnb.log            # Path of the log file.
  all_level: debug                # Logging level applied to all layers.
pcap:
  mac_enable: true                 # Set to true to enable MAC-layer PCAPs.
  mac_filename: /tmp/gnb_mac.pcap   # Path where the MAC PCAP is stored.
  ngap_enable: true                # Set to true to enable NGAP PCAPs.
  ngap_filename: /tmp/gnb_ngap.pcap # Path where the NGAP PCAP is stored.
```

```
 sudo build/apps/gnb/gnb -c gnb.conf
Available radio types: uhd.

--== srsRAN gNB (commit 77be7d339) ==--

[INFO] [UHD] linux; Clang version 14.0.0 ; Boost_107400; UHD_4.4.0.0-0-g5fac246b
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Making USRP object with args 'type=b200'
[DEBUG] [B200] the firmware image: /usr/share/uhd/images/usrp_b200_fw.hex
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[DEBUG] [AD936X] baseband bandwidth too large for current sample rate. Setting bandwidth to: 5e+07
[DEBUG] [AD936X] baseband bandwidth too large for current sample rate. Setting bandwidth to: 5e+07
[DEBUG] [AD936X] baseband bandwidth too large for current sample rate. Setting bandwidth to: 5e+07
[DEBUG] [AD936X] baseband bandwidth too large for current sample rate. Setting bandwidth to: 5e+07
[DEBUG] [AD936X] baseband bandwidth too large for current sample rate. Setting bandwidth to: 5e+07
[DEBUG] [AD936X] baseband bandwidth too large for current sample rate. Setting bandwidth to: 5e+07
[DEBUG] [AD936X] baseband bandwidth too large for current sample rate. Setting bandwidth to: 5e+07
[DEBUG] [AD936X] baseband bandwidth too large for current sample rate. Setting bandwidth to: 5e+07
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[DEBUG] [AD936X] Performing CODEC loopback test... 
[DEBUG] [AD936X] CODEC loopback test passed.
[DEBUG] [AD936X] Performing CODEC loopback test... 
[DEBUG] [AD936X] CODEC loopback test passed.
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz... 
[INFO] [B200] Actually got clock rate 16.000000 MHz.
[DEBUG] [CORES] Performing timer loopback test... 
[DEBUG] [CORES] Timer loopback test passed.
[DEBUG] [CORES] Performing timer loopback test... 
[DEBUG] [CORES] Timer loopback test passed.
[INFO] [MULTI_USRP] Setting master clock rate selection to 'manual'.
[INFO] [B200] Asking for clock rate 11.520000 MHz... 
[INFO] [B200] Actually got clock rate 11.520000 MHz.
[DEBUG] [CORES] Performing timer loopback test... 
[DEBUG] [CORES] Timer loopback test passed.
[DEBUG] [CORES] Performing timer loopback test... 
[DEBUG] [CORES] Timer loopback test passed.
[DEBUG] [CONVERT] get_converter: For converter ID: conversion ID
  Input format:  fc32
  Num inputs:    1
  Output format: sc16_item32_le
  Num outputs:   1
 Using best available prio: 3
[DEBUG] [CONVERT] get_converter: For converter ID: conversion ID
  Input format:  sc16_item32_le
  Num inputs:    1
  Output format: fc32
  Num outputs:   1
 Using best available prio: 3
Cell pci=1, bw=10 MHz, dl_arfcn=632628 (n78), dl_freq=3489.42 MHz, dl_ssb_arfcn=632640, ul_freq=3489.42 MHz

==== gNodeB started ===
Type <t> to view trace
Late: 79; Underflow: 211; Overflow: 2;
Late: 77; Underflow: 227; Overflow: 0;
...
```
## UE

```yaml
[rf]
freq_offset = 0
tx_gain = 50
rx_gain = 60
[rat.eutra]
dl_earfcn = 3350
nof_carriers = 1
[rat.nr]
bands = 78
dl_nr_arfcn = 632628
ssb_nr_arfcn = 632640
[pcap]
enable = none
mac_filename = /tmp/ue_mac.pcap
mac_nr_filename = /tmp/ue_mac_nr.pcap
nas_filename = /tmp/ue_nas.pcap
[log]
all_level = warning
phy_lib_level = none
all_hex_limit = 32
filename = /tmp/ue.log
file_max_size = -1
[usim]
mode = soft
algo = milenage
opc  = 63BFA50EE6523365FF14C1F45F88737D
k    = 00112233445566778899aabbccddeeff
imsi = 001010123456780
imei = 353490069873319
[rrc]
[nas]
[gw]
[gui]
enable = false
[channel.dl]
[channel.dl.awgn]
[channel.dl.fading]
[channel.dl.delay]
[channel.dl.rlf]
[channel.dl.hst]
[channel.ul]
[channel.ul.awgn]
[channel.ul.fading]
[channel.ul.delay]
[channel.ul.rlf]
[channel.ul.hst]
[phy]
[phy.nr]
[cfr]
[sim]
[general]
```

```
sudo srsue ue.conf
Active RF plugins: libsrsran_rf_uhd.so libsrsran_rf_zmq.so
Inactive RF plugins: 
Reading configuration file ue.conf...
WARNING: cpu0 scaling governor is not set to performance mode. Realtime processing could be compromised. Consider setting it to performance mode before running the application.

Built in Debug mode using commit 80cc511 on branch main.

Opening 1 channels in RF device=default with args=default
Supported RF device list: UHD zmq file
Trying to open RF device 'UHD'
[INFO] [UHD] linux; Clang version 14.0.0 ; Boost_107400; UHD_4.4.0.0-0-g5fac246b
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=1, args: type=b200,master_clock_rate=23.04e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] Actually got clock rate 23.040000 MHz.
RF device 'UHD' successfully opened
Waiting PHY to initialize ... done!
Attaching UE...
.
.
Found Cell:  Mode=FDD, PCI=445, PRB=100, Ports=2, CP=Normal, CFO=-7.3 KHz
RF status: O=6, U=0, L=0
.
Found Cell:  Mode=FDD, PCI=445, PRB=100, Ports=2, CP=Normal, CFO=-5.6 KHz
.
.
RF status: O=15, U=0, L=0
..
...
.
Found Cell:  Mode=FDD, PCI=445, PRB=100, Ports=2, CP=Normal, CFO=-7.7 KHz
.
.
```

## spectrum analyzer
I would further expect a peak around 3489.42 MHz; but I only see noise
```
 /usr/lib/uhd/examples/rx_ascii_art_dft --freq 3489420000 --rate 5e6 --gain 60 --bw 5e6 --ref-lvl -30
```

But as the cell is found I guess that is a wrong assumption

brendan-mcauliffe, 2023-03-01:
Hi, srsUE is not yet compatible with the new gNB, we are working on a fix for this and should have something soon. We will also update our docs with a guide on using the srsRAN 4G UE and srsRAN Project gNB when the changes have been made.

eulenleber, 2023-03-02:
thanks for the feedback
shouldnt at least the rx_ascii_art_dft show something?

eulenleber, 2023-03-02:
after checking again with a COTS UE it does not find any net.

eulenleber, 2023-03-03:
Its not clear from my initial post but the gNB console output shows continuously `lates`
like
```
Late: 95; Underflow: 268; Overflow: 0;
Late: 64; Underflow: 243; Overflow: 0;
Late: 59; Underflow: 230; Overflow: 0;
Late: 102; Underflow: 278; Overflow: 0;
Late: 61; Underflow: 236; Overflow: 0;
Late: 84; Underflow: 256; Overflow: 0;
Late: 96; Underflow: 262; Overflow: 0;
Late: 57; Underflow: 236; Overflow: 0;
Late: 83; Underflow: 243; Overflow: 0;
Late: 96; Underflow: 257; Overflow: 0;
Late: 74; Underflow: 261; Overflow: 0;
Late: 99; Underflow: 271; Overflow: 0;
Late: 67; Underflow: 254; Overflow: 0;
Late: 100; Underflow: 277; Overflow: 0;
Late: 61; Underflow: 229; Overflow: 0;
Late: 78; Underflow: 245; Overflow: 0;
Late: 82; Underflow: 251; Overflow: 0;
Late: 124; Underflow: 296; Overflow: 0;
Late: 81; Underflow: 258; Overflow: 0;
Late: 87; Underflow: 261; Overflow: 0;
Late: 68; Underflow: 253; Overflow: 0;
Late: 95; Underflow: 265; Overflow: 0;
Late: 78; Underflow: 255; Overflow: 0;
Late: 84; Underflow: 251; Overflow: 0;
Late: 102; Underflow: 275; Overflow: 0;
Late: 73; Underflow: 247; Overflow: 0;
Late: 86; Underflow: 251; Overflow: 0;
Late: 73; Underflow: 248; Overflow: 0;
Late: 81; Underflow: 257; Overflow: 0;
Late: 91; Underflow: 263; Overflow: 0;
Late: 102; Underflow: 283; Overflow: 0;
Late: 93; Underflow: 274; Overflow: 0;
Late: 64; Underflow: 237; Overflow: 0;
Late: 82; Underflow: 253; Overflow: 0;
Late: 116; Underflow: 281; Overflow: 0;
Late: 85; Underflow: 256; Overflow: 0;
Late: 66; Underflow: 236; Overflow: 0;
Late: 84; Underflow: 255; Overflow: 0;
Late: 89; Underflow: 260; Overflow: 0;
Late: 91; Underflow: 253; Overflow: 0;
Late: 75; Underflow: 245; Overflow: 0;
Late: 112; Underflow: 285; Overflow: 0;
Late: 88; Underflow: 273; Overflow: 0;
Late: 107; Underflow: 283; Overflow: 0;
Late: 117; Underflow: 290; Overflow: 0;
Late: 97; Underflow: 264; Overflow: 0;
Late: 90; Underflow: 258; Overflow: 0;
Late: 103; Underflow: 287; Overflow: 0;
Late: 88; Underflow: 259; Overflow: 0;
Late: 87; Underflow: 253; Overflow: 0;
Late: 77; Underflow: 261; Overflow: 0;
Late: 84; Underflow: 256; Overflow: 0;
Late: 74; Underflow: 257; Overflow: 0;
Late: 89; Underflow: 271; Overflow: 0;
Late: 69; Underflow: 245; Overflow: 0;
Late: 79; Underflow: 258; Overflow: 0;
Late: 81; Underflow: 257; Overflow: 0;
Late: 82; Underflow: 321; Overflow: 0;
Late: 101; Underflow: 267; Overflow: 0;
Late: 122; Underflow: 296; Overflow: 0;
Late: 90; Underflow: 258; Overflow: 0;
Late: 84; Underflow: 255; Overflow: 0;
Late: 85; Underflow: 256; Overflow: 0;
Late: 97; Underflow: 258; Overflow: 0;
Late: 106; Underflow: 279; Overflow: 0;
Late: 93; Underflow: 269; Overflow: 0;
Late: 101; Underflow: 280; Overflow: 0;
Late: 79; Underflow: 250; Overflow: 0;
Late: 77; Underflow: 253; Overflow: 0;
Late: 62; Underflow: 239; Overflow: 0;
Late: 87; Underflow: 258; Overflow: 0;
Late: 104; Underflow: 276; Overflow: 0;
Late: 77; Underflow: 245; Overflow: 0;
Late: 84; Underflow: 255; Overflow: 0;
Late: 92; Underflow: 272; Overflow: 0;
Late: 90; Underflow: 259; Overflow: 0;
Late: 72; Underflow: 246; Overflow: 0;
Late: 103; Underflow: 278; Overflow: 0;
Late: 64; Underflow: 232; Overflow: 0;
Late: 88; Underflow: 265; Overflow: 0;
Late: 85; Underflow: 249; Overflow: 0;
Late: 95; Underflow: 291; Overflow: 0;
Late: 99; Underflow: 269; Overflow: 0;
Late: 68; Underflow: 245; Overflow: 0;
Late: 83; Underflow: 259; Overflow: 0;
Late: 83; Underflow: 257; Overflow: 0;
Late: 73; Underflow: 258; Overflow: 0;
Late: 61; Underflow: 231; Overflow: 0;
Late: 96; Underflow: 263; Overflow: 0;
Late: 80; Underflow: 244; Overflow: 0;
Late: 79; Underflow: 316; Overflow: 0;
Late: 117; Underflow: 287; Overflow: 0;
Late: 99; Underflow: 274; Overflow: 0;
Late: 82; Underflow: 271; Overflow: 0;
Late: 78; Underflow: 248; Overflow: 0;
Late: 66; Underflow: 240; Overflow: 0;
Late: 80; Underflow: 264; Overflow: 0;
Late: 78; Underflow: 251; Overflow: 0;
Late: 70; Underflow: 248; Overflow: 0;
Late: 74; Underflow: 262; Overflow: 0;
Late: 97; Underflow: 274; Overflow: 0;
Late: 131; Underflow: 300; Overflow: 0;
Late: 118; Underflow: 293; Overflow: 0;
Late: 103; Underflow: 281; Overflow: 0;
Late: 80; Underflow: 253; Overflow: 0;
Late: 67; Underflow: 263; Overflow: 0;
Late: 78; Underflow: 255; Overflow: 0;
Late: 130; Underflow: 291; Overflow: 0;
Late: 92; Underflow: 282; Overflow: 0;
Late: 86; Underflow: 258; Overflow: 0;
Late: 93; Underflow: 258; Overflow: 0;
Late: 78; Underflow: 260; Overflow: 0;
Late: 75; Underflow: 249; Overflow: 0;
```

Is that related to the issue?

gustavobsch, 2023-03-04:
I'm having the same issue here, band 41, I can't pickup any signal when I run gnb.

brendan-mcauliffe, 2023-03-06:
@eulenleber, you should first focus on getting the gNB running without lates and underflows. Take a look [here](https://docs.srsran.com/projects/4g/en/latest/app_notes/source/5g_sa_amari/source/index.html) in the documentation for tips on fixing this. We will have an app note soon on connecting a COTS UE to the gNB. 

@gustavobsch, we recommend that you run the gNB with similar configurations to those given in the example configuration files. You can find them [here](https://github.com/srsran/srsRAN_Project/tree/main/configs). You should try to get a stable connection using either of these, then feel free to try customizing and changing the config.

brendan-mcauliffe, 2023-03-06:
@eulenleber lates, underflows, and overflows occur due to data buffer and processing issues, most likely due to a lack of processing power or misconfiguration. 

By setting the CPU to performance mode, reducing the level of logging, and adjusting the level of processing that has to be done to run the gNB you can reduce lates and underflows. By adjusting the level of processing I mean reducing the overall bandwidth and throughput of the gNB. A set-up that uses 50 PRBs will be more computationally heavy than a set-up that uses 25 PRBs. Lowering the log levels means your PC is not "wasting" resources writing out to logs when they're not needed and can give more resources to actually running the gNB. 

Excessive Lates and Underfows mean that a UE will not receive messages from the gNB on time, and as a result will not connect.

eulenleber, 2023-03-06:
Ah, I see.
We have an Intel nuc with an i5-6260U. Am I correct to assume that this way to weak to provide gNB functionalities?

brendan-mcauliffe, 2023-03-06:
We expect to have some documentation soon about running the gNB on less powerful machines. For now, I suggest you try some of the above fixes.

mubk00, 2023-03-06:
@eulenleber I think the problem here is your host PC. As Brendan mentioned, underflows could be occurring because the host computer is too slow to process the information or because the CPU governor or other forms of power control are incorrectly set up. For further debugging, can you please send the output of _lscpu_ 

Although I highly doubt it will fix the overflow problem, have you attempted connecting the B210 to a USB 3.0 port to see if it fixes the late issue?

eulenleber, 2023-03-06:
yes, as i said the problem may in deed be the cpu (i5-6260U):
```
Architecture:            x86_64
  CPU op-mode(s):        32-bit, 64-bit
  Address sizes:         39 bits physical, 48 bits virtual
  Byte Order:            Little Endian
CPU(s):                  4
  On-line CPU(s) list:   0-3
Vendor ID:               GenuineIntel
  Model name:            Intel(R) Core(TM) i5-6260U CPU @ 1.80GHz
    CPU family:          6
    Model:               78
    Thread(s) per core:  2
    Core(s) per socket:  2
    Socket(s):           1
    Stepping:            3
    CPU max MHz:         2900.0000
    CPU min MHz:         400.0000
    BogoMIPS:            3600.00
    Flags:               fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse
                         3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb invpcid_single pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid mpx
                          rdseed adx smap clflushopt intel_pt xsaveopt xsavec xgetbv1 xsaves dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp md_clear flush_l1d arch_capabilities
Virtualization features: 
  Virtualization:        VT-x
Caches (sum of all):     
  L1d:                   64 KiB (2 instances)
  L1i:                   64 KiB (2 instances)
  L2:                    512 KiB (2 instances)
  L3:                    4 MiB (1 instance)
NUMA:                    
  NUMA node(s):          1
  NUMA node0 CPU(s):     0-3
Vulnerabilities:         
  Itlb multihit:         KVM: Mitigation: VMX disabled
  L1tf:                  Mitigation; PTE Inversion; VMX conditional cache flushes, SMT vulnerable
  Mds:                   Mitigation; Clear CPU buffers; SMT vulnerable
  Meltdown:              Mitigation; PTI
  Mmio stale data:       Mitigation; Clear CPU buffers; SMT vulnerable
  Retbleed:              Mitigation; IBRS
  Spec store bypass:     Mitigation; Speculative Store Bypass disabled via prctl
  Spectre v1:            Mitigation; usercopy/swapgs barriers and __user pointer sanitization
  Spectre v2:            Mitigation; IBRS, IBPB conditional, RSB filling, PBRSB-eIBRS Not affected
  Srbds:                 Mitigation; Microcode
  Tsx async abort:       Not affected
```

the b210 is already connected via usb3

thanks for your support though :+1:

mubk00, 2023-03-06:
> Intel(R) Core(TM) i5-6260U CPU @ 1.80GHz

Yeah. Just from looking at the specs, running it might be difficult.  You can monitor the system performance in the background, which gives you a small idea about how it does handling the workload. 

Could you attempt reducing the bandwidth and adjusting the samp rate accordingly? I'm just curious to see how much you can reduce this underflow/late/overflow problem.

eulenleber, 2023-03-06:
How would I achieve that?
What I just tested unsuccessfully:
```
cell_cfg:
  channel_bandwidth_MHz: 5         # Bandwith in MHz. Number of PRBs will be automatically derived.
  common_scs: 15
```

Is there another other "screws to adjust" ?

gustavobsch, 2023-03-06:
Sorry to interject but I'm also seeing underflows and I have a pretty decent computer with a Intel(R) Core(TM) i9-10900 CPU @ 2.80GHz 

# gnb  -c /etc/srsran/gnb.yaml 
Available radio types: uhd.

--== srsRAN gNB (commit 0523be699) ==--

[INFO] [UHD] linux; GNU C++ version 9.2.1 20200304; Boost_107100; UHD_3.15.0.0-2build5
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Making USRP object with args 'type=b200'
Cell pci=1, bw=10 MHz, dl_arfcn=499200 (n41), dl_freq=2496.0 MHz, dl_ssb_arfcn=499230, ul_freq=2496.0 MHz

==== gNodeB started ===
Type <t> to view trace
Late: 0; Underflow: 4; Overflow: 0;
Late: 0; Underflow: 2; Overflow: 0;
Late: 0; Underflow: 5; Overflow: 0;
Late: 0; Underflow: 6; Overflow: 0;
Late: 0; Underflow: 2; Overflow: 0;
Late: 0; Underflow: 6; Overflow: 0;
Late: 0; Underflow: 3; Overflow: 0;

# top

top - 10:27:59 up 25 days, 53 min,  1 user,  load average: 3.15, 2.77, 3.21
Tasks: 1053 total,   1 running, 905 sleeping,   0 stopped, 147 zombie
%Cpu(s):  7.8 us,  3.5 sy,  0.0 ni, 87.7 id,  0.3 wa,  0.0 hi,  0.7 si,  0.0 st
MiB Mem :  96458.8 total,   1550.6 free,  68073.5 used,  26834.6 buff/cache
MiB Swap:      0.0 total,      0.0 free,      0.0 used.  27189.6 avail Mem 

    PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND                                                                                  
3895415 root      20   0 1849916 869552  26464 S  30.2   0.9   0:07.74 gnb     <<<<<<<<<<<<<<<< using just 30% of one core

eulenleber, 2023-03-07:
for me its approx 130% and 170% respectively, just fyi

ismagom, 2023-03-07:
Hi @eulenleber , @gustavobsch , 

Thanks for your help debugging the issue. Could you try running this command and paste here the output, please?  

```
sudo /usr/lib/uhd/examples/benchmark_rate --args type=b200 --rx_rate 23.04e6 --tx_rate 23.04e6

[INFO] [UHD] linux; GNU C++ version 9.2.1 20200304; Boost_107100; UHD_3.15.0.0-2build5
[00:00:00.000001] Creating the usrp device with: type=b200...
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz... 
[INFO] [B200] Actually got clock rate 16.000000 MHz.
Using Device: Single USRP:
  Device: B-Series Device
  Mboard 0: B210
  RX Channel: 0
    RX DSP: 0
    RX Dboard: A
    RX Subdev: FE-RX2
  RX Channel: 1
    RX DSP: 1
    RX Dboard: A
    RX Subdev: FE-RX1
  TX Channel: 0
    TX DSP: 0
    TX Dboard: A
    TX Subdev: FE-TX2
  TX Channel: 1
    TX DSP: 1
    TX Dboard: A
    TX Subdev: FE-TX1

[00:00:01.006084] Setting device timestamp to 0...
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] Actually got clock rate 23.040000 MHz.
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] OK
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] OK
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] OK
[00:00:01.464820] Testing receive rate 23.040000 Msps on 1 channels
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] OK
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] OK
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] OK
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] OK
[00:00:01.489160] Testing transmit rate 23.040000 Msps on 1 channels
[00:00:01.589959] Tx timeouts: 1
[00:00:11.742031] Benchmark complete.

Benchmark rate summary:
  Num received samples:     235554807
  Num dropped samples:      0
  Num overruns detected:    0
  Num transmitted samples:  230462880
  Num sequence errors (Tx): 0
  Num sequence errors (Rx): 0
  Num underruns detected:   0
  Num late commands:        0
  Num timeouts (Tx):        2
  Num timeouts (Rx):        0

Done!
```

This will run a benchmark test between the CPU and the USRP. Note that this tool might be somewhere else, depending on how you installed UHD. 

After that, could you run again the gnb binary and send us the log please?

eulenleber, 2023-03-07:
### i5-6260U

<details>
<summary>benchmark</summary>

```
sudo /usr/lib/uhd/examples/benchmark_rate --args type=b200 --rx_rate 23.04e6 --tx_rate 23.04e6

[INFO] [UHD] linux; Clang version 14.0.0 ; Boost_107400; UHD_4.4.0.0-0-g5fac246b
[00:00:00.009717] Creating the usrp device with: type=b200...
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz... 
[INFO] [B200] Actually got clock rate 16.000000 MHz.
Using Device: Single USRP:
  Device: B-Series Device
  Mboard 0: B210
  RX Channel: 0
    RX DSP: 0
    RX Dboard: A
    RX Subdev: FE-RX2
  RX Channel: 1
    RX DSP: 1
    RX Dboard: A
    RX Subdev: FE-RX1
  TX Channel: 0
    TX DSP: 0
    TX Dboard: A
    TX Subdev: FE-TX2
  TX Channel: 1
    TX DSP: 1
    TX Dboard: A
    TX Subdev: FE-TX1

[00:00:01.87124557] Setting device timestamp to 0...
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] Actually got clock rate 23.040000 MHz.
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] OK
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] OK
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] OK
[00:00:01.552242552] Testing receive rate 23.040000 Msps on 1 channels
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] OK
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] OK
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] OK
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] OK
Setting TX spp to 2040
[00:00:01.578342772] Testing transmit rate 23.040000 Msps on 1 channels
UOUO[00:00:11.581427411] Benchmark complete.

Benchmark rate summary:
  Num received samples:     230898937
  Num dropped samples:      94599
  Num overruns detected:    2
  Num transmitted samples:  230442480
  Num sequence errors (Tx): 0
  Num sequence errors (Rx): 0
[... 1334 lines cut ...]
2023-03-07T11:16:10.502223 [Error notifier] [I] Unavailable data to transmit for sector 0, slot 160.2 and symbol 13.
2023-03-07T11:16:10.502266 [Error notifier] [I] Unavailable data to transmit for sector 0, slot 160.3 and symbol 14.
2023-03-07T11:16:10.502346 [Error notifier] [I] Unavailable data to transmit for sector 0, slot 160.3 and symbol 15.
2023-03-07T11:16:10.502365 [Error notifier] [I] Unavailable data to transmit for sector 0, slot 160.3 and symbol 16.
2023-03-07T11:16:10.502380 [Error notifier] [I] Unavailable data to transmit for sector 0, slot 160.3 and symbol 17.
2023-03-07T11:16:10.502397 [Error notifier] [I] Unavailable data to transmit for sector 0, slot 160.3 and symbol 18.
2023-03-07T11:16:10.502408 [FAPI    ] [W] [     0.0] Received DL_TTI.request message out of time. Current slot is 160.5 while message corresponds to 160.2
2023-03-07T11:16:10.502415 [FAPI    ] [W] [     0.0] Received UL_TTI.request message out of time. Current slot is 160.5 while message corresponds to 160.2
2023-03-07T11:16:10.502529 [FAPI    ] [W] [     0.0] Received DL_TTI.request message out of time. Current slot is 160.5 while message corresponds to 160.3
2023-03-07T11:16:10.502531 [FAPI    ] [W] [     0.0] Received UL_TTI.request message out of time. Current slot is 160.5 while message corresponds to 160.3
2023-03-07T11:16:10.502645 [FAPI    ] [W] [     0.0] Received DL_TTI.request message out of time. Current slot is 160.5 while message corresponds to 160.4
2023-03-07T11:16:10.502647 [FAPI    ] [W] [     0.0] Received UL_TTI.request message out of time. Current slot is 160.5 while message corresponds to 160.4
2023-03-07T11:16:22.973148 [GNB     ] [I] Stopping lower PHY...
2023-03-07T11:16:22.973222 [GNB     ] [I] Lower PHY stopped successfully
2023-03-07T11:16:22.973224 [GNB     ] [I] Closing network connections...
2023-03-07T11:16:22.973233 [IO-EPOLL] [E] epoll_ctl failed for fd=-1
2023-03-07T11:16:22.973235 [GNB     ] [I] Network connections closed successfully
2023-03-07T11:16:22.973236 [GNB     ] [I] Stopping executors...
2023-03-07T11:16:22.973948 [RF      ] [W] Radio realtime event: Type=underflow Source=tx Timestamp=na
2023-03-07T11:16:22.973951 [ALL     ] [E] Cannot push anymore tasks into the radio worker queue. maximum size is 2048
2023-03-07T11:16:22.974095 [GNB     ] [I] Executors stopped successfully
2023-03-07T11:16:23.010365 [SCTP-GW ] [E] Socket not initialized
```

</details>

smunaut, 2023-03-07:
@eulenleber To change cell bandwidth you need to adjust `srate` and `channel_bandwidth_MHz`.
However there are other limits. For instance `n78` band has a minimum 10 MHz bandwidth (and even then I could only get my phone to connect when using 20 MHz bandwidth). Also you tried setting scs to 15 kHz but `srsRAN_Project` only supports 30 kHz  for `n78`.

And about your earlier comment with not seeing the cell on the ascii dft : 5G cells are much more power efficient and really only TX when they have something to TX (as opposed to LTE for instance which always transmits pilots signals). So that means, they are "harder" to see when inactive. You'd need to use something like fosphor to make sure not to miss anything to see them.

`Error connecting to 127.0.1.100:38412` also means no connection to Open5GS which in my experience also prevents the network from showing up.

eulenleber, 2023-03-07:
@smunaut you are right, but the default values seem to be already to cpu performant. I wouldnt know how to reduce the bandwith even further in order to achieve better cpu results

I already assumed that, thanks for the clarification

We have to distinguish between the default config which I only use to debug that Late/Underflow thingy while the working Open5gs core is used in the non-default config. I just guessed that it doesnt make any sense to go further if that Late/Underflow thing still persists.

ismagom, 2023-03-07:
You mentioned you have a working setup where it connects to Open5gs. Could you send us logs and config for that setup please?

eulenleber, 2023-03-07:
@ismagom

that just on the weaker i5-6260U:

<details>
<summary>gnb.conf</summary>

```
amf:
  addr: 127.0.0.5                 # The address or hostname of the AMF.
  bind_addr: 127.0.0.1              # A local IP that the gNB binds to for traffic from the AMF.

rf_driver:
  device_driver: uhd                # The RF driver name.
  device_args: type=b200            # Optionally pass arguments to the selected RF driver.
  clock:                            # Specify the clock source used by the RF.
  sync:                             # Specify the sync source used by the RF.
  srate: 11.52                      # RF sample rate might need to be adjusted according to selected bandwidth.
  #srate: 30.72                      # RF sample rate might need to be adjusted according to selected bandwidth.
  tx_gain: 50                       # Transmit gain of the RF might need to adjusted to the given situation.
  rx_gain: 60                       # Receive gain of the RF might need to adjusted to the given situation.

cell_cfg:
  dl_arfcn: 632628                  # ARFCN of the downlink carrier (center frequency).
  band: 78                          # The NR band.
  channel_bandwidth_MHz: 10         # Bandwith in MHz. Number of PRBs will be automatically derived.
  common_scs: 30                    # Subcarrier spacing in kHz used for data.
  #dl_arfcn: 368500                  # ARFCN of the downlink carrier (center frequency).
  #band: 3                          # The NR band.
  #channel_bandwidth_MHz: 15         # Bandwith in MHz. Number of PRBs will be automatically derived.
  #common_scs: 15                    # Subcarrier spacing in kHz used for data.
  plmn: "90170"                     # PLMN broadcasted by the gNB.
  tac: 1                            # Tracking area code (needs to match the core configuration).
  pci: 1                            # Physical cell ID.
      
log:
  filename: /tmp/gnb.log            # Path of the log file.
  all_level: warning                # Logging level applied to all layers.
  #all_level: debug                # Logging level applied to all layers.

pcap:
  mac_enable: false                 # Set to true to enable MAC-layer PCAPs.
  #mac_enable: true                 # Set to true to enable MAC-layer PCAPs.
  mac_filename: /tmp/gnb_mac.pcap   # Path where the MAC PCAP is stored.
  ngap_enable: false                # Set to true to enable NGAP PCAPs.
  #ngap_enable: true                # Set to true to enable NGAP PCAPs.
  ngap_filename: /tmp/gnb_ngap.pcap # Path where the NGAP PCAP is stored.
```
</details>

<details>
<summary>stdout</summary>

```
sudo gnb -c gnb.conf
Available radio types: uhd.

--== srsRAN gNB (commit 77be7d339) ==--

[INFO] [UHD] linux; Clang version 14.0.0 ; Boost_107400; UHD_4.4.0.0-0-g5fac246b
Making USRP object with args 'type=b200'
Cell pci=1, bw=10 MHz, dl_arfcn=632628 (n78), dl_freq=3489.42 MHz, dl_ssb_arfcn=632640, ul_freq=3489.42 MHz

==== gNodeB started ===
Type <t> to view trace
Late: 38; Underflow: 173; Overflow: 0;
Late: 4; Underflow: 49; Overflow: 0;
Late: 3; Underflow: 40; Overflow: 0;
Late: 9; Underflow: 64; Overflow: 0;
Late: 0; Underflow: 39; Overflow: 0;
[... 899 lines cut ...]
2023-03-07T15:09:05.724993 [RF      ] [W] Radio realtime event: Type=start-of-burst Source=tx Timestamp=105690813
2023-03-07T15:09:05.791676 [RF      ] [W] Radio realtime event: Type=end-of-burst Source=tx Timestamp=106451544
2023-03-07T15:09:05.791679 [RF      ] [W] Radio realtime event: Type=underflow Source=tx Timestamp=na
2023-03-07T15:09:05.791816 [RF      ] [W] Radio realtime event: Type=start-of-burst Source=tx Timestamp=106454010
2023-03-07T15:09:05.792243 [RF      ] [W] Radio realtime event: Type=underflow Source=tx Timestamp=na
2023-03-07T15:09:05.792247 [RF      ] [W] Radio realtime event: Type=end-of-burst Source=tx Timestamp=106458537
2023-03-07T15:09:05.792252 [RF      ] [W] Radio realtime event: Type=underflow Source=tx Timestamp=na
2023-03-07T15:09:05.792381 [RF      ] [W] Radio realtime event: Type=start-of-burst Source=tx Timestamp=106460592
2023-03-07T15:09:05.875633 [RF      ] [W] Radio realtime event: Type=underflow Source=tx Timestamp=na
2023-03-07T15:09:05.875652 [RF      ] [W] Radio realtime event: Type=end-of-burst Source=tx Timestamp=107418396
2023-03-07T15:09:05.875799 [RF      ] [W] Radio realtime event: Type=start-of-burst Source=tx Timestamp=107420457
2023-03-07T15:09:05.875903 [RF      ] [W] Radio realtime event: Type=late Source=tx Timestamp=na
2023-03-07T15:09:05.875910 [RF      ] [W] Radio realtime event: Type=end-of-burst Source=tx Timestamp=107421690
2023-03-07T15:09:05.876013 [RF      ] [W] Radio realtime event: Type=underflow Source=tx Timestamp=na
2023-03-07T15:09:05.885222 [RF      ] [W] Radio realtime event: Type=start-of-burst Source=tx Timestamp=107536479
2023-03-07T15:09:05.922095 [GNB     ] [I] Stopping lower PHY...
2023-03-07T15:09:05.922127 [GNB     ] [I] Lower PHY stopped successfully
2023-03-07T15:09:05.922129 [GNB     ] [I] Closing network connections...
2023-03-07T15:09:05.922142 [GNB     ] [I] Network connections closed successfully
2023-03-07T15:09:05.922149 [GNB     ] [I] Stopping executors...
2023-03-07T15:09:05.922770 [RF      ] [W] Radio realtime event: Type=underflow Source=tx Timestamp=na
2023-03-07T15:09:05.922795 [ALL     ] [E] Cannot push anymore tasks into the radio worker queue. maximum size is 2048
2023-03-07T15:09:05.922985 [GNB     ] [I] Executors stopped successfully
```
</details>

ismagom, 2023-03-07:
Hi again @eulenleber . We've been testing your same commit in a variety of CPUs, using B210 and UHD versions 3.15 and 4.1. Unfortunately we haven't been able to reproduce your issue. To give some explanation about what is happening here, all these log messages: 

```
Radio realtime event:...
Unavailable data to transmit for sector...
Received UL_TTI.request message out of time...
```

Are all real-time processing errors at different stages of the processing. Essentially the CPU is late to generate a given result. In a small CPU you could expect some of these messages to show up, specially if you are running a GUI, have your wifi enabled or NIC card doing internet traffic, your GPU etc, etc. All these devices interrupt the CPU and can cause these sort of issues. However, in your case these errors are seen continuously from the start, even with no users in the system, which is not typical.

We have tested the following CPUs: i7-8550U, i7-7700, i7-8559U and i7-6770HQ. We don't have anything as small as your i5-6260U which is a dual-core. In all these 4 i7 CPUs, we can run a cell with up to TDD 20 MHz and we just see a few of these RT errors. 

My latest recommendation to try would be to make sure the CPU governor is set to maximum (sometimes using GUI tools don't set it correctly), make sure you disable the wifi card, maybe try to run without GUI (in headless mode), and make sure you don't have background processes that could be using CPU time. I guess you have also compiled in Release mode (is the default). 

If I think of something else to try, I'll let you know.

gustavobsch, 2023-03-07:
> Hi again @eulenleber . We've been testing your same commit in a variety of CPUs, using B210 and UHD versions 3.15 and 4.1. Unfortunately we haven't been able to reproduce your issue. To give some [...]

Thanks for this input. I moved my testing to another, dedicated, baremetal node with an Intel(R) Core(TM) i7-3770S CPU @ 3.10GHz and now I don't see more underflows 

Ensure the CPU governor is performance, I think Ubuntu defaults to powersave

`# echo performance | tee /sys/devices/system/cpu/cpu*/cpufreq/scaling_governor
`

Here's the output of the benchmark 

```
[00:00:20.805364459] Benchmark complete.
Benchmark rate summary:
  Num received samples:     236749621
  Num dropped samples:      0
  Num overruns detected:    0
  Num transmitted samples:  236223840
  Num sequence errors (Tx): 0
  Num sequence errors (Rx): 0
  Num underruns detected:   0
  Num late commands:        0
  Num timeouts (Tx):        0
  Num timeouts (Rx):        0
Done!
```

And the output of gnb with default config, no more underflows! I let it a couple of minutes running

```
# gnb -c /usr/local/share/srsran/gnb_rf_b200_tdd_n78_10mhz.yml
Available radio types: uhd.

--== srsRAN gNB (commit 0523be699) ==--

[INFO] [UHD] linux; GNU C++ version 11.2.0; Boost_107400; UHD_4.1.0.5-3
Making USRP object with args 'type=b200'
Cell pci=1, bw=10 MHz, dl_arfcn=632628 (n78), dl_freq=3489.42 MHz, dl_ssb_arfcn=632640, ul_freq=3489.42 MHz

==== gNodeB started ===
Type <t> to view trace
^CStopping ..

```

This node as a slower CPU and doesn't see this issue. I think the other worker node has too many CPU interrupts because the thousand or so tasks running..

The bad news is that I still can't see the network from the UE or see any activity on the GQRX waterfall. With the enb I was able to see it.

I'm waiting for the GPSDO to arrive, it might be related to that..

smunaut, 2023-03-07:
GQRX does very infrequent sampling you're unlikely to see it.

As for seeing it on the phone, I'm not sure what UE you use but :
(1) Did you ever see your 5G network on it ?  COTS UE are _VERY_ fussy ...
(2) Even on a known working setup I was never able to see n78 10 MHz network, I had to make it 20 MHz wide for the phone to see it ...

gustavobsch, 2023-03-07:
I have a Galaxy A22 and I have not been able to make it work yet. This phone only supports band 28 and 41 so kind of limited there.

My testing has been limited to band 41, catching errors like this underflow issue and monitoring the waterfall when testing other bands. With enb I see a lot of activity once I run the program on different bands, without any UE connecting to it.. so I was using that behavior as baseline. Do you know if gnb behaves differently? 

I purchased a 'One Plus Nord N10' UE which supports more 5G bands and an internal GPSDO, waiting for them to arrive and test again. At least the underflow issue is resolved now. 

Do you know if fosphor is better than GQRX for monitoring?

smunaut, 2023-03-07:
Yes, as I mentioned before `gnb` behaves very differently. In LTE eve if there is no phone active there pilot carriers are always transmitted and they "fill out" enough of the spectrum over enough of the time to make the cell very visible.  In 5G/NR, those don't exists, which mean when a phone is not active, only the SSB is transmitted and it's very short, doesn't fill much bandwidth and doesn't happen very often (in percentage of time).
And fosphor will undoubtedly catch the cell since it's designed to catch short events.

I'm assuming you mean the  Galaxy A22 5G ?  (since the plain A22 doesn't have 5G at all). And AFAICT it supports more band.
However as I mentioned COTS phones are fussy ... I tried 3 of them before finding one that worked ... And even then it only works with PLMN 001/01 and when in SIM1  (same phone/same sim in the SIM2 slot and it doesn't work).

You also need to make sure your SIM is properly setup for 5G.

gustavobsch, 2023-03-07:
What UE are you using?

Correct, Galaxy A22 5G, I was using the information from https://cacombos.com/device/SM-A226B and they only have bands 28 and 41 listed as supported... but I see on other websites they have listed more bands, as you mentioned, that's great news I will test other bands.

I will also test with PLMN 001/01 on SIM1 slot and monitoring with fosphor

smunaut, 2023-03-07:
Well that site also only list 5G NR NSA band, no SA bands at all ... (and gnb is a SA network)

I'm using a One Plus Nord CE 5G (EB2103).

ismagom, 2023-03-07:
For reference, we have the One Plus Nord 5G model AC2003 and also works well in both TDD and FDD in 5G SA. 

However, the minimum bandwidth for FDD is 10 MHz and for TDD is 20 MHz. 

We'll gather a list of supported COTS UE and publish it soon.

eulenleber, 2023-03-08:
Well, thanks for the advice. I could at least reduce the values, I guess the biggest impact had indeed the Release build 🫣
But Im still not able to eliminate the underflows completely.

I disabled most of the systemd services, isolated multi user, killed unnecessary processes.
@gustavobsch have you also an open5gs core net running on the same machine?
(well without open5gs core running I can improve the performance a little bit more :D)

But thank you all for the explanations.

smunaut, 2023-03-08:
Other things you can look at:
* Make sure you memory is dual-channel
* Use a discrete GPU and/or make sure to disable any GUI ( built-in GPU uses main memory as display mem, this hogs memory bandwidth and blocks the bus for long stretch, increasing latency of other peripherals)

eulenleber, 2023-03-08:
just for documentation purposes:
```
sudo dmidecode -t 17  | grep Locator
	Locator: ChannelA-DIMM0
	Bank Locator: BANK 0
	Locator: ChannelB-DIMM0
	Bank Locator: BANK 2
```
since there are 2 channels (A and B) the memory is in dual channel

Im not able to change the GPU

I will change the system and will report back if that does not work.

Thanks for the support

closed 2023-03-08 by eulenleber

gustavobsch, 2023-03-08:
> Well, thanks for the advice. I could at least reduce the values, I guess the biggest impact had indeed the Release build 🫣 But Im still not able to eliminate the underflows completely.
>
> I disabled [...]

I don't have open5gs installed on this node. The node giving me issues also is also running GPU tasks so I think it was just a matter of too many CPU interrupts.

I definitely recommend fosphor, it's so much more sensible than GQRX.

Have you checked your usb bus usage? for reference this is the usage I observe running band 41 with 10mhz bw 

```

# lsusb 
Bus 004 Device 002: ID 2500:0020 Ettus Research LLC USRP B210

# usbtop --bus usbmon4

Bus ID 4 (Raw USB traffic, bus number 4)	To device	From device
  Device ID 1 :			0.00 kb/s	0.00 kb/s
  Device ID 2 :			46363.56 kb/s	46770.42 kb/s

# top
top - 08:41:00 up 23 min,  2 users,  load average: 0.49, 0.48, 0.35
Tasks: 177 total,   1 running, 176 sleeping,   0 stopped,   0 zombie
%Cpu(s):  6.8 us,  4.8 sy,  0.0 ni, 86.0 id,  0.0 wa,  0.0 hi,  2.4 si,  0.0 st
MiB Mem :  15911.0 total,  14141.8 free,   1121.7 used,    647.5 buff/cache
MiB Swap:   1956.0 total,   1956.0 free,      0.0 used.  14525.6 avail Mem 

    PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND                                                                            
  34548 root      20   0 1955060 868904  25088 S  78.1   5.3   0:20.92 gnb   <<<< one core Intel i7-3770S

# cat /sys/devices/system/cpu/cpu*/cpufreq/scaling_governor
performance
performance
performance
performance
performance
performance
performance
performance

# gnb -c /root/gnb.yml 
Available radio types: uhd.

--== srsRAN gNB (commit 0523be699) ==--

[INFO] [UHD] linux; GNU C++ version 11.2.0; Boost_107400; UHD_4.1.0.5-3
Making USRP object with args 'type=b200'
Cell pci=1, bw=10 MHz, dl_arfcn=499200 (n41), dl_freq=2496.0 MHz, dl_ssb_arfcn=499230, ul_freq=2496.0 MHz

==== gNodeB started ===
Type <t> to view trace

```

alvaroalfaro612, 2023-03-15:
> What UE are you using?
>
> Correct, Galaxy A22 5G, I was using the information from https://cacombos.com/device/SM-A226B and they only have bands 28 and 41 listed as supported... but I see on other [...]

Hello @gustavobsch , did you get to connect this UE to the gnb? 
I have a doubt, should be possible to see the network with a standard SIM of a telecom provider? If it is possible, it will be easier to identify which UE are able to see the network. I have tried with Xiami Redmi Note 10 5G and Poco F3 and I don't see the network in any of them.

eulenleber, 2023-03-16:
I just changed the system to:
```
Architecture:            x86_64
  CPU op-mode(s):        32-bit, 64-bit
  Address sizes:         39 bits physical, 48 bits virtual
  Byte Order:            Little Endian
CPU(s):                  12
  On-line CPU(s) list:   0-11
Vendor ID:               GenuineIntel
  Model name:            Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz
    CPU family:          6
    Model:               158
    Thread(s) per core:  2
    Core(s) per socket:  6
    Socket(s):           1
    Stepping:            10
    CPU max MHz:         4100,0000
    CPU min MHz:         800,0000
    BogoMIPS:            4399.99
    Flags:               fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse
                         3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb invpcid_single pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust sgx bmi1 avx2 smep bmi2 erms invpcid
                          mpx rdseed adx smap clflushopt intel_pt xsaveopt xsavec xgetbv1 xsaves dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp sgx_lc md_clear flush_l1d arch_capabilities
Virtualization features: 
  Virtualization:        VT-x
Caches (sum of all):     
  L1d:                   192 KiB (6 instances)
  L1i:                   192 KiB (6 instances)
  L2:                    1,5 MiB (6 instances)
  L3:                    9 MiB (1 instance)
NUMA:                    
  NUMA node(s):          1
  NUMA node0 CPU(s):     0-11
Vulnerabilities:         
  Itlb multihit:         KVM: Mitigation: VMX disabled
  L1tf:                  Mitigation; PTE Inversion; VMX conditional cache flushes, SMT vulnerable
  Mds:                   Mitigation; Clear CPU buffers; SMT vulnerable
  Meltdown:              Mitigation; PTI
  Mmio stale data:       Mitigation; Clear CPU buffers; SMT vulnerable
  Retbleed:              Mitigation; IBRS
  Spec store bypass:     Mitigation; Speculative Store Bypass disabled via prctl
  Spectre v1:            Mitigation; usercopy/swapgs barriers and __user pointer sanitization
  Spectre v2:            Mitigation; IBRS, IBPB conditional, RSB filling, PBRSB-eIBRS Not affected
  Srbds:                 Mitigation; Microcode
  Tsx async abort:       Not affected
```
```
sudo /usr/bin/gnb -c ~/5g/srsran-config/gnb.conf
Available radio types: uhd.

--== srsRAN gNB (commit 0523be699) ==--

[INFO] [UHD] linux; Clang version 14.0.0 ; Boost_107400; UHD_4.4.0.0-47-gd18647dd
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Making USRP object with args 'type=b200'
Cell pci=1, bw=10 MHz, dl_arfcn=632628 (n78), dl_freq=3489.42 MHz, dl_ssb_arfcn=632640, ul_freq=3489.42 MHz

==== gNodeB started ===
Type <t> to view trace
Late: 0; Underflow: 2; Overflow: 0;
Late: 0; Underflow: 1; Overflow: 0;
Late: 0; Underflow: 5; Overflow: 0;
Late: 0; Underflow: 3; Overflow: 0;
```

The amount and frequency of the underflows is heavily reduced but still not eliminated completely

Is that normal behaviour?

gustavobsch, 2023-04-02:
Just an update on my issue. After using external GPSDO I can attach correctly to the network on both b3 and b78 using the One Plus Nord N10. The galaxy A22 will not see the network using the same SIM / srsRAN config combination.

andreaterni7, 2024-10-24:
Hello @gustavobsch, can you please share your gNodeB config files for both the bands? Thank you in advance