srsran/srsRAN_Project #11: Could not find valid 5QI 5

EugeneBogush, 2023-03-02:
## Issue Description ##
2023-03-02T10:01:33.390175 [CU-CP   ] [W] Could not find valid 5QI 5. QoS map size 2
2023-03-02T10:01:33.390176 [CU-CP   ] [E] ue=2: "PDU Session Creation Routine" QoS flow 5QI is not configured. id 1 5QI 5
2023-03-02T10:01:33.390177 [CU-CP   ] [E] ue=2: "PDU Session Creation Routine" failed.

## Setup Details ##
I connected Fibocom FM150  with srsran + open5gs.

The problem occurs when creating pdusession for dnn="IMS"

andrepuschmann, 2023-03-03:
The default 5QI config has only 7 and 9 populated. We should be able to provide an extended config as well. I'll follow up on this. Could you still send full logs please and also PCAPs for this attach? Thanks

andrepuschmann, 2023-03-03:
Hey,
attached please find a example config with FiveQI 5, 7 and 9 configured. Give it a try and let us know how it goes. Can you share PCAPs and logs with us?

```
amf:
  addr: 127.0.1.100                 # The address or hostname of the AMF.
  bind_addr: 127.0.0.1              # A local IP that the gNB binds to for traffic from the AMF.

rf_driver:
  device_driver: uhd                # The RF driver name.
  device_args: type=b200            # Optionally pass arguments to the selected RF driver.
  clock:                            # Specify the clock source used by the RF.
  sync:                             # Specify the sync source used by the RF.
  srate: 11.52                      # RF sample rate might need to be adjusted according to selected bandwidth.
  tx_gain: 50                       # Transmit gain of the RF might need to adjusted to the given situation.
  rx_gain: 60                       # Receive gain of the RF might need to adjusted to the given situation.

cell_cfg:
  dl_arfcn: 632628                  # ARFCN of the downlink carrier (center frequency).
  band: 78                          # The NR band.
  channel_bandwidth_MHz: 10         # Bandwith in MHz. Number of PRBs will be automatically derived.
  common_scs: 30                    # Subcarrier spacing in kHz used for data.
  plmn: "00101"                     # PLMN broadcasted by the gNB.
  tac: 7                            # Tracking area code (needs to match the core configuration).
  pci: 1                            # Physical cell ID.
      
log:
  filename: /tmp/gnb.log            # Path of the log file.
  all_level: warning                # Logging level applied to all layers.

pcap:
  mac_enable: false                 # Set to true to enable MAC-layer PCAPs.
  mac_filename: /tmp/gnb_mac.pcap   # Path where the MAC PCAP is stored.
  ngap_enable: false                # Set to true to enable NGAP PCAPs.
  ngap_filename: /tmp/gnb_ngap.pcap # Path where the NGAP PCAP is stored.

qos:
  -
    five_qi: 5
    rlc:
      mode: am
      am:
        tx:
          sn: 18
          t-poll-retransmit: 80
          max-retx-threshold: 4
          poll-pdu: 64
          poll-byte: 125
        rx:
          sn: 18
          t-reassembly: 80
          t-status-prohibit: 10
  -
    five_qi: 7
    rlc:
      mode: um-bidir
      um-bidir:
        tx:
          sn: 12
        rx:
          sn: 12
          t-reassembly: 50
  -
    five_qi: 9
    rlc:
      mode: am
      am:
        tx:
          sn: 18
          t-poll-retransmit: 80
          max-retx-threshold: 4
          poll-pdu: 64
          poll-byte: 125
        rx:
          sn: 18
          t-reassembly: 80
          t-status-prohibit: 10
```

EugeneBogush, 2023-03-06:
hello,
not helped(.
logs&config in attach 

[gnblogs.tar.gz](https://github.com/srsran/srsRAN_Project/files/10900053/gnblogs.tar.gz)

andrepuschmann, 2023-03-07:
Noted, we are working on this.

andrepuschmann, 2023-03-20:
Hey, can you retry with the current version. It should be fixed. I am closing the issue. Please feel free to reopen if needed.

closed 2023-03-20 by andrepuschmann

huahuaLover, 2024-11-11:
I encountered the same question," _[W] No valid config for 5QI=0x8, [W] Configuration for QFI=2 can't be derived_",I configured  it according to the above,but the terminal reports an error._QoS config, item #0 requires mac_ my srsRAN_Project is 9d5dd742a70e82c0813c34f57982f9507f1b6d5d