srsran/srsRAN_Project #17: This is discussed further in the relevant app note.

eulenleber, 2023-03-06:
> If you plan to connect the gNB to a COTS UE we recommend that you use an external clock source such as an Octoclock of GPSDO that is compatible with your RF-frontend, as the on-board clock within the USRP is not accurate enough to enable a connection with the UE. This is discussed further in the relevant app note.

https://docs.srsran.com/projects/project/en/latest/user_manuals/source/1_getting_started.html

[image]

brendan-mcauliffe, 2023-03-06:
Hey, the app note is coming shortly - we are currently preparing it. 

To your second point, srsUE is no longer under active development and its 5G feature set is limited. I am also not sure if you are referring to a commercial gNB or something like an Amarisoft Callbox. Connecting srsUE to a Callbox is shown [here](https://docs.srsran.com/projects/4g/en/latest/app_notes/source/5g_sa_amari/source/index.html). You need to give more information on your intended use case if this is something you plan on doing. You should also direct any srsUE (and **all** other srsRAN 4G) questions to the [mailing list ](https://lists.srsran.com/mailman/listinfo/srsran-users) as they most likely won't be answered here.

closed 2023-03-06 by brendan-mcauliffe

eulenleber, 2023-03-06:
Thanks,

https://github.com/srsran/srsRAN_Project/issues/10#issuecomment-1450263148
here you said 
> srsUE is not yet compatible with the new gNB, we are working on a fix for this and should have something soon.

Just for clarification of the operative objectives here: is a softUE implementation planned (or is there something already available, if srsUE is not?)