srsran/srsRAN_Project #21: Payload protocol identifier is not specified in SCTP

iman432, 2023-03-07:
## Issue Description ##
Payload protocol identifier is not specified in SCTP

## Setup Details ##
The latest code was built, USRP B200, UHD 4.1

## Expected Behavior ##
Payload protocol identifier should be specified as NGAP for N2 in SCTP

## Steps to reproduce the problem ##
This case can be found in any NGAP packet from gNB to AMF

- gNB to AMF
[image]

- AMF to gNB
[image]

mubk00, 2023-03-07:
Hi.
I was able to reproduce the same problem when testing earlier today (using B210 and X300).

Here are the debugging logs:
```
[2023-03-07T09:17:36.068790] [AMF] [sctp   ] [debug] SCTP Association Change event received
[2023-03-07T09:17:36.068810] [AMF] [sctp   ] [debug] Add new association with id (17812)
[2023-03-07T09:17:36.068850] [AMF] [ngap   ] [debug] Ready to handle new NGAP SCTP association request (id 17812)
[2023-03-07T09:17:36.068853] [AMF] [ngap   ] [debug] Create a new gNB context with assoc_id (17812)
[2023-03-07T09:17:36.070346] [AMF] [sctp   ] [error] Received data from peer with unsolicited PPID (0), 
expecting (60)
```

alvasMan, 2023-03-07:
Thanks for the report.
Something as simple as this should fix the issue:
```
commit ec07389145175e86f9468363e67e934ca3ff1e44 (HEAD -> sctp_ppid, origin/sctp_ppid)
Author: Pedro Alvarez <pedro.alvarez@softwareradiosystems.com>
Date:   Tue Mar 7 18:48:52 2023 +0000

    sctp: fix NGAP ppid

diff --git a/apps/gnb/gnb.cpp b/apps/gnb/gnb.cpp
index ab5b8a00b..fe1a4b295 100644
--- a/apps/gnb/gnb.cpp
+++ b/apps/gnb/gnb.cpp
@@ -92,6 +92,7 @@ static void compute_derived_args(const gnb_appconfig& gnb_params)
   ngap_nw_config.connect_address = gnb_params.amf_cfg.ip_addr;
   ngap_nw_config.connect_port    = gnb_params.amf_cfg.port;
   ngap_nw_config.bind_address    = gnb_params.amf_cfg.bind_addr;
+  ngap_nw_config.ppid            = NGAP_PPID;
 }
 
 namespace {
diff --git a/include/srsran/gateways/sctp_network_gateway.h b/include/srsran/gateways/sctp_network_gateway.h
index 2b897c44e..fba94e626 100644
--- a/include/srsran/gateways/sctp_network_gateway.h
+++ b/include/srsran/gateways/sctp_network_gateway.h
@@ -19,10 +19,13 @@
 
 namespace srsran {
 
+constexpr uint16_t NGAP_PPID = 60; // NGAP PPID, see TS 38.412, section 7.
+
 /// \brief Configuration for SCTP network gateway
 struct sctp_network_gateway_config : common_network_gateway_config {
   std::string connect_address;
   int         connect_port = 0;
+  int         ppid         = 0; /// the Payload Protocol Identifier
   // TODO add SCTP specific options
 };
 
diff --git a/lib/gateways/sctp_network_gateway_impl.cpp b/lib/gateways/sctp_network_gateway_impl.cpp
index 48e374660..0cb6c0cd1 100644
--- a/lib/gateways/sctp_network_gateway_impl.cpp
+++ b/lib/gateways/sctp_network_gateway_impl.cpp
@@ -546,7 +546,7 @@ void sctp_network_gateway_impl::handle_pdu(const byte_buffer& pdu)
                                 pdu_span.size_bytes(),
                                 (struct sockaddr*)&server_addr,
                                 server_addrlen,
-                                ppi,
+                                htonl(config.ppid),
                                 0,
                                 stream_no,
                                 0,
diff --git a/lib/gateways/sctp_network_gateway_impl.h b/lib/gateways/sctp_network_gateway_impl.h
index 45df3cbe2..4d1b594d6 100644
--- a/lib/gateways/sctp_network_gateway_impl.h
+++ b/lib/gateways/sctp_network_gateway_impl.h
@@ -90,7 +90,6 @@ private:
   int                     server_ai_socktype = 0;
   int                     server_ai_protocol = 0;
 
-  unsigned ppi       = 0; /// the Payload Protocol Identifier (e.g. 18 for S1AP)
   unsigned stream_no = 0; /// the stream number to use for sending
 };
```

mubk00, 2023-03-07:
Thanks.

Earlier today, I attempted to hard-code the ppid value to 60 but obtained the same results. I have also now just applied the patch you provided here, recompiled the project, and re-tested running gNB and 5GC. However, I am still getting the same issue with `[2023-03-07T20:00:21.495902] [AMF] [sctp   ] [error] Received data from peer with unsolicited PPID (0), expecting (60)`

Can you @iman432 confirm this to see if it works for you? I'll have to re-check my setup, I might be doing something wrong somewhere.

alvasMan, 2023-03-08:
Hum... strange. I tested it, but only with the NG setup, tough.
I'll test again with the full attach.

Do you have a pcap?

alvasMan, 2023-03-08:
@mubk00, just tested again (now with a full attach) and it seems fine to me. Did you apply the patch correctly?
You should be able to just do `git am < 0001-sctp-fix-NGAP-ppid.patch.txt` and re-compile.

[0001-sctp-fix-NGAP-ppid.patch.txt](https://github.com/srsran/srsRAN_Project/files/10920913/0001-sctp-fix-NGAP-ppid.patch.txt)

If you did apply the patch and are still seeing the error can you capture the traffic between the gNB and the core and attach the PCAP?

mubk00, 2023-03-08:
@alvasMan  I apologize for the delay in my response. Due to unforeseen circumstances, I was unable to reply to your message earlier. I'll re-try again on Friday when I have tested it with the setup, and I'll provide the logs. Thanks.

mubk00, 2023-03-09:
Hi.

I tried again with the B210. I re-cloned the project, applied the patch, and re-compiled the project. I seem to be getting the same results.

Here are the logs:
[AMF log (error occurs at line line 152)](https://pastebin.com/6E2z4Nsw)
[SPGWU log](https://pastebin.com/cndiCtTP)

* Also I have attached the PCAP file. Which shows the `NGSetupRequest` at packet no. 59, and this shows the criticality reject (0) for Ngsetup.

Thanks. Let me know if there is anything else that you want me to try out to help troubleshoot this.
...

Also, @iman432, were you able to fix this problem using the patch? since you were also experiencing the same problem

robertfalkenberg, 2023-03-09:
Confirming patch by @alvasMan is setting `payload protocol identifier correctly:

Before:
[image]

After:
[image]

mubk00, 2023-03-09:
Hi.

I just want to mention I got it to work, thanks for the patch @alvasMan

closed 2023-03-11 by iman432