srsran/srsRAN_Project #26: long duration data test dont work

anurag7860, 2023-03-09:
## Issue Description ##
we are trying iperf in DL but data session wont last more than 10 -20 seconds , we tried external clock 10 Mhz also but results are same 

## Setup Details ##
I am  using USRP 205mini and below is config for clock . 
 
rf_driver:
  device_driver: uhd                # The RF driver name.
  device_args: type=b200            # Optionally pass arguments to the selected RF driver.  
  clock: external                      # Specify the  clock source used by the RF.  
 sync: external  
## Expected Behavior ##
[What you expect to happen]

## Actual Behaviour ##
[What happens instead e.g. error message]

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

mubk00, 2023-03-09:
Hi,

I believe it may be difficult for others to troubleshoot without any kind of logs. It is difficult to extrapolate from what you have described, as the issue could be related to connection stability or even other things. Could you please provide some logs?

andrepuschmann, 2023-03-20:
Closing issue as no further information has been provided by creator.

closed 2023-03-20 by andrepuschmann