srsran/srsRAN_Project #27: USRP X410 Configs

MaryamAmini, 2023-03-09:
## USRP X410 Configs ##
Hi there,
I have been trying to work with the new srsRAN-project with our USRP devices. We have USRP B210, and X410. We have successfully run the gNB using the default configurations with B210, but encountered underflow errors when running the gNB with the X410 (the UE is able to attach to the network, but the connection is unstable). I believe that there should be specific configuration files for the X410 that could potentially resolve these errors.

## Setup Details ##

Core Platform: Open5GS
RF device: USRP X410
UHD version: 4.4.0
Os: Ubuntu 20.04.5 LTS
(CPU scaling governor is set to maximum performance)

## Actual Behaviour ##

--== srsRAN gNB (commit 77be7d339) ==--

[INFO] [UHD] linux; GNU C++ version 9.4.0; Boost_107100; UHD_4.4.0.0-0ubuntu1~focal1
Making USRP object with args 'type=x4xx'
[WARNING] [UDP] The send buffer could not be resized sufficiently.
Target sock buff size: 25000000 bytes.
Actual sock buff size: 1048576 bytes.
See the transport application notes on buffer resizing.
Please run: sudo sysctl -w net.core.wmem_max=25000000
[WARNING] [UDP] The send buffer could not be resized sufficiently.
Target sock buff size: 25000000 bytes.
Actual sock buff size: 1048576 bytes.
See the transport application notes on buffer resizing.
Please run: sudo sysctl -w net.core.wmem_max=25000000
[WARNING] [0/Radio#0] Attempting to set tick rate to 0. Skipping.
[WARNING] [UDP] The send buffer could not be resized sufficiently.
Target sock buff size: 25000000 bytes.
Actual sock buff size: 1048576 bytes.
See the transport application notes on buffer resizing.
Please run: sudo sysctl -w net.core.wmem_max=25000000
Cell pci=1, bw=20 MHz, dl_arfcn=368640 (n3), dl_freq=1843.2 MHz, dl_ssb_arfcn=368670, ul_freq=1748.2 MHz

==== gNodeB started ===
Type <t> to view trace
Late: 0; Underflow: 1; Overflow: 0;
Late: 0; Underflow: 1; Overflow: 0;
Late: 0; Underflow: 1; Overflow: 0;
Late: 0; Underflow: 1; Overflow: 0;
Late: 0; Underflow: 1; Overflow: 0;
Late: 0; Underflow: 1; Overflow: 0;
Late: 0; Underflow: 1; Overflow: 0;
Late: 0; Underflow: 1; Overflow: 0;
Late: 0; Underflow: 1; Overflow: 0;
Late: 0; Underflow: 1; Overflow: 0;
Late: 0; Underflow: 1; Overflow: 0;
Late: 0; Underflow: 1; Overflow: 0;

## Steps to reproduce the problem ##
[gnb_rf_X410.txt](https://github.com/srsran/srsRAN_Project/files/10933055/gnb_rf_X410.txt)

The gNB logs and configs are attached

[gnb.log](https://github.com/srsran/srsRAN_Project/files/10933049/gnb.log)

ismagom, 2023-03-09:
Could you try running the command: 

```
sudo sysctl -w net.core.wmem_max=25000000
```

If you still find issues, could you set the log level to info and run again?

```
log: 
  all_level: info
```
thanks

MaryamAmini, 2023-03-09:
I believe I have done it before. But to make sure, I did it and reset the gNB. 
I run into the same issue:
--== srsRAN gNB (commit 77be7d339) ==--

[INFO] [UHD] linux; GNU C++ version 9.4.0; Boost_107100; UHD_4.4.0.0-0ubuntu1~focal1
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Making USRP object with args 'type=x4xx'
[INFO] [MPMD] Initializing 1 device(s) in parallel with args: mgmt_addr=192.168.10.2,type=x4xx,product=x410,serial=327AFE0,name=ni-x4xx-327AFE0,fpga=X4_200,claimed=False,addr=192.168.10.2
[INFO] [MPM.PeriphManager] init() called with device args `fpga=X4_200,mgmt_addr=192.168.10.2,name=ni-x4xx-327AFE0,product=x410,clock_source=internal,time_source=internal'.
[WARNING] [0/Radio#0] Attempting to set tick rate to 0. Skipping.
Cell pci=1, bw=20 MHz, dl_arfcn=368640 (n3), dl_freq=1843.2 MHz, dl_ssb_arfcn=368670, ul_freq=1748.2 MHz

==== gNodeB started ===
Type <t> to view trace
Late: 0; Underflow: 1; Overflow: 0;
Late: 0; Underflow: 1; Overflow: 0;
[March9-gNB.log.zip]

Kindly find the new logs:
(https://github.com/srsran/srsRAN_Project/files/10933499/March9-gNB.log.zip)

Thanks for the super fast response btw! :)

ismagom, 2023-03-09:
I don't think that Underflows are an issue here. I can barely see them in the log. Even a few underflows per second are OK. 

What phone are you using? Is this a COTS UE? 

I just see a couple of issues: 

1) the UL SNR is very low. If you are using a COTS UE, try reducing the tx_gain. This will make the UE believe it's farther from the gNB and will transmit higher power. Alternatively, you can try increasing the rx_gain. 

2) Even without that, the link was stable and it seems the open5Gs core releases the UE. It could be due to some authentication issue. 

Can you enable pcap at both mac and ngap and send us the files? Also if you can send us the AMF log would be useful. 

Thanks

MaryamAmini, 2023-03-09:
Hi again,
I did all the changes and attached the logs here.
Our UE is a Quectel 5G modem. The strange thing is that with USRP B210 we can get up to 50 Mbps in dl in a single UE connection, but with USRP X410, our best dl rate is about 20 Mbps, and we get occasional disconnections. But will not encounter these issues, with the same core, same RAN, and USRP B210!
That's why I initially thought the configs that we're using must be changed. I changed the tx_gain to 40 for this test.

https://drive.google.com/file/d/1u1S8TtwrQallMCzmSDcVRCV_8IY1jFbY/view?usp=sharing

ismagom, 2023-03-09:
The UL SNR looks good now. It seems the UE is disconnecting and reconnecting. I don't see anything indicating what could be the reason. Can you share a log (no pcap needed) of the same run with the B210?

MaryamAmini, 2023-03-09:
Hi again,

After changing the gains on USRP X410 (tx_gain = 30, rx_gain = 20), we could get 53 Mbps on DL and 70 Mbps on UL with a very strong signal strength.  Do you think we can improve these numbers if we use a COTS phone? at the moment we think that the USB connection between the 5G modem and the laptop might limit the rate. What is your opinion? Do you have any 5G-SA phone recommendations for us that works with X410? Also, we are looking for good antennas for this USRP X410, any recommendations?

[March9-T3.zip](https://github.com/srsran/srsRAN_Project/files/10935006/March9-T3.zip)

(I attached the logs for USRP B210 here anyways incase you're interested :))

Thanks very much for the time and help :)

mubk00, 2023-03-09:
@MaryamAmini  What type of 5G modem are you using?

MaryamAmini, 2023-03-09:
Hi there,
we are using Quectel 5G modules.

ismagom, 2023-03-09:
Great you made it work! The UL bitrate is what you can expect for a 20 MHz FDD cell. The DL could be slightly better, maybe up to 60 Mbps, not much more than that. 

What you can try is increasing the BW to 50 MHz. 

The Quectel 5G should be fine, will give you the same bitrate than a COTS phone. 

> Hi again,
>
> After changing the gains on USRP X410 (tx_gain = 30, rx_gain = 20), we could get 53 Mbps on DL and 70 Mbps on UL with a very strong signal strength. Do you think we can improve these [...]

closed 2023-03-09 by ismagom