srsran/srsRAN_Project #30: AMF-UE-NGAP-ID mismatch between the authentication request and response messages.

Shashank6669, 2023-03-10:
## Issue Description ##
An issue previously discovered in the srsRAN 4G 21.10 ([previous discussion here](https://github.com/srsran/srsRAN_4G/issues/819)) has been encountered again. The srsGNB appears to be truncating the AMF-UE-NGAP ID in the Authentication Response.  This results in the AMF responding with an ErrorIndication message containing the error: **radioNetwork: unknown-local-UE-NGAP-ID. (14)**

## Setup Details ##
srsRAN gNB used with USRP B210 and UHD 4.1. 
OS: Ubuntu 22.04 
COTS UE

## Expected Behavior ##
srsRAN should return the correct id-AMF-UE-NGAP-ID assigned by the AMF/5GC in the authentication response message.

## Actual Behaviour ##
srsRAN returns an incorrect id-AMF-UE-NGAP-ID in the authentication response message to the 5GC causing the authentication handshake to produce an error: unknown-local-UE-NGAP-ID (pcap below)

1. Registration request is sent from the UE to  the AMF/5GC.
2. AMF/5GC responds with a DownlinkNASTransport, Authentication request which contains the AMF-UE-NGAP-ID.
3. srsRAN responds to AMF/5GC with a UplinkNASTransport, Authentication response containing the wrong AMF-UE-NGAP-ID.
4. AMF/5GC responds with ErrorIndication message with the error radioNetwork: unknown-local-UE-NGAP-ID (14).

## Steps to reproduce the problem ##

1.  Send Registration request from UE.
2. Verify that the AMF-UE-NGAP-ID is consistent between the authentication request and response messages.

## Additional Information ##

Here are some partial logs for this issue:

**Authentication Request**
```
NG Application Protocol (DownlinkNASTransport)
	NGAP-PDU: initiatingMessage (0)
		initiatingMessage
			procedureCode: id-DownlinkNASTransport (4)
			criticality: ignore (1)
			value
				DownlinkNASTransport
					protocolIEs: 3 items
						Item 0: id-AMF-UE-NGAP-ID
							ProtocolIE-Field
								id: id-AMF-UE-NGAP-ID (10)
								criticality: reject (0)
								value
									AMF-UE-NGAP-ID: 4294969659

```

**Authentication Response**
```
NG Application Protocol (UplinkNASTransport)
	NGAP-PDU: initiatingMessage (0)
		initiatingMessage
			procedureCode: id-UplinkNASTransport (46)
			criticality: ignore (1)
				value
					UplinkNASTransport
						protocolIEs: 4 items
							Item 0: id-AMF-UE-NGAP-ID
								ProtocolIE-Field
									id: id-AMF-UE-NGAP-ID (10)
									criticality: reject (0)
									value
AMF-UE-NGAP-ID: 2364
```

**Error Indication**
```
NG Application Protocol (ErrorIndication)
	NGAP-PDU: initiatingMessage (0)
		initiatingMessage
			procedureCode: id-ErrorIndication (9)
			criticality: ignore (1)
			value
				ErrorIndication
					protocolIEs: 3 items
						Item 0: id-AMF-UE-NGAP-ID
							ProtocolIE-Field
								id: id-AMF-UE-NGAP-ID (10)
								criticality: ignore (1)
								value
									AMF-UE-NGAP-ID: 2364
						Item 1: id-RAN-UE-NGAP-ID
							ProtocolIE-Field
								id: id-RAN-UE-NGAP-ID (85)
								criticality: ignore (1)
								value
									RAN-UE-NGAP-ID: 0
						Item 2: id-Cause
							ProtocolIE-Field
								id: id-Cause (15)
								criticality: ignore (1)
								value
								        Cause: radioNetwork (0)
                                                                                radioNetwork: unknown-local-UE-NGAP-ID (14)
```

srsRAN gNB changes the AMF-UE-NGAP-ID from 4294969659 in the authentication request to 2364 in the authentication response.

andrepuschmann, 2023-03-10:
Thanks for reporting this bug @Shashank6669 . We'll look into it.

andrepuschmann, 2023-03-17:
Hey, we've just pushed an update to `main` that should include a fix for this. Please give it a try. Closing this issue - please reopen if it's still not working for you.

closed 2023-03-17 by andrepuschmann

MaryamAmini, 2023-06-13:
Hi again,
sorry for the delay in answering here.
I'm still having the same issue. I've tried a different UE and also looked at the logs on the core side. To my understanding, the UE doesn't complete the RRC connection and I'm not sure why. There might be an issue with the core + RAN connection, but I don't see any errors and the NGAP connection is established successfully. Anyways, I'm still working on it, but thanks for the help. :)