srsran/srsRAN_Project #32: Does srsue can connect srsgnb and open5Gs

LongRiver-hash, 2023-03-16:
I am currently working on a 5G SA project that requires that I deploy nrUE and gNB with TDD frontends. I have already deployed the gNB from the srsRAN project andOpen5gs as CN.

However, I wonder if the existing srsUE code will attach with the gNB? The documentation states that the gNB can work with any third-party UE or COTS UE, is there support of srsUE already? If anyone has tried, kindly let me know.

andrepuschmann, 2023-03-17:
Thanks @mubk00 for the explanation - it's exactly right. We will provide an update to srsUE with the upcoming srsRAN 4G release that includes the fixes to attach to the srsRAN Project gNB.

dayal007, 2023-04-13:
Hello @andrepuschmann Thanks a lot for letting know that srsUE will not work with the new gNB. Do you know when we can anticipate the functionality? 

We dont use any radios for our research, so we are kind of blocked without the zmq support in the new gNB. Is there a work around using any other non-radio UE?

andrepuschmann, 2023-05-05:
As in #56 and #65 - we've updated both repos and added an appnote describing the required steps in detail https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html

closed 2023-05-05 by andrepuschmann

wmwwimor, 2023-07-13:
> As in #56 and #65 - we've updated both repos and added an appnote describing the required steps in detail https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html

I would like to know if it is possible to implement this architecture：multiple UEs+ZeroMQ+one gnb+5gc?”Can it currently be achieved using GNU Radio? Is it possible to further expand it into such an architecture:multiple UEs+ZeroMQ+multiple gnbs+5gc? I don't want to use physical devices. Is there something similar to ZeroMQ available now that can assist in simulating radio frequencies?