srsran/srsRAN_Project #34: Late, Overflow and underflow error

SusmitaSumaiya, 2023-03-21:
Hello,
I am trying to run a private network using USRP B210. Our licenced frequency band is 3700-3800 MHz, n78, TDD. I integrated USRP B210 to one pc as gNb and having late, overflow and underflow error. I would like to have suggestion to overcome these errors. My gNb configuration -
amf:
  addr: 127.0.0.2             	# The address or hostname of the AMF.
  bind_addr: 127.0.0.1          	# A local IP that the gNB binds to for traffic from the AMF.

rf_driver:
  device_driver: uhd            	# The RF driver name.
  device_args: type=b200        	# Optionally pass arguments to the selected RF driver.
  clock: external                         	# Specify the clock source used by the RF.
  sync:                         	# Specify the sync source used by the RF.
  srate: 11.52                  	# RF sample rate might need to be adjusted according to selected bandwidth.
  tx_gain: 50                   	# Transmit gain of the RF might need to adjusted to the given situation.
  rx_gain: 60                   	# Receive gain of the RF might need to adjusted to the given situation.

cell_cfg:
  dl_arfcn: 653332              	# ARFCN of the downlink carrier (center frequency).
  band: 78                      	# The NR band.
  channel_bandwidth_MHz: 10     	# Bandwith in MHz. Number of PRBs will be automatically derived.
  common_scs: 30                	# Subcarrier spacing in kHz used for data.
  plmn: "00101"                 	# PLMN broadcasted by the gNB.
  tac: 7                        	# Tracking area code (needs to match the core configuration).
  pci: 1                        	# Physical cell ID.

log:
  filename: /tmp/gnb.log        	# Path of the log file.
  all_level: warning            	# Logging level applied to all layers.

pcap:
  mac_enable: false             	# Set to true to enable MAC-layer PCAPs.
  mac_filename: /tmp/gnb_mac.pcap   # Path where the MAC PCAP is stored.
  ngap_enable: false            	# Set to true to enable NGAP PCAPs.
  ngap_filename: /tmp/gnb_ngap.pcap # Path where the NGAP PCAP is stored.
   

I am having following output after starting the gNb
--== srsRAN gNB (commit 0523be699) ==--

[INFO] [UHD] linux; GNU C++ version 9.4.0; Boost_107100; UHD_4.4.0.0-47-gd18647dd
Making USRP object with args 'type=b200'
Cell pci=1, bw=10 MHz, dl_arfcn=653332 (n78), dl_freq=3799.98 MHz, dl_ssb_arfcn=653376, ul_freq=3799.98 MHz

==== gNodeB started ===
Type <t> to view trace
Late: 94; Underflow: 452; Overflow: 40;
Late: 116; Underflow: 507; Overflow: 43;
Late: 120; Underflow: 529; Overflow: 43;
Late: 123; Underflow: 550; Overflow: 45

ismagom, 2023-03-22:
Hi @SusmitaSumaiya. Thanks for your feedback. Could you try the latest version and read the documentation please?

https://docs.srsran.com/projects/project/en/latest/user_manuals/source/running.html#manual-running

SusmitaSumaiya, 2023-03-22:
Hello @ismagom thanks for your swift reply. 
Yes, I tried it with the latest version of srsRAN_project and followed the documentation you mentioned but still having the same output. Is it related to the incompatibility of arfcn value with srate and channelbandwidth? but it's not showing any other error or warning. Which value would you recommend for dl_arfcn: 653332?

ismagom, 2023-03-22:
Can you paste the stdout and attach logs for the new version?

ismagom, 2023-03-22:
Hi @SusmitaSumaiya , I strongly recommend you read the documentation and application notes first to get familiar with the application. 

Also, please use issues to report errors in the software. For general help use "Discussions".

closed 2023-03-22 by ismagom