srsran/srsRAN_Project #37: External GPSDO Error

alvaroalfaro612, 2023-03-22:
## Issue Description ##
When configuring for using external clock, the radio session cannot be created:
```
sudo gnb -c gnb_avag.yml 
Available radio types: uhd.

--== srsRAN gNB (commit 0523be699) ==--

[INFO] [UHD] linux; GNU C++ version 9.4.0; Boost_107100; UHD_4.4.0.HEAD-0-g5fac246b
Making USRP object with args 'type=b200'
srsGNB ERROR: Unable to create radio session.

```
Same external GPSDO function is working on srsRAN 4G gnb by using **device_args = clock=external** in config file, so it should not be a problem related to this external clock...

## Setup Details ##
I am using a OnePlus Nord 2 as UE, USRP b200 mini and Open5Gs as 5G Core.
For using the external clock, I am configuring the yaml with the following lines:
```
rf_driver:
  device_driver: uhd                # The RF driver name.
  device_args: type=b200        # Optionally pass arguments to the selected RF driver.
  clock: external             # Specify the clock source used by the RF.
  sync: external                       # Specify the sync source used by the RF.
  srate: 30.72                   # RF sample rate might need to be adjusted according to selected bandwidth.
  tx_gain: 60                       # Transmit gain of the RF might need to adjusted to the given situation.
  rx_gain: 50   
```

## Expected Behavior ##
Gnb executes using the external GPSDO as reference.

## Actual Behaviour ##
Gnb has an error when tryin to use the external clock.

andrepuschmann, 2023-03-22:
Please check out the latest code and try again - you seem to be using an old commit. We've fixed the logic there.

closed 2023-03-27 by alvaroalfaro612

ismagom, 2023-03-27:
Hi @alvaroalfaro612 , what error does it throw when you select both? Maybe it can not find the PPS signal.

alvaroalfaro612, 2023-03-27:
> Hi @alvaroalfaro612 , what error does it throw when you select both? Maybe it can not find the PPS signal.
Hi, this is the stdout of the console:

```
Error: couldn't set sync source: ValueError: external reference cannot be both a clock source and a time source
srsGNB ERROR: Unable to create radio session.
```

ismagom, 2023-03-27:
That means you are using the b200mini which has only external clock. Then yes, set only clock to external and don't select sync