srsran/srsRAN_Project #42: srsRAN gNB crash after UE trying to establish multiple pdu sessions

ChengDaHaI, 2023-03-24:
Hi,

I am trying to establish multiple pdu sessions from COTS UE to srsRAN gNB. My cots UE (OnePlus Nord & Sierra Wireless 5g modem) work perfect when there is only one pdu session. However, when I try to establish the second PDU session via AT commands, the gNB crashes and throws the error:
```
terminate called after throwing an instance of 'std::out_of_range'
  what():  map::at
```
Does srsRAN support multiple pdu sessions so far?

Thanks
Hai

andrepuschmann, 2023-03-24:
Hey, could you please send logs and PCAPs?

ChengDaHaI, 2023-03-24:
pcap files can be find [here](https://drive.google.com/drive/folders/1ihtn4N-ut1DWd5xEkks_6ly4rb7ZL5Jl?usp=share_link)

The gNB crashes immediately (everytime) after UE try to establish another PDU session
[gnb.log](https://github.com/srsran/srsRAN_Project/files/11067031/gnb.log)

ismagom, 2023-03-25:
Thanks for your feedback. This feature is currently not supported. Your logs/pcaps are very helpful. We'll try to support it soon.

Thanks

closed 2023-03-25 by ismagom

ChengDaHaI, 2023-03-28:
Thanks for your working on this project. Is there a developer guide I could read? I would like to help build this feature on srsRAN project.