srsran/srsRAN_Project #48: COTS UE Cannot find the network

pricalinat, 2023-04-06:
Using the default configuration file, the network can be found in srsRAN_4G, but not in srsRAN_Project. The log is as follows.
[gnb.log](https://github.com/srsran/srsRAN_Project/files/11167569/gnb.log)
[pcap.zip](https://github.com/srsran/srsRAN_Project/files/11167571/pcap.zip)

andrepuschmann, 2023-04-06:
Please send logs and configs and stdout for both cases

pricalinat, 2023-04-06:
srsRAN_4G
[image]
[enb_mac_nr.zip](https://github.com/srsran/srsRAN_Project/files/11168150/enb_mac_nr.zip)
[enb.log](https://github.com/srsran/srsRAN_Project/files/11168152/enb.log)
[conf.zip](https://github.com/srsran/srsRAN_Project/files/11168221/conf.zip)

srsRAN_Project
[image]
[gnb.log](https://github.com/srsran/srsRAN_Project/files/11168194/gnb.log)
[pcap2.zip](https://github.com/srsran/srsRAN_Project/files/11168199/pcap2.zip)
[gnb_rf_b200_tdd_n78_20mhz.zip](https://github.com/srsran/srsRAN_Project/files/11168219/gnb_rf_b200_tdd_n78_20mhz.zip)

closed 2023-04-13 by pricalinat