srsran/srsRAN_Project #49: Low data transfer bandwith

alvaroalfaro612, 2023-04-11:
## Issue Description ##
I have low values (specially in upload speed): DL: 15.6Mb/s and UL: 0.56 Mb/s.
In the trace of the gnb I can see most of the packets are using MCS 28, but in the mobile phone using NSG, i see 64QAM modulation is used (shouldnt it be 256?).

## Setup Details ##
Xiaomi Redmi Note 10 5G, Open5Gs, i7 vPro CPU, band n7, b200 mini.
bw=20 MHz, dl_arfcn=632628 (n78), dl_freq=3489.42 MHz, dl_ssb_arfcn=632544, ul_freq=3489.42 MHz

## Expected Behavior ##
  Higher speeds.

## Actual Behaviour ##
Low speeds.

## Additional Information ##
What parameters can be modified for improving this bandwith? I have seen in documentation that 140 Mbps for DL and 120 Mbps have been achieved... 

Thanks in advance

ismagom, 2023-04-11:
Can you attach logs in info please?

alvaroalfaro612, 2023-04-11:
> Can you attach logs in info please?

Sure, I attach here the gnb.log file with log with info level:
[gnb.zip](https://github.com/srsran/srsRAN_Project/files/11201240/gnb.zip)

I have some underflow messages in stdout, but I it should not be the problem (less than 5 per minute).

ismagom, 2023-04-11:
What kind of traffic are you using? Have you tried iperf UDP?

alvaroalfaro612, 2023-04-12:
> What kind of traffic are you using? Have you tried iperf UDP?

Hello,
when iperf3 server is located on UE, the percentage of packet loss is quite low:

[image]

But in the other case, with the iperf3 server located on the RAN computer, this value is much higher:
[image]

Is this related to the rx/tx gains? How can I improve the uplink connection by setting these values?

ryanwwest, 2023-04-12:
Doesn't your second screenshot have a ~40% _lower_ bitrate than the first? Depending on your iperf setup that could be measuring UE Uplink which would feasibly be lower than UE Downlink assuming srsRAN assigns less Physical Resource Blocks (PRBs) to UL than DL.

alvaroalfaro612, 2023-04-13:
> Doesn't your second screenshot have a ~40% _lower_ bitrate than the first? Depending on your iperf setup that could be measuring UE Uplink which would feasibly be lower than UE Downlink assuming srsRAN assigns less Physical Resource Blocks (PRBs) to UL than DL.

I have looked into the meaning of the outputs of these tests. In both cases the bitrate of the client (who is sending udp packets) is the same. But as it is UDP protocol, the client is not aware of the number of packets that are being received in the other side, so this information should be checked in the server side. 

The conclusion should be that there is a 37% of lost udp packets. What is a reasonable value for this data and how it can be improved?

ismagom, 2023-07-11:
Hi, do you still the same issue with the latest version of the code? Thanks

alvaroalfaro612, 2023-07-13:
No! With the last version it was improved : 30-35 DL and 7-10 UL. Thanks!

closed 2023-07-13 by alvaroalfaro612