srsran/srsRAN_Project #56: When will srsRAN Project support srsUE?

dayal007, 2023-04-13:
## Issue Description ##
[Describe the issue in detail]

srsUE is not working with the new gNB. Do you know when we can anticipate the functionality?

We dont use any radios for our research, so we are kind of blocked without the zmq support in the new gNB. Is there a work around using any other non-radio UE?

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions

- Run srsRAN Project gNB
- Use srsUE from srsRAN_4G

## Expected Behavior ##
[What you expect to happen]
srsUE successfully connect to gNB

## Actual Behaviour ##
[What happens instead e.g. error message]
srsUE doesnt connect to the gNB

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]
Already mentioned above.

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

brendan-mcauliffe, 2023-04-19:
Hi, this functionality will be added to srsUE in the next release of srsRAN 4G, expected by the end of this month. We will also have accompanying documentation in the srsRAN Project docs outlining how to create a set-up using srsUE and the srsRAN Project gNB using ZMQ.

andrepuschmann, 2023-05-05:
We've updated both repos recently and there's also a new appnote describing the required steps in detail https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html

closed 2023-05-05 by andrepuschmann