srsran/srsRAN_Project #57: Error, Band is not Valid: NR Band 46

alex-arknetworx, 2023-04-14:
## Issue Description ##
When configuring a gnb .yml file under the cell_cfg: section with various dl_arfcn values from table 5.4.2.3-1 and 5.4.2.3-2 of 3GPP TS 38.101-1 version 16.5.0 Release 16 for NR Band N46 and then running the gnb with -c argument pointed at the .yml, the following error occurs:

Invalid DL ARFCN=7xxxxx for band 46. Cause: Band is not valid.
srsGNB ERROR: Invalid configuration detected.

Is NR Band 46 (5150 to 5925 MHz) currently supported by srsRAN_Project as Feature List lists "FDD/TDD supported, all FR1 bands"?

Using the supplied example TDD config with various dl_arfcn, this error does not occur (e.g. 620000, 78, 20Mhz, 30kHz).

## Setup Details ##
Open5gs core on external server, x86 machine running Debian 11.6.0 with UHD drivers, B200 and compiled GNB.

## Expected Behavior ##
Proceed to "Making USRP Object" with n46 config .yml, just like the n78 example.

## Actual Behaviour ##
Invalid DL ARFCN=785668 for band 46. Cause: Band is not valid.
srsGNB ERROR: Invalid configuration detected.

## Steps to reproduce the problem ##
cell_cfg:
 dl_arfcn: 785668
 band: 46
 channel_bandwidth_MHz: 20
 common_scs: 15
 plmn: "99970"
 tac: 7
 pci: 1

## Additional Information ##

ismagom, 2023-04-14:
The current release only supports Release 15 bands. We'll add new ones to the next release. Expect it within the next month.

alex-arknetworx, 2023-04-14:
Thanks for the feedback @ismagom 
Also good to know which 3GPP release (15) the current version (23.3) of the project aligns with.
Looking forward to testing a future version on band n46. In Australia there is no shared or SAS/CBRS allocated spectrum compatible with LTE/NR other than the ISM bands, so currently confined to testing with coax and sdr/sdr.

closed 2023-04-14 by alex-arknetworx