srsran/srsRAN_Project #58: Error with SCTP Socket on ARM64 Build

theundyingphoenix, 2023-04-18:
## Question ##
1. Is there a procedure to build srsRAN_Project source code for ARM64?
2. Does the srsRAN_Project's "gnb" use the sctp.ko file instead of the .so files?
    i. What is the cause of this issue?

## Issue Description ##
When running srsRAN_Project "gnb" executable, the following error is produced:
"Failed to create SCTP socket: {}. Hint: Please ensure 'sctp' kernel module is available on the system."

## Setup Details ##
1. Connect Xilinx ARM64 board to laptop (for access to the board via SSH).
4. SSH into the board and run the "gnb" executable with the TDD default configuration in srsRAN_Project's "configs" directory.
5. Error is output.

## Expected Behavior ##
A session starting on the USRP b205-mini and showing the output of a correct setup with the RF device.

## Actual Behaviour ##
"gnb" program outputs the following:
"Failed to create SCTP socket: {}. Hint: Please ensure 'sctp' kernel module is available on the system."

## Steps to reproduce the problem ##
/path/to/srsRAN_Project/apps/gnb/gnb -c /path/to/srsRAN_Project/configs/gnb_rf_b200_tdd_n78_20mhz.yml

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

andrepuschmann, 2023-04-18:
This is a not an srsRAN issue but something that your board support package should cater for. Have you checked the SCTP kernel module is loaded or the module is built into the kernel as the error message indicates?

closed 2023-04-18 by andrepuschmann

theundyingphoenix, 2023-04-18:
I checked and it is not present on my board's kernel. Does this mean my board's support package build needs to build the sctp.ko file and be loaded using "modprobe" or "insmod" prior to running "gnb" executable?

theundyingphoenix, 2023-04-20:
@jao-tm for an ARM64, not yet. However, I'm in the process of using buildroot framework to do so.

jao-tm, 2023-04-21:
OK, I'm trying to build on a Raspberry Pi as I saw on Twitter that @andrepuschmann did that

robertfalkenberg, 2023-04-21:
Don't forget to also install
```
sudo apt-get install linux-modules-extra-raspi
```
to get the SCTP kernel module.

jao-tm, 2023-04-21:
Great, will try that. Thanks!

shirinebadi, 2024-04-18:
hi @theundyingphoenix. I'm also trying to run srsran project on buildroot. Could you make any progress?