srsran/srsRAN_Project #65: srsUE with srsRAn_Project: nrof_harq_processes_for_pdsch not present

asheeshtripathi, 2023-04-27:
## Issue Description ##
Configured srsUE and srsRAN_project gNB as per the [srsRAN gNB with srsUE](https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html) document. UE logs show `nrof_harq_processes_for_pd` not present and UE fails to initiate connect setup procedure. 

**UE console shows RRC connected:**
`Opening 1 channels in RF device=uhd with args=type=x300,serial=323DF3B,clock=external,sampling_rate=23.04e6,lo_freq_offset_hz=23.04e6,None
Supported RF device list: UHD zmq file
[INFO] [UHD] linux; GNU C++ version 9.2.1 20200304; Boost_107100; UHD_3.15.0.0-2build5
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=1, args: type=x300,serial=323DF3B,lo_freq_offset_hz=23.04e6,None=,master_clock_rate=184.32e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 1472 bytes.
[INFO] [X300] Radio 1x clock: 184.32 MHz
[INFO] [GPS] No GPSDO found
[INFO] [0/DmaFIFO_0] Initializing block control (NOC ID: 0xF1F0D00000000000)
[INFO] [0/DmaFIFO_0] BIST passed (Throughput: 1291 MB/s)
[INFO] [0/DmaFIFO_0] BIST passed (Throughput: 1308 MB/s)
[INFO] [0/Radio_0] Initializing block control (NOC ID: 0x12AD100000000001)
[INFO] [0/Radio_1] Initializing block control (NOC ID: 0x12AD100000000001)
[INFO] [0/DDC_0] Initializing block control (NOC ID: 0xDDC0000000000000)
[INFO] [0/DDC_1] Initializing block control (NOC ID: 0xDDC0000000000000)
[INFO] [0/DUC_0] Initializing block control (NOC ID: 0xD0C0000000000000)
[INFO] [0/DUC_1] Initializing block control (NOC ID: 0xD0C0000000000000)
Waiting PHY to initialize ... done!
Attaching UE...
**Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=5614
Random Access Complete.     c-rnti=0x4601, ta=9
RRC Connected**
t
`
**srsUE log:** 
`2023-04-27T15:07:55.182359 [UE     ] [I] Using binary ./srsue with arguments: ue.conf
2023-04-27T15:07:55.213222 [UE     ] [I] Built in Release mode using commit 00c972ac4 on branch master.
2023-04-27T15:07:58.637573 [ASN1   ] [W] [    0] Option nrof_harq_processes_for_pdsch not present
2023-04-27T15:07:58.637573 [RRC-NR ] [W] Failed to make dl_harq_cfg_nr config
2023-04-27T15:07:58.637573 [RRC-NR ] [E] Failed to initiate connection setup procedure
nrof_harq_processes_for_pd
`

**gNB console output and logs:**  
Connecting to AMF on 127.0.0.5:38412
Making USRP object with args 'type=x300,addr=192.168.40.2,master_clock_rate=184.32e6,send_frame_size=8000,recv_frame_size=8000'
[INFO] [UHD] linux; GNU C++ version 11.2.0; Boost_107400; UHD_4.1.0.5-3
[INFO] [LOGGING] Fastpath logging disabled at runtime.
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 8000 bytes.
[INFO] [X300] Radio 1x clock: 184.32 MHz
[WARNING] [0/Radio#0] Attempting to set tick rate to 0. Skipping.
Cell pci=1, bw=10 MHz, dl_arfcn=368500 (n3), dl_freq=1842.5 MHz, dl_ssb_arfcn=368410, ul_freq=1747.5 MHz

==== gNodeB started ===
Type <t> to view trace
t

           -------------DL----------------|------------------UL--------------------
 pci rnti  cqi  mcs  brate   ok  nok  (%) | pusch  mcs  brate   ok  nok  (%)    bsr
   1 4601  n/a    1   6.8k    0    0   0% |   n/a    0      0    0    0   0%    0.0
   1 4601  n/a    0      0    0    0   0% | -21.0    5    20k    0    5   0%    0.0
   1 4601  n/a    0      0    0    0   0% |   n/a    0      0    0    0   0%    0.0
   1 4601  n/a    0      0    0    0   0% |   n/a    0      0    0    0   0%    0.0
   1 4601  n/a    0      0    0    0   0% |   n/a    0      0    0    0   0%    0.0
   1 4601  n/a    0      0    0    0   0% | -20.4    5    20k    0    5   0%    0.0
   1 4601  n/a    0      0    0    0   0% |   n/a    0      0    0    0   0%    0.0
   1 4601  n/a    0      0    0    0   0% |   n/a    0      0    0    0   0%    0.0
   1 4601  n/a    0      0    0    0   0% | -21.2    5    20k    0    5   0%    0.0
   1 4601  n/a    0      0    0    0   0% |   n/a    0      0    0    0   0%    0.0
   1 4601  n/a    0      0    0    0   0% | -22.7    5    41k    0   10   0%    0.0

## Setup Details ##
Intel NUC-1: Open5GS and srsRAN_Project gNB using x310
Intel NUC-2: srsRAN_4G (srsUE)
## Expected Behavior ##
 The srsUE should attach to the network.

## Actual Behaviour ##
With same srsRAN_project gNB and Open5GS core, COTS UE is able to do the attach and proceed upto PDU session response.

## Steps to reproduce the problem ##
srsUE and srsRAN_project gNB as per the [srsRAN gNB with srsUE](https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html) document.

## Additional Information ##
external clock is connected to UE but not gNB. ( However same gNB attach woroks fine with COTS UE)

asheeshtripathi, 2023-04-27:
logs: 
[srsRAN_with_srsUE.zip](https://github.com/srsran/srsRAN_Project/files/11345691/srsRAN_with_srsUE.zip)

brendan-mcauliffe, 2023-05-02:
It looks like you're running an older version of srsUE, can you update to the latest version of srsRAN 4G and try again?

andrepuschmann, 2023-05-02:
We'll release a new version of srsRAN Project this week which will fix the issue you're seeing.

asheeshtripathi, 2023-05-02:
Thanks @andrepuschmann, looking forward to it.

andrepuschmann, 2023-05-05:
We've pushed the code to https://github.com/srsran/srsRAN_Project so please have a look again. There's also a new appnote describing the required steps in detail https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html

Closing the issue. Please feel free to reopen if you still have problems.

closed 2023-05-05 by andrepuschmann

asheeshtripathi, 2023-05-05:
Thanks @andrepuschmann .