srsran/srsRAN_Project #80: Failed to connect SCTP socket to 127.0.1.100:38412. error="Connection refused"

rajugbv, 2023-05-16:
## Issue Description ##
unable establish srsrRAN gNB 

amf:
  addr: 127.0.1.100                 # The address or hostname of the AMF.
  bind_addr: 127.0.0.1              # A local IP that the gNB binds to for traffic from the AMF.
what is addr of amf here and how can i find it in my local computer

## Setup Details ##
os Ubuntu 22.04 ,usrp B200, latest srsran project 5g and uhd software versions installed 

## Expected Behavior ##
Available radio types: uhd.

--== srsRAN gNB (commit 77be7d339) ==--

[INFO] [UHD] linux; GNU C++ version 9.4.0; Boost_107100; UHD_4.2.0.HEAD-0-g197cdc4f
Making USRP object with args 'type=b200'
Cell pci=1, bw=10 MHz, dl_arfcn=632628 (n78), dl_freq=3489.42 MHz, dl_ssb_arfcn=632640, ul_freq=3489.42 MHz

==== gNodeB started ===
Type <t> to view trace

## Actual Behaviour ##
Lower PHY in dual executor mode.
Available radio types: uhd.

--== srsRAN gNB (commit 16cd38851) ==--

Connecting to AMF on 127.0.1.100:38412
Failed to connect SCTP socket to 127.0.1.100:38412. error="Connection refused" timeout=0ms
srsGNB ERROR: Failed to create SCTP gateway.

## Steps to reproduce the problem ##
sudo gnb -c /home/raju/srsRAN_Project/configs/gnb_rf_b200_tdd_n78_20mhz.ym

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

asheeshtripathi, 2023-05-17:
@rajugbv you need to check the config in open5GS conf files and gnb conf file. The IP addresses should match.

brendan-mcauliffe, 2023-05-19:
@rajugbv going through the documentation is a clear start as mentioned above. If you want a detailed tutorial on how to set-up and run the srsRAN Project gNB, I suggest you look at [this tutorial](https://docs.srsran.com/projects/project/en/latest/tutorials/source/cotsUE/source/index.html), take a look at the example configuration files provided for the gNB and Core, and notice how the addresses match up.

brendan-mcauliffe, 2023-06-13:
@rajugbv have you managed to fix this issue?

ismagom, 2023-07-11:
closing due to inactivity

closed 2023-07-11 by ismagom