srsran/srsRAN_Project #85: showing late/underflow/overflow while running gnb

widyaputranti, 2023-05-30:
--== srsRAN gNB (commit fbe73a49c) ==--

Connecting to AMF on 127.0.0.5:38412
[INFO] [UHD] linux; GNU C++ version 9.4.0; Boost_107100; UHD_4.4.0.0-0ubuntu1~focal1
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Making USRP object with args 'type=b200,num_recv_frames=64,num_send_frames=64'
[INFO] [B200] Loading firmware image: /usr/share/uhd/images/usrp_b200_fw.hex...
[INFO] [B200] Detected Device: B210
[INFO] [B200] Loading FPGA image: /usr/share/uhd/images/usrp_b210_fpga.bin...
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Detecting internal GPSDO.... 
[INFO] [GPS] Found an internal GPSDO: GPSTCXO , Firmware Rev 0.929a
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz... 
[INFO] [B200] Actually got clock rate 16.000000 MHz.
[INFO] [MULTI_USRP] Setting master clock rate selection to 'manual'.
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] Actually got clock rate 23.040000 MHz.
Cell pci=1, bw=10 MHz, dl_arfcn=632628 (n78), dl_freq=3489.42 MHz, dl_ssb_arfcn=632640, ul_freq=3489.42 MHz

==== gNodeB started ===
Type <t> to view trace
Late: 0; Underflow: 1; Overflow: 0;
Late: 0; Underflow: 1; Overflow: 0;
Late: 0; Underflow: 1; Overflow: 0;
Late: 0; Underflow: 2; Overflow: 0;
Late: 0; Underflow: 1; Overflow: 0;
Late: 0; Underflow: 1; Overflow: 0;

fyi : im using open5gs as core and Usim from sysmocom and USRP B210

what should i do to fix this problem?

brendan-mcauliffe, 2023-06-01:
Hi, 

If you take a look at the[ troubleshooting section](https://docs.srsran.com/projects/project/en/latest/user_manuals/source/troubleshooting.html) of the documentation you will see details on setting the CPU to performance mode. 

Typically lates and underflows are caused by a lack of available computational resources.

widyaputranti, 2023-06-01:
i did run 4g before, and it ran well and UE (handphone) connected. so that cpu is already in performance mode while running 5g. is there any other possibilities issue?

ismagom, 2023-06-01:
> i did run 4g before, and it ran well and UE (handphone) connected. so that cpu is already in performance mode while running 5g. is there any other possibilities issue?

Hi @widyaputranti, this number of Underflows are not an issue. You don't need to worry unless you see >10 per second

widyaputranti, 2023-06-01:
but unfortunately UE it didnt show connected to UE and UE still cant access internet. still showing late/overflow/underflow while running gnb.

ismagom, 2023-06-01:
That's a different thing then. Can you set log level to info and send us logs please?

widyaputranti, 2023-06-01:
2023-06-01T11:32:40.743517 [FAPI    ] [W] [     0.0] Real-time failure in FAPI: Received late DL_TTI.request. Current slot is 302.17 while message corresponds to 302.14
2023-06-01T11:32:40.743525 [FAPI    ] [W] [     0.0] Real-time failure in FAPI: Received UL_TTI.request message out of time. Current slot is 302.17 while message corresponds to 302.14
2023-06-01T11:32:40.743529 [FAPI    ] [W] [     0.0] Real-time failure in FAPI: Received late DL_TTI.request. Current slot is 302.17 while message corresponds to 302.15
2023-06-01T11:32:40.743536 [FAPI    ] [W] [     0.0] Real-time failure in FAPI: Received UL_TTI.request message out of time. Current slot is 302.17 while message corresponds to 302.15
2023-06-01T11:32:40.743539 [FAPI    ] [W] [     0.0] Real-time failure in FAPI: Received late DL_TTI.request. Current slot is 302.17 while message corresponds to 302.16
2023-06-01T11:32:40.743546 [FAPI    ] [W] [     0.0] Real-time failure in FAPI: Received UL_TTI.request message out of time. Current slot is 302.17 while message corresponds to 302.16
2023-06-01T11:32:46.679417 [FAPI    ] [W] [     0.0] Real-time failure in FAPI: Received late DL_TTI.request. Current slot is 896.9 while message corresponds to 896.5
2023-06-01T11:32:46.679425 [FAPI    ] [W] [     0.0] Real-time failure in FAPI: Received UL_TTI.request message out of time. Current slot is 896.9 while message corresponds to 896.5
2023-06-01T11:32:46.679430 [FAPI    ] [W] [     0.0] Real-time failure in FAPI: Received late DL_TTI.request. Current slot is 896.9 while message corresponds to 896.6
2023-06-01T11:32:46.679435 [FAPI    ] [W] [     0.0] Real-time failure in FAPI: Received UL_TTI.request message out of time. Current slot is 896.9 while message corresponds to 896.6
2023-06-01T11:32:46.679439 [FAPI    ] [W] [     0.0] Real-time failure in FAPI: Received late DL_TTI.request. Current slot is 896.9 while message corresponds to 896.7
2023-06-01T11:32:46.679445 [FAPI    ] [W] [     0.0] Real-time failure in FAPI: Received UL_TTI.request message out of time. Current slot is 896.9 while message corresponds to 896.7
2023-06-01T11:32:46.679449 [FAPI    ] [W] [     0.0] Real-time failure in FAPI: Received late DL_TTI.request. Current slot is 896.9 while message corresponds to 896.8
2023-06-01T11:32:46.679455 [FAPI    ] [W] [     0.0] Real-time failure in FAPI: Received UL_TTI.request message out of time. Current slot is 896.9 while message corresponds to 896.8
2023-06-01T11:32:46.684780 [FAPI    ] [W] [     0.0] Real-time failure in FAPI: Received late DL_TTI.request. Current slot is 897.0 while message corresponds to 896.19
2023-06-01T11:32:46.684795 [FAPI    ] [W] [     0.0] Real-time failure in FAPI: Received UL_TTI.request message out of time. Current slot is 897.0 while message corresponds to 896.19
2023-06-01T11:32:48.855801 [RF      ] [W] Real-time failure in RF: underflow
2023-06-01T11:32:48.860496 [RF      ] [W] Real-time failure in RF: underflow
2023-06-01T11:32:52.695507 [FAPI    ] [W] [     0.0] Real-time failure in FAPI: Received late DL_TTI.request. Current slot is 474.1 while message corresponds to 473.17
2023-06-01T11:32:52.695515 [FAPI    ] [W] [     0.0] Real-time failure in FAPI: Received UL_TTI.request message out of time. Current slot is 474.1 while message corresponds to 473.17
2023-06-01T11:32:52.695525 [FAPI    ] [W] [     0.0] Real-time failure in FAPI: Received late DL_TTI.request. Current slot is 474.1 while message corresponds to 473.18
2023-06-01T11:32:52.695540 [FAPI    ] [W] [     0.0] Real-time failure in FAPI: Received UL_TTI.request message out of time. Current slot is 474.1 while message corresponds to 473.18
2023-06-01T11:32:52.695546 [FAPI    ] [W] [     0.0] Real-time failure in FAPI: Received late DL_TTI.request. Current slot is 474.1 while message corresponds to 473.19
2023-06-01T11:32:52.695551 [FAPI    ] [W] [     0.0] Real-time failure in FAPI: Received UL_TTI.request message out of time. Current slot is 474.1 while message corresponds to 473.19

ismagom, 2023-06-01:
Can you enable logs in info please? `logs --all_level info` and also send the file please, don't cut and paste here. Thanks

widyaputranti, 2023-06-01:
Sure, I attach here the gnb.log file with log with info level: 
[gnb log](https://github.com/srsran/srsRAN_Project/files/11624772/gnb.log)

ismagom, 2023-06-01:
Thanks @widyaputranti . Certainly there are a few lates, but nothing you need to worry about. If you want to optimize your system, read [this section](https://docs.srsran.com/projects/project/en/latest/user_manuals/source/troubleshooting.html#performance-configuration-script) of the manual. It could improve a bit.

closed 2023-06-01 by ismagom