srsran/srsRAN_Project #93: Cannot connect srsUE to gNB through ZMQ. Gets stuck on RRC Connected.

fabio-patrizi, 2023-06-08:
### Discussed in https://github.com/srsran/srsRAN_Project/discussions/92

<div type='discussions-op-text'>

<sup>Originally posted by **fabio-patrizi** June  7, 2023</sup>
Hi all,
I'm having troubles in connecting the srsUE to the gNB via ZMQ. 

I've followed the steps in the  [tutorial](https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html#srsran-gnb-with-srsue). I have built open5gs, srsRAN_Project and srsRAN_4G from sources on Ubuntu 22.04.2 LTS (actually, I'm on a Mac M1 Max).  Each of them seems to work fine and apparently the gNB and open5gs connect with each other. However, when the srsUE seems to fail in connecting. Details follow.

gNB  commit is ce86e4770
srsUE commit is 921f17484

(Expected behavior)
As per the tutorial, the srsUE output should end by something as follows:
...
RRC Connected
PDU Session Establishment successful. IP: 10.45.10.2
RRC NR reconfiguration successful.

(Observed behavior)
Instead, I get the following:

Active RF plugins: libsrsran_rf_uhd.so libsrsran_rf_zmq.so
Inactive RF plugins: 
Reading configuration file ./srsRAN_4G/build/ue_zmq.conf...

Built in Release mode using commit 921f17484 on branch master.

Opening 1 channels in RF device=zmq with args=tx_port=tcp://127.0.0.1:2001,rx_port=tcp://127.0.0.1:2000,base_srate=11.52e6
Supported RF device list: UHD zmq file
CHx base_srate=11.52e6
Current sample rate is 1.92 MHz with a base rate of 11.52 MHz (x6 decimation)
CH0 rx_port=tcp://127.0.0.1:2000
CH0 tx_port=tcp://127.0.0.1:2001
Current sample rate is 11.52 MHz with a base rate of 11.52 MHz (x1 decimation)
Current sample rate is 11.52 MHz with a base rate of 11.52 MHz (x1 decimation)
Waiting PHY to initialize ... done!
Attaching UE...
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=334
Random Access Complete.     c-rnti=0x4601, ta=0
RRC Connected

In particular, I do not get the lines: 

PDU Session Establishment successful. IP: 10.45.10.2
RRC NR reconfiguration successful.

which indicate successful connection (according to the tutorial). 

When I test the connection with ping, it doesn't work.

gNB output is: 
Lower PHY in executor blocking mode.

--== srsRAN gNB (commit ce86e4770) ==--

Connecting to AMF on 127.0.0.5:38412
Available radio types: uhd and zmq.
Cell pci=1, bw=10 MHz, dl_arfcn=368500 (n3), dl_freq=1842.5 MHz, dl_ssb_arfcn=368410, ul_freq=1747.5 MHz

==== gNodeB started ===
Type <t> to view trace

Configuration files are attached:
[gnb_zmq.txt](https://github.com/srsran/srsRAN_Project/files/11680309/gnb_zmq.txt)
[ue_zmq.conf.txt](https://github.com/srsran/srsRAN_Project/files/11680315/ue_zmq.conf.txt)
[amf.yaml.txt](https://github.com/srsran/srsRAN_Project/files/11682492/amf.yaml.txt)
[upf.yaml.txt](https://github.com/srsran/srsRAN_Project/files/11682493/upf.yaml.txt)

Subscriber configuration is attached below. I tried AMF both 8000 and 9000, none work. BTW, where do I get the correct value?

[image]
[image]
[image]

As an aside, the field rf_driver in gnb_zmq.txt does not seem to be correctly parsed. Indeed I get the following error when running the gNB:
INI was not able to parse rf_driver.++
Run with --help for more information.

For this reason, after inspecting the config files provided with the release, I switched to ru_sdr. If I do so, it works fine. This is commented inline in the gNB configuration file.

Also, using "ss_type: common" in gnb_zmq.txt does not work and produces the following error:
INI was not able to parse cell_cfg.ss_type
Run with --help for more information.

To reproduce the issue, just run the open5gs, followed by the gNB, then by the srsUE. If needed, I can provide the specific command lines used, as well as the logs. 

Thank you in advance for any help you can provide.
Best! 

</div>

brendan-mcauliffe, 2023-06-08:
Hi @fabio-patrizi, 

With the latest version of srsRAN Project, we have made some updates to the gNB configuration. 

When configuring the gNB to connect to srsUE you now need to correctly set the `dci_format_0_1_and_1_1` in the `cell_cfg` as well as the `ss_type`. By default non-fallback is used as the DCI format, this needs to be changed so that fallback is used. srsUE requires a common search space and fallback DCI to be used. 

These changes are the same for both ZMQ and RF-based setups. 

You can find updated config files reflecting these changes [here](https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html).

brendan-mcauliffe, 2023-06-09:
Can you send the pcap files for the UE and the gNB NGAP?

fabio-patrizi, 2023-06-09:
I guess this is the gNB NGAP pcap file:
[gnb_ngap.zip](https://github.com/srsran/srsRAN_Project/files/11705172/gnb_ngap.zip)

I cannot produce the UE's. This is the configuration I currently have for UE's pcap section:

[pcap]
enable = none
mac_filename = /tmp/ue_mac.pcap
mac_nr_filename = /tmp/ue_mac_nr.pcap
nas_filename = /tmp/ue_nas.pcap 

I also tried enable = true and enable = yes but /tmp does not contain any of the files above.

brendan-mcauliffe, 2023-06-09:
It looks like the gNB and core are successfully creating the NG connection. So it is probably down how the UE is registered in the core. I will need the UE pcaps to confirm this. 

For the UE you can enable the pcaps on a per-layer basis. For example you can enable all pcaps using `enable = mac, mac_nr, nas`. 

Can you enable these and send the relevant pcap files.

fabio-patrizi, 2023-06-09:
Here they are: 
[ue_pcap_files.zip](https://github.com/srsran/srsRAN_Project/files/11705403/ue_pcap_files.zip)

brendan-mcauliffe, 2023-06-09:
All of those files are empty, can you re-do the tests and make sure the pcaps get populated correctly.

fabio-patrizi, 2023-06-09:
Uh, sorry, I didn't stop the UE and it didn't write the files. Now they are updated.

brendan-mcauliffe, 2023-06-09:
With the latest version of the public code, there have been some changes to the DMRS format of the PUSCH. This is causing the RRC Setup to fail. We have highlighted the issue and are working on fix internally that should be available soon. 

For now, you should revert to the last tagged release of srsRAN Project, version 23.5. You can find it [here](https://github.com/srsran/srsRAN_Project/releases/tag/release_23_5). You will also need to revert to the original configuration file. This means you should use `rf_driver` instead of `ru_sdr`, and remove the `dci_format_0_1_and_1_1` from the `cell_cfg` option.

fabio-patrizi, 2023-06-09:
Tried also with 23.5 but still the same:

Active RF plugins: libsrsran_rf_uhd.so libsrsran_rf_zmq.so
Inactive RF plugins: 
Reading configuration file ./srsRAN_4G/build/ue_zmq.conf...

Built in Release mode using commit 921f17484 on branch master.

Opening 1 channels in RF device=zmq with args=tx_port=tcp://127.0.0.1:2001,rx_port=tcp://127.0.0.1:2000,base_srate=11.52e6
Supported RF device list: UHD zmq file
CHx base_srate=11.52e6
Current sample rate is 1.92 MHz with a base rate of 11.52 MHz (x6 decimation)
CH0 rx_port=tcp://127.0.0.1:2000
CH0 tx_port=tcp://127.0.0.1:2001
Current sample rate is 11.52 MHz with a base rate of 11.52 MHz (x1 decimation)
Current sample rate is 11.52 MHz with a base rate of 11.52 MHz (x1 decimation)
Waiting PHY to initialize ... done!
Attaching UE...
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=334
Random Access Complete.     c-rnti=0x4601, ta=0
RRC Connected

I attach all configuration and pcap files below. Do you confirm configuration files are OK? 

[amf.yaml.txt](https://github.com/srsran/srsRAN_Project/files/11709334/amf.yaml.txt)
[upf.yaml.txt](https://github.com/srsran/srsRAN_Project/files/11709335/upf.yaml.txt)
[gnb_zmq.yaml.txt](https://github.com/srsran/srsRAN_Project/files/11709282/gnb_zmq.yaml.txt)
[ue_zmq.conf.txt](https://github.com/srsran/srsRAN_Project/files/11709293/ue_zmq.conf.txt)

[pcap_files.zip](https://github.com/srsran/srsRAN_Project/files/11709322/pcap_files.zip)

Thanks

brendan-mcauliffe, 2023-06-12:
@fabio-patrizi using the latest version of srsUE and srsRAN Project v23.5 should give you a stable connection. I can see in the attached ue_mac_nr pcap that the cause of the RRC connection not succeeding is due to a known issue with the latest version of the srsRAN Project code.

fabio-patrizi, 2023-06-12:
@brendan-mcauliffe I tried once again, and I confirm it works. Probably, I inadvertently launched the latest gnb instead of v. 23.5. For the time being, I think I can work with this. Thanks a lot for the support!