srsran/srsRAN_Project #94: srsRAN gNB with srsUE - UE stability issues

alejandroBlancoPizarro, 2023-06-12:
## Issue Description ##
I'm following the recent tutorial 'srsRAN gNB with srsUE' from here https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html. The UE attached and got and IP from the 5G core. In addition, ping from UE to gw was working fine. However, we observed some stability issues on the UE side, specifically, the UE lost connection and tried the random access several times. 
## Setup Details ##
2 USRP b210 as RF frontend
Open5gs
srsRAN_Project for gnb
srsRAN_4G for the UE 
Leo Bodnar GPSDO (you recommend it)

## Expected Behavior ##
The UE does not need to repeat the random access several times

## Actual Behaviour ##
The UE losses connection and repeat the random access. See a screenshot of the UE terminal below

[image]
## Steps to reproduce the problem ##
I follow the tutorial

## Additional Information ##
Find below the logs of the UE and gNB.
[ue.log](https://github.com/srsran/srsRAN_Project/files/11721533/ue.log)

[gnb.log](https://github.com/srsran/srsRAN_Project/files/11721530/gnb.log)

andrepuschmann, 2023-06-12:
Hey, the UE has an underflow, loses sync and fails to recover from it. It should be able to recover from that but somehow the RF is in a weird state after losing samples.

I see you're using a recent UHD version on the UE - could you perhaps try to revert back to UHD 3.15 (only UE) and see if the behavior after loosing samples is better? I spot this in the logs `2023-05-31T14:39:07.201070 [RF     ] [I] Detected RF gap of -10000.0 us. Sending end-of-burst.` and I know that EOB handling has changed from version to version a bit. 

See logs:
```
2023-05-31T14:39:07.145821 [PHY0-NR] [I] [ 4795] PUCCH: cc=0, f=2 prb=2:1 symb=12:2 rnti=0x4601 cqi=15 sr=0 , tti_tx=4799
2023-05-31T14:39:07.190769 [PHY-SA ] [E] [ 4840] SYNC: detected out-of-sync... skipping slot ...
2023-05-31T14:39:07.191766 [PHY-SA ] [E] [ 4840] SYNC: detected out-of-sync... skipping slot ...
2023-05-31T14:39:07.192898 [PHY-SA ] [E] [ 4840] SYNC: detected out-of-sync... skipping slot ...
2023-05-31T14:39:07.193731 [PHY-SA ] [E] [ 4840] SYNC: detected out-of-sync... skipping slot ...
2023-05-31T14:39:07.194823 [PHY-SA ] [E] [ 4840] SYNC: detected out-of-sync... skipping slot ...
2023-05-31T14:39:07.195839 [PHY-SA ] [E] [ 4840] SYNC: detected out-of-sync... skipping slot ...
2023-05-31T14:39:07.196780 [PHY-SA ] [E] [ 4840] SYNC: detected out-of-sync... skipping slot ...
2023-05-31T14:39:07.197831 [PHY-SA ] [E] [ 4840] SYNC: detected out-of-sync... skipping slot ...
2023-05-31T14:39:07.198886 [PHY-SA ] [E] [ 4840] SYNC: detected out-of-sync... skipping slot ...
2023-05-31T14:39:07.199782 [PHY-SA ] [E] [ 4840] SYNC: detected out-of-sync... skipping slot ...
2023-05-31T14:39:07.200897 [PHY0-NR] [E] [ 4840] PBCH-MIB: NR SFN (485) does not match current SFN (484)
2023-05-31T14:39:07.201070 [RF     ] [I] Detected RF gap of -10000.0 us. Sending end-of-burst.
2023-05-31T14:39:07.210685 [PHY0-NR] [E] [ 4850] PBCH-MIB: NR SFN (486) does not match current SFN (485)
2023-05-31T14:39:07.220780 [PHY0-NR] [E] [ 4860] PBCH-MIB: NR SFN (487) does not match current SFN (486)
2023-05-31T14:39:07.230687 [PHY0-NR] [E] [ 4870] PBCH-MIB: NR SFN (488) does not match current SFN (487)
2023-05-31T14:39:07.235734 [MAC    ] [I] [ 4885] BSR:   triggered_bsr_type=none, LCID QUEUE status: 0: 0 1: 0 
2023-05-31T14:39:07.236094 [PHY1-NR] [I] [ 4875] PUCCH: cc=0, f=2 prb=2:1 symb=12:2 rnti=0x4601 cqi=15 sr=0 , tti_tx=4879
2023-05-31T14:39:07.238780 [RF     ] [I] Underflow
2023-05-31T14:39:07.240709 [PHY-SA ] [E] [ 4880] SYNC: detected out-of-sync... skipping slot ...
2023-05-31T14:39:07.241946 [PHY-SA ] [E] [ 4880] SYNC: detected out-of-sync... skipping slot ...
2023-05-31T14:39:07.242817 [PHY-SA ] [E] [ 4880] SYNC: detected out-of-sync... skipping slot ...
2023-05-31T14:39:07.243900 [PHY-SA ] [E] [ 4880] SYNC: detected out-of-sync... skipping slot ...
2023-05-31T14:39:07.244949 [PHY-SA ] [E] [ 4880] SYNC: detected out-of-sync... skipping slot ...
2023-05-31T14:39:07.245827 [PHY-SA ] [E] [ 4880] SYNC: detected out-of-sync... skipping slot ...
2023-05-31T14:39:07.246890 [PHY-SA ] [E] [ 4880] SYNC: detected out-of-sync... skipping slot ...
2023-05-31T14:39:07.247891 [PHY-SA ] [E] [ 4880] SYNC: detected out-of-sync... skipping slot ...
2023-05-31T14:39:07.248828 [PHY-SA ] [E] [ 4880] SYNC: detected out-of-sync... skipping slot ...
2023-05-31T14:39:07.249931 [PHY-SA ] [E] [ 4880] SYNC: detected out-of-sync... skipping slot ...
2023-05-31T14:39:07.251064 [PHY0-NR] [W] [ 4880] PBCH-MIB: CRC failed
2023-05-31T14:39:07.251263 [RF     ] [I] Detected RF gap of -10045.6 us. Sending end-of-burst.
2023-05-31T14:39:07.260769 [PHY0-NR] [E] [ 4890] PBCH-MIB: NR SFN (491) does not match current SFN (489)
```

alejandroBlancoPizarro, 2023-06-12:
Hi @andrepuschmann 

Many thanks for the fast reply. I will try with the UHD 3.15 and let you know. 

Best,
Alejandro

alejandroBlancoPizarro, 2023-06-14:
Hi @andrepuschmann 

Changing to UHD 3.15 helps a bit but I got the same issue.

I changed the laptops to PCs (more powerful) and I was able to run ping for an hour, this is a big improvement. However, once I run iperf, the UE got underflow. 

It seems that I need better machines. What are the specifications of the PC that you use as UE?

In any case, I am attaching the UE and gnb logs. 

Best,
Alejandro

[gnb.log](https://github.com/srsran/srsRAN_Project/files/11747926/gnb.log)
[ue.log](https://github.com/srsran/srsRAN_Project/files/11747928/ue.log)

[image]

andrepuschmann, 2023-06-15:
Hey - I'd advise to run the performance script [srsran_performance](https://github.com/srsran/srsRAN_Project/blob/main/scripts/srsran_performance) as you have warnings in the gNB logs. But do the same on the UE as well to make sure the CPU isn't scaling down the frequency. 

In total I don;t think there is an issue with srsRAN Project or srsUE - it's more that you setup isn't tuned and configured correctly. I am closing the issue therefore.

closed 2023-06-15 by andrepuschmann

andrepuschmann, 2023-06-15:
Here is some more infos about performance tuning https://docs.srsran.com/projects/project/en/latest/user_manuals/source/troubleshooting.html?highlight=performance#performance-tuning

matth3wh, 2025-01-30:
Hey @alejandroBlancoPizarro, did you ever get iperf working? I am running into a similar issue, but with srsRAN 4G. I am running with 30 MHz (carrier aggregation of 2 75 PRB cells) and as soon as I start iperf, I get an underflow and then the link just dies. Once I stop the iperf on the UE, the UE then does the reconnect process.