srsran/srsRAN_Project #97: Cannot build tests

polhenarejos, 2023-06-13:
## Issue Description ##
[Describe the issue in detail]
Cannot build tests. It happens with commit 56a771d but also with previous.

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions
Just clone and build
```
git clone https://github.com/srsran/srsRAN_Project
mkdir build && cd build
cmake ..
make
```

It fails with the following log:
```

[ 52%] Building CXX object tests/unittests/du_high/CMakeFiles/paging_du_high_test.dir/test_utils/du_high_test_bench.cpp.o
In file included from /home/phenarejos/srsRAN_Project/tests/unittests/du_high/test_utils/du_high_test_bench.h:25,
                 from /home/phenarejos/srsRAN_Project/tests/unittests/du_high/test_utils/du_high_test_bench.cpp:23:
/home/phenarejos/srsRAN_Project/tests/unittests/du_high/test_utils/du_high_worker_manager.h: In constructor 'constexpr srsran::du_high_worker_manager::du_high_worker_manager()':
/home/phenarejos/srsRAN_Project/tests/unittests/du_high/test_utils/du_high_worker_manager.h:32:8: error: use of deleted function 'srsran::task_worker::task_worker(srsran::task_worker&&)'
   32 | struct du_high_worker_manager {
      |        ^~~~~~~~~~~~~~~~~~~~~~
In file included from /home/phenarejos/srsRAN_Project/tests/unittests/du_high/test_utils/du_high_worker_manager.h:28,
                 from /home/phenarejos/srsRAN_Project/tests/unittests/du_high/test_utils/du_high_test_bench.h:25,
                 from /home/phenarejos/srsRAN_Project/tests/unittests/du_high/test_utils/du_high_test_bench.cpp:23:
/home/phenarejos/srsRAN_Project/include/srsran/support/executors/task_worker.h:57:3: note: declared here
   57 |   task_worker(task_worker&&)                 = delete;
      |   ^~~~~~~~~~~
In file included from /home/phenarejos/srsRAN_Project/tests/unittests/du_high/test_utils/du_high_test_bench.h:25,
                 from /home/phenarejos/srsRAN_Project/tests/unittests/du_high/test_utils/du_high_test_bench.cpp:23:
/home/phenarejos/srsRAN_Project/tests/unittests/du_high/test_utils/du_high_worker_manager.h:32:8: error: use of deleted function 'srsran::task_worker::task_worker(srsran::task_worker&&)'
   32 | struct du_high_worker_manager {
      |        ^~~~~~~~~~~~~~~~~~~~~~
In file included from /home/phenarejos/srsRAN_Project/tests/unittests/du_high/test_utils/du_high_worker_manager.h:28,
                 from /home/phenarejos/srsRAN_Project/tests/unittests/du_high/test_utils/du_high_test_bench.h:25,
                 from /home/phenarejos/srsRAN_Project/tests/unittests/du_high/test_utils/du_high_test_bench.cpp:23:
/home/phenarejos/srsRAN_Project/include/srsran/support/executors/task_worker.h:57:3: note: declared here
   57 |   task_worker(task_worker&&)                 = delete;
      |   ^~~~~~~~~~~
/home/phenarejos/srsRAN_Project/tests/unittests/du_high/test_utils/du_high_test_bench.cpp: In constructor 'srsran::srs_du::du_high_test_bench::du_high_test_bench()':
/home/phenarejos/srsRAN_Project/tests/unittests/du_high/test_utils/du_high_test_bench.cpp:176:58: note: synthesized method 'constexpr srsran::du_high_worker_manager::du_high_worker_manager()' first required here
  176 |   next_slot(0, test_rgen::uniform_int<unsigned>(0, 10239))
      |                                                          ^
make[2]: *** [tests/unittests/du_high/CMakeFiles/paging_du_high_test.dir/build.make:63: tests/unittests/du_high/CMakeFiles/paging_du_high_test.dir/test_utils/du_high_test_bench.cpp.o] Error 1
make[2]: Leaving directory '/home/phenarejos/srsRAN_Project/build'
make[1]: *** [CMakeFiles/Makefile2:8948: tests/unittests/du_high/CMakeFiles/paging_du_high_test.dir/all] Error 2
make[1]: Leaving directory '/home/phenarejos/srsRAN_Project/build'
make: *** [Makefile:141: all] Error 2
```

Of course, if I disable test build it builds successfully:
```
cmake .. -DBUILD_TESTS=0
```

## Expected Behavior ##
[What you expect to happen]
Build should success without errors with tests.

## Actual Behaviour ##
[What happens instead e.g. error message]
It cannot build all tests. Tests `du_high_test`, `paging_du_high_test` and `cu_du_test` fail with the same error.

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]
```
git clone https://github.com/srsran/srsRAN_Project
mkdir build && cd build
cmake ..
make
```

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]
- Debian 11.7 
- gcc 10.2.1-6

Related with #12 

As highlighted, copy constructor of `task_worker` is deleted, but struct in `du_high_worker_manager.h` initializes `task_worker` members with constant objects (which use copy constructor in turn). I do not know why the copy constructor is deleted and if is safe to put it back. Cannot figure out the consequences of putting back.

ismagom, 2023-06-21:
Hi @polhenarejos, is this issue still happening with the latest release?

polhenarejos, 2023-06-21:
Yes, same errors.

ismagom, 2023-07-25:
Hi @polhenarejos, can you check again with the latest release when you got a chance please?

polhenarejos, 2023-08-14:
Hi @ismagom now it works oob. Thanks!

closed 2023-08-14 by ismagom