srsran/srsRAN_Project #99: gNb gets down on reconnection

alvaroalfaro612, 2023-06-14:
## Issue Description ##
When the UE is connected and the data is going without problem through ogstun interface, if I turn off the mobile data in UE, and turn on again, the gNb  gets down with the following message:

> --== srsRAN gNB (commit 56a771dfe) ==--
> 
> Connecting to AMF on 127.0.0.5:38412
> Available radio types: uhd.
> [INFO] [UHD] linux; GNU C++ version 9.4.0; Boost_107100; UHD_4.4.0.HEAD-0-g5fac246b
> [INFO] [LOGGING] Fastpath logging disabled at runtime.
> Making USRP object with args 'type=b200'
> [INFO] [B200] Detected Device: B200mini
> [INFO] [B200] Operating over USB 3.
> [INFO] [B200] Initialize CODEC control...
> [INFO] [B200] Initialize Radio control...
> [INFO] [B200] Performing register loopback test... 
> [INFO] [B200] Register loopback test passed
> [INFO] [B200] Setting master clock rate selection to 'automatic'.
> [INFO] [B200] Asking for clock rate 16.000000 MHz... 
> [INFO] [B200] Actually got clock rate 16.000000 MHz.
> [INFO] [MULTI_USRP] Setting master clock rate selection to 'manual'.
> [INFO] [B200] Asking for clock rate 38.400000 MHz... 
> [INFO] [B200] Actually got clock rate 38.400000 MHz.
> Cell pci=1, bw=20 MHz, dl_arfcn=632628 (n78), dl_freq=3489.42 MHz, dl_ssb_arfcn=632544, ul_freq=3489.42 MHz
> 
> ==== gNodeB started ===
> Type <t> to view trace
> Late: 0; Underflow: 1; Overflow: 0;
> Late: 0; Underflow: 1; Overflow: 0;
> Late: 0; Underflow: 1; Overflow: 0;
> Late: 0; Underflow: 1; Overflow: 0;
> terminate called after throwing an instance of 'std::out_of_range'
>   what():  map::at
> Aborted

without more details in gnb.log or open5Gs amf.log.

Also, I am having problems getting the data through ogstun interface, I have to restart everything many times until get it working, even though the UE equipment is connected to the network always. I can see it by looking the arrows in the network icon not working and also in wireshark, seeing that no packets are seen in ogstun interface. 

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
USRP B200mini, external GPSDO, Open5gs, xiaomi redmi note 10 5g.

## Expected Behavior ##
Reconnection without problems in the case of activating back the data.}
Data transmission working everytime the UE is connected to the network.

andrepuschmann, 2023-06-15:
Hey @alvaroalfaro612  - Can you please share logs with us then this is happening? What would be even better is if you could run the gnb with `gdb` and send us the output. You can start the gnb with the binary and args you use but just prepend `gbd --args gnb arguments` to it. Then type `catch throw std::out_of_range` in the gdb console and then `run`. Then do the same experiment and send us the output of `bt`.

alvaroalfaro612, 2023-06-26:
Hello, 

sorry for the delay. When trying to run : 

> sudo gdb  gnb  -c ./xiaomi/gnb_b210_20MHz_oneplus_8t.yml 

i am getting the following output and the gnb is not initialized:

```
GNU gdb (Ubuntu 9.2-0ubuntu1~20.04.1) 9.2
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from gnb...
**(No debugging symbols found in gnb)
"/home/claudia-sc06/workarea/srsRAN_Project/./xiaomi/gnb_b210_20MHz_oneplus_8t.yml" is not a core dump: file format not recognized**
warning: File "/home/claudia-sc06/workarea/srsRAN_Project/.gdbinit" auto-loading has been declined by your `auto-load safe-path' set to "$debugdir:$datadir/auto-load".
To enable execution of this file add
	add-auto-load-safe-path /home/claudia-sc06/workarea/srsRAN_Project/.gdbinit
line to your configuration file "/root/.gdbinit".
To completely disable this security protection add
	set auto-load safe-path /
line to your configuration file "/root/.gdbinit".
For more information about this security protection see the
"Auto-loading safe path" section in the GDB manual.  E.g., run from the shell:
	info "(gdb)Auto-loading safe path"
(gdb) catch throw std::out_of_range
```