srsran/srsRAN_Project #101: srsRAN gNB with srsUE failed to assgin UE IP address

zikunliu6, 2023-06-21:
## Issue Description ##
I am trying to do the example in 'srsRAN gNB with srsUE' with zeroMQ and open5GS. UE always get stuck at RRC Connected and fail to get an IP address. PS: gNB can get started correctly, but it shows"lower PHY in executor blocking mode", not sure what this is.

## Setup Details ##
ubuntu 22.04LTS, 
srsue commit version 'af4b9589b', 
srsran gNB commit version '56a771dfe', 
open5GS release-17,
libzmp 4.3.4-2

## Expected Behavior ##
UE finishes the RRC connection successfully and get an IP address so that I can do iperf on it

## Actual Behaviour ##
```
zikun@zikunP1:~/srsRAN_Project/configs$ sudo srsue ./ue_zmq.conf 
[sudo] password for zikun: 
Active RF plugins: libsrsran_rf_zmq.so
Inactive RF plugins: 
Reading configuration file ./ue_zmq.conf...

Built in Release mode using commit af4b9589b on branch master.

Opening 1 channels in RF device=zmq with args=tx_port=tcp://127.0.0.1:2001,rx_port=tcp://127.0.0.1:2000,base_srate=11.52e6
Supported RF device list: zmq file
CHx base_srate=11.52e6
Current sample rate is 1.92 MHz with a base rate of 11.52 MHz (x6 decimation)
CH0 rx_port=tcp://127.0.0.1:2000
CH0 tx_port=tcp://127.0.0.1:2001
Current sample rate is 11.52 MHz with a base rate of 11.52 MHz (x1 decimation)
Current sample rate is 11.52 MHz with a base rate of 11.52 MHz (x1 decimation)
Waiting PHY to initialize ... done!
Attaching UE...
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=334
Random Access Complete.     c-rnti=0x4601, ta=0
RRC Connected
t
Enter t to stop trace.
```

```
zikun@zikunP1:~/srsRAN_Project/configs$ sudo gnb -c ./gnb_zmq.yaml 
[sudo] password for zikun: 
Lower PHY in executor blocking mode.

--== srsRAN gNB (commit 56a771dfe) ==--

Connecting to AMF on 127.0.1.100:38412
Available radio types: zmq.
Cell pci=1, bw=10 MHz, dl_arfcn=368500 (n3), dl_freq=1842.5 MHz, dl_ssb_arfcn=368410, ul_freq=1747.5 MHz

==== gNodeB started ===
Type <t> to view trace
```

## Steps to reproduce the problem ##
Config files and logs are _[here](https://drive.google.com/drive/folders/1knqd85WYBnnwXd_H_zGs-Z7wMCZcBe2u?usp=sharing)_
## Additional Information ##

herlesupreeth, 2023-06-21:
@zikunliu6 please use the latest srsUE code (commit id - fa56836b14dc6ad7ce0c3484a1944ebe2cdbe63b) where this issue has been fixed

zikunliu6, 2023-06-21:
Closing this issue, fixed by using commit 'fa56836b14dc6ad7ce0c3484a1944ebe2cdbe63b'

closed 2023-06-21 by zikunliu6