srsran/srsRAN_Project #104: GNB failed to bind SCTP socket, Cannot assign requested address

IngArcos, 2023-06-26:
Hello everyone!

I'm trying to run the ZQM setup with Open5GS. I used the conf files from the docs, and obtained this:

[image]

Hope someone can help me with this.

gustavobsch, 2023-06-26:
Check the bind IP address you have configured in gnb.yml. 
You need to use an IP that:
1. It's available on the host you are running gnb from
2. It's reachable from the AMF
```
amf:
  addr: a.b.c.d                 # The address or hostname of the AMF. 
  bind_addr:  a.b.c.d        # A local IP that the gNB binds to for traffic from the AMF.  <<<<<<< here
```

IngArcos, 2023-06-27:
I changed the address multiple times, but nothing seems to be working. I used the exact addresses from an example that use usrp instead of ZMQ:

amf:
  addr: 127.0.1.100                                               # The address or hostname of the AMF.
  bind_addr: 127.0.0.1                                            # A local IP that the gNB binds to for traffic from the AMF.

[image]

I don´t know why this work and the others no, but thanks for the advice to change the address :)

NobleK3N8, 2023-06-27:
Hallo again...

Y tested several addresses, and I realized that the GNB app only can bind the socket if its local IP address is in the 127.b.c.d segment.

I don't know what else to do, because when the UE obtains an IP is in the 10.45.0.x segment and I'm unable to test de uplink.

gustavobsch, 2023-06-27:
Are you sure you are binding to an IP available on your system? Generally one has to modify the IP in the default config files.

You should use the IP address associated to your default route 
```
$ ip route
default via a.b.c.d dev enx70b5e804da99 <<<<
$ ip add show dev enx70b5e804da99
4: enx70b5e804da99: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether MAC ff:ff:ff:ff:ff:ff
    inet a.b.c.d scope global dynamic noprefixroute enx70b5e804da99 <<<<<<<<<<<
```

catalin-nechifor, 2023-06-27:
I just had this issue today and I managed to bypass it by setting a route so that the sctp messages were sent through a specific device.

Before setting the route, the sctp messages were sent through a network that had no connectivity with the AMF IP.

ALISROURR, 2023-06-30:
> Check the bind IP address you have configured in gnb.yml. You need to use an IP that:
>
>     1. It's available on the host you are running gnb from
>
>     2. It's reachable from the AMF
>
>
> ```
> amf: [...]

i have the same problem and i configure the same ips but it didnt work.is better to build open5gs from source

closed 2023-07-06 by NobleK3N8