srsran/srsRAN_Project #116: srsUE cannot connect to srsgNB with no attempt in cell searching

lgs96, 2023-07-07:
## Issue Description ##
[Describe the issue in detail]

srsUE cannot connect to srsgNB with no attempt in cell searching. Compared to the LTE cell searching, which keeps trying to connect to LTE BS, srsUE with 5G setting only try cell selection once. 

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions

OS: gNB for ubuntu 22.04, UE for ubuntu 20.04
Hardware: both with i9-10940X CPU
RF: both with USRP2943R (X310)
library: UHD 4.4.0

## Expected Behavior ##
[What you expect to happen]

srsUE should attempt to attach to the srsGNB until it connects to it.

## Actual Behaviour ##
[What happens instead e.g. error message]

srsUE does not attempt to connect to srsGNB.

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]

[ue.log.txt](https://github.com/srsran/srsRAN_Project/files/11982559/ue.log.txt)

[ue.conf.txt](https://github.com/srsran/srsRAN_Project/files/11982563/ue.conf.txt)

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

brendan-mcauliffe, 2023-07-07:
Hi @lgs96 can you send the pcaps for both the gNB and UE?

lgs96, 2023-07-07:
@brendan-mcauliffe Thank you for the reply. However, I'm sorry that the both gNB MAC and UE MAC do not generate pcap file since there is no attempt to connection (e.g., random access). So I attach only ngap pcap file of gNB. The below figure shows that no attempt for connection occurs even though I turn on srsUE about 1 minute ago. 
 
[image]

[gnb_ngap.zip](https://github.com/srsran/srsRAN_Project/files/11983993/gnb_ngap.zip)

ismagom, 2023-07-25:
Can you explain a bit better your setup? What CPUs, USRPs are you using? cables, attenuators, external clocks etc. Then copy and paste the console output of both UE and gNB (do not paste images) and logs of both in info mode. Thanks

lgs96, 2023-07-28:
Hi, I attach settings and info logs. Please hope this can help.

CPU:  i9-10940X CPU
USRP: X310 (USRP-2943R)
cables: vert antenna (distance between two USRP is about 10cm)
attenuators: none
external clocks: none

INFO mode logs of gnb and ue
[gnb.log](https://github.com/srsran/srsRAN_Project/files/12189328/gnb.log)
[ue.log](https://github.com/srsran/srsRAN_Project/files/12189329/ue.log)

Thanks.

ismagom, 2023-11-08:
Hi @lgs96 , sorry for the late follow up here. Can you try disabling the UE metrics and running again? It seems the UE gets stuck at the start of the execution. 

One more question: are you running the UE and gNB in the same machine?

closed 2024-02-29 by pgawlowicz