srsran/srsRAN_Project #123: gnb gets an error while we have 15kHz SCS in band n78

emadrezz, 2023-07-11:
## Issue Description ##
As what we see in the standard 38.101-1 v17.5 - Table 5.3.5-1  and https://www.sharetechnote.com/html/5G/5G_FR_Bandwidth.html#Spectrum_and_Bandwidth the band n78 supports 15KHz subcarrier spacing as well as 30KHz scs in different bandwidths but srsGNB will run perfectly on 30KHz but gets an error with 15KHz. I've tested this exact configuration but on band n3 with 15KHz and it worked fine
screenshot of gnb with band n3 and 15khz scs : 

[image]
screenshot of gnb with band n78 and 30khz scs : 
[image]
screenshot of gnb with band n78 and 15khz scs : 
[image]

gnb log for band 78 and 15 khz scs:
[image]

## Setup Details ##
I'm using usrp x300 with latest version of uhd and ubuntu 22 and my configuration file goes as below:
[image]

## Expected Behavior ##
I supposed that gnb works fine with 15khz scs in band n78 with srate 11.52

## Actual Behaviour ##
I've got the error about SSB SCS must be equal to SCS common jst with 15KHZ scs in band n78

ismagom, 2023-07-11:
Hi, band n78 with 15 KHz requires mixed numerology, since SSB is still 30 KHz. Unfortunately mixed numerology is not yet supported.

closed 2023-07-11 by ismagom

emadrezz, 2023-07-11:
Thanks for quick responce.