srsran/srsRAN_Project #126: Trying to run gNB

freicyvillalva, 2023-07-11:
## Issue Description ##
Hi, I am trying to install srsRAN_Project on a Virtual Machine (Ubuntu).
At first, I followed these instructions:
https://docs.srsran.com/projects/project/en/latest/user_manuals/source/installation.html#manual-installation
Now, I am trying to follow the instructions at https://docs.srsran.com/projects/project/en/latest/user_manuals/source/running.html#manual-running
I have installed the gnb using the command sudo make install.
Although there is written "If the gNB has been installed using sudo make install or installed from packages then you will be able to run the gNB from anywhere on your machine.", I can execute the command "sudo gnb -c gnb_rf_b200_tdd_n78_20mhz.yml" from the configs directory only.
Moreover, when running the gNB it does NOT generate the console output it should generate:
*********************************************************************************************************
Available radio types: uhd.

--== srsRAN gNB (commit 77be7d339) ==--

[INFO] [UHD] linux; GNU C++ version 9.4.0; Boost_107100; UHD_4.2.0.HEAD-0-g197cdc4f
Making USRP object with args 'type=b200'
Cell pci=1, bw=10 MHz, dl_arfcn=632628 (n78), dl_freq=3489.42 MHz, dl_ssb_arfcn=632640, ul_freq=3489.42 MHz

==== gNodeB started ===
Type <t> to view trace
*********************************************************************************************************
What I get is the following:
*********************************************************************************************************
Lower PHY in dual executor mode.
/home/fre/srsRAN_Project/include/srsran/adt/bounded_bitset.h:1062: void srsran::bounded_bitset<N, LowestInfoBitIsMSB>::assert_within_bounds_(size_t, bool) const [with long unsigned int N = 1024; bool LowestInfoBitIsMSB = false; size_t = long unsigned int]: 
Assertion `pos < size() or (not strict and pos == size())' failed - ERROR: index='3' is out-of-bounds for bitset of size='2'.
Stack trace (most recent call last):
#10   Object "", at 0xffffffffffffffff, in 
#9    Object "/usr/local/bin/gnb", at 0x55da2733e334, in _start
#8    Source "../csu/libc-start.c", line 392, in __libc_start_main_impl [0x7f6055c29e3f]
#7    Source "../sysdeps/nptl/libc_start_call_main.h", line 58, in __libc_start_call_main [0x7f6055c29d8f]
#6    Object "/usr/local/bin/gnb", at 0x55da27332443, in main
#5    Object "/usr/local/bin/gnb", at 0x55da273436c4, in (anonymous namespace)::worker_manager::create_executors(bool, srsran::optional<srsran::lower_phy_thread_profile>, unsigned int) [clone .isra.0]
#4    Object "/usr/local/bin/gnb", at 0x55da2736375d, in srsran::os_sched_affinity_bitmask::os_sched_affinity_bitmask(unsigned long)
#3    Object "/usr/local/bin/gnb", at 0x55da273630f5, in srsran::detail::print_and_abort(char const*, int, char const*, char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#2    Source "./stdlib/abort.c", line 79, in abort [0x7f6055c287f2]
#1    Source "../sysdeps/posix/raise.c", line 26, in raise [0x7f6055c42475]
#0  | Source "./nptl/pthread_kill.c", line 89, in __pthread_kill_internal
    | Source "./nptl/pthread_kill.c", line 78, in __pthread_kill_implementation
      Source "./nptl/pthread_kill.c", line 44, in __pthread_kill [0x7f6055c96a7c]
Aborted (Signal sent by tkill() 2065 0)
Aborted
******************************************************************************************************
How can I solve this issue?

ismagom, 2023-07-11:
Thanks for reporting. As indicated [in this Discussion post reply](https://github.com/srsran/srsRAN_Project/discussions/107#discussioncomment-6351505) this is a known issue and will be fixed in the next release. 

In the meantime, a workaround is configuring the VM with at least 4 cores.

ismagom, 2023-07-25:
This issue has been fixed in the past release. Please try again and reopen if still present.

closed 2023-07-25 by ismagom