srsran/srsRAN_Project #128: UE no detect a carrier with bandwidth 10mhz or 15mhz in tdd

EugeneBogush, 2023-07-12:
## Issue Description ##
I using Fibocom 150 for connecting to gnb.
and all work great in tdd mode with bandwidth 20 Mhz:
```
cell_cfg:
  dl_arfcn: 632610            
  band: 78                    
  channel_bandwidth_MHz: 20       
  common_scs: 30                             
  plmn: "46601"                                                   
  tac: 1                                                          
  pci: 13          
```                                               
When i change bandwidth to 10mhz or 15mhz my modem not detect a carrier at all.
```
cell_cfg:
  dl_arfcn: 632610            
  band: 78                    
  channel_bandwidth_MHz: 10       
  common_scs: 30                             
  plmn: "46601"                                                   
  tac: 1                                                          
  pci: 13          
```

andrepuschmann, 2023-07-12:
Share the MAC PCAP when the modules attaches to e.g. 20MHz and we can check the capabilities. Pretty sure it's simply not supporting this bandwidth on the particular band.

EugeneBogush, 2023-07-12:
>Share the MAC PCAP

[gnb_mac.pcap.gz](https://github.com/srsran/srsRAN_Project/files/12028534/gnb_mac.pcap.gz)

but it detect a carrier in fdd mode bandwidth 5 mhz and 10mhz.

andrepuschmann, 2023-07-13:
Each band is different - see here for n78 20 MHz is not supported.

[image]

closed 2023-07-13 by andrepuschmann