srsran/srsRAN_Project #150: srsRAN_Project issue with USRP X410

MaryamAmini, 2023-07-27:
Hello srsRAN team,

I recently updated srsRAN_Project and after the update, srsRAN-gNB doesn't run with USRP X410. I was able to work with USRP X410 and srsRAN_Project before, but using the same configs I'm not able to do so anymore. 
At first, I was getting this error:

_--== srsRAN gNB (commit 583c92b16) ==--

Connecting to AMF on 127.0.0.2:38412
Available radio types: uhd.
[INFO] [UHD] linux; GNU C++ version 9.4.0; Boost_107100; UHD_4.4.0.0-0ubuntu1~focal1
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Making USRP object with args 'type=x4xx,master_clock_rate=491.52e6,send_frame_size=8000,recv_frame_size=8000'
[INFO] [MPMD] Initializing 1 device(s) in parallel with args: mgmt_addr=192.168.10.2,type=x4xx,product=x410,serial=327AFE0,name=ni-x4xx-327AFE0,fpga=X4_200,claimed=False,addr=192.168.10.2,master_clock_rate=491.52e6,send_frame_size=8000,recv_frame_size=8000
[ERROR] [RPC] master_clock_rate 491520000.0 is not compatible with FPGA which expected decimation 2
[ERROR] [MPM.PeriphManager] master_clock_rate 491520000.0 is not compatible with FPGA which expected decimation 2
[ERROR] [MPM.RPCServer] init() failed with error: master_clock_rate 491520000.0 is not compatible with FPGA which expected decimation 2
Failed to open device with address 'type=x4xx': RuntimeError: Error during RPC call to `init'. Error message: master_clock_rate 491520000.0 is not compatible with FPGA which expected decimation 2
srsGNB ERROR: Unable to create radio session.
Aborted (core dumped)_

Then, after adding the **master_clock_rate=250e6** in the configs (I didn't have to do this before), that error disappeared and this one showed up:

_--== srsRAN gNB (commit 583c92b16) ==--

Connecting to AMF on 127.0.0.2:38412
Available radio types: uhd.
[INFO] [UHD] linux; GNU C++ version 9.4.0; Boost_107100; UHD_4.4.0.0-0ubuntu1~focal1
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Making USRP object with args 'type=x4xx,master_clock_rate=250e6,send_frame_size=8000,recv_frame_size=8000'
[INFO] [MPMD] Initializing 1 device(s) in parallel with args: mgmt_addr=192.168.10.2,type=x4xx,product=x410,serial=327AFE0,name=ni-x4xx-327AFE0,fpga=X4_200,claimed=False,addr=192.168.10.2,master_clock_rate=250e6,send_frame_size=8000,recv_frame_size=8000
[INFO] [MPM.PeriphManager] init() called with device args `fpga=X4_200,master_clock_rate=250e6,mgmt_addr=192.168.10.2,name=ni-x4xx-327AFE0,product=x410,recv_frame_size=8000,send_frame_size=8000,clock_source=internal,time_source=internal'.
Error: setting Tx sampling rate. Tx Rate 30.72 MHz is invalid. The nearest valid value is 31.25.
srsGNB ERROR: Unable to create radio session.
Aborted (core dumped)_

Now, I don't understand why the s_rate that was previously supported, is not supported anymore. To be sure, I changed the s_rate to 31.25 as recommended and I got this error:

_The sampling rate must be multiple of 1.92 MHz.
srsGNB ERROR: Invalid configuration detected.

Aborted (core dumped)_

Any idea why this is happening?
I've already checked USRP X410, and UHD with another RAN solution and everything is working as they used to.
Also, srsRAN_Project is working fine with USRP B210.
Attached you can find the configs I'm using.

Thank you,
Maryam.
[gnb_rf_x410.txt](https://github.com/srsran/srsRAN_Project/files/12186414/gnb_rf_x410.txt)

ismagom, 2023-07-27:
Hi Maryam, 

In that device you need to use the master clock rate 491.52 MHz. I'm not sure why is not accepted by the USRP. Are you sure nothing has changed in the FPGA? see: 

```
master_clock_rate 491520000.0 is not compatible with FPGA which expected decimation 2
```

MaryamAmini, 2023-07-27:
Hey Ismael,

I'm pretty sure that nothing has changed on the USRP side. I have just tested it with OAI, and it was working fine!

One more question, I was checking the device arguments of X410 in Ettus (https://kb.ettus.com/USRP_X410_Getting_Started_Guide), and the master_clock_rate is supposed to be master_clock_rate=250e6, right?

ismagom, 2023-07-27:
From that link you shared: 

```
The analog bandwidth determines the available master clock rates. As of UHD 4.1, only the X4_200 image is shipped with UHD, which allows a 245.76 MHz or 250 MHz master clock rate. The other images are considered experimental (unsupported).
```

so we need to use master_clock_rate=245.76e6, can you try passing it as arg?

MaryamAmini, 2023-07-27:
Hi again,

So this master clock rate (245.76 MHz) actually worked :).
Now everything is working. 
Just curious, do you have any idea where "master_clock_rate 491520000.0 is not compatible with FPGA which expected decimation 2", was coming from?

Thanks so much for the help,
Maryam.

xavierarteaga, 2023-07-28:
Hi Maryam, 

First of all, many thanks for your interest in using srsRAN.

> Just curious, do you have any idea where "master_clock_rate 491520000.0 is not compatible with FPGA which expected decimation 2", was coming from?

The software detects the USRP type, and it selects a default value if the `master_clock_rate` is not specified in the `device_args`. 

I selected the highest master clock rate of `491.52e6` from the [x4xx manual page](https://files.ettus.com/manual/page_usrp_x4xx.html) which states:

> The 200 MHz images allow master clock rates of 245.76 MHz or 250 MHz. The 400 MHz images allow master clock rates of 491.52 MHz or 500 MHz.

In any case, the default master clock rate will be soon updated in the master branch to `245.76e6`.

closed 2023-07-28 by xavierarteaga

mianrazo, 2023-12-01:
Hi Maryam, sorry to bother you, but I just get a X410 and starting to explore the options to use it for some testing, I used to have a B210 working with eNB but now I would like to setup the X410 with gNB already, I see you pretty much get it to work already, quick question , may I ask you to check how are you connecting the X410 to your gNB, since B210 was way much easier with USB but 410 have a lot of QSFPs and I'm not sure exactly how to connect it for testing within 100 MHz bandwidth, thanks in advance for your help!

MaryamAmini, 2023-12-04:
> Hi Maryam, sorry to bother you, but I just get a X410 and starting to explore the options to use it for some testing, I used to have a B210 working with eNB but now I would like to setup the X410 [...]

Hello @mianrazo 
If I were to start working with this USRP again, I would probably start with the USRP manual, and make sure that the device is working fine with the UHD installed on the host PC, and then would focus on working with this SDR and srsRAN project.
Please read the instructions to set up the USRP at:
https://kb.ettus.com/USRP_X410/X440_Getting_Started_Guide

and then refer to the configuration files provided in this thread.
If you have more specific questions I'd be happy to help :).
bests,
Maryam.