srsran/srsRAN_Project #156: gNB does not send SIB1

dconnollyapl, 2023-08-03:
## Issue Description ##
The srsRAN_Project does not send SIB1 as part of the resource grid in some 10MHz configurations. 

## Setup Details ##
Radio: Ettus B210
srsRAN_Project tag release_23_5
open5gs Core (this problem should be agnostic to the specific core used)

## Expected Behavior ##
SIB1 should show up in the transmitted resource grid.

## Actual Behaviour ##
SIB1 does not appear in the transmitted resource grid. Only the SSB is present, meaning that UEs cannot connect.

## Steps to reproduce the problem ##
Run the SRS gNB with the configuration below on an Ettus B210 and use the SanJole WaveJudge to view the transmitted resource grid.

Configuration options specific to this setup:

cell_cfg:
   dl_arfcn: 176300
   band: 5
   channel_bandwidth_MHz: 10
   common_scs: 30

## Additional Information ##
Stepping through the srsRAN_Project code, it appears that the allocate_sib1( ) function is called, despite nothing showing up in the transmitted waveform.

The same configuration, modified to use 20MHz channel bandwidth, does produce SIB1 in the TX resource grid.

Attempting to use the same setup on the latest srsRAN_Project code (as of 8/1/23), and modifying the config appropriately for new configuration options, the SRS software produces an error that DMRS Mapping type B is not supported.

frankist, 2023-08-04:
Can you share the gnb mac logs in debug?

dconnollyapl, 2023-08-04:
Are you looking for the `gnb.log` or the mac pcap? In my logs with `broacast_enabled: true`, I see the following despite SIB1 not having actually been transmitted:

`SIB1 PDSCH: prb=[0..8) symb=[3..14) tbs=80 mcs=5 rv=0`

carlo-gal, 2023-08-04:
We'll probably need to check ourselves internally, but for the time being, could you please share the logs you collected?
We would like to see what is going on there.
Thanks!

frankist, 2023-08-09:
We see that the FAPI dected an issue in the SIB1 generated:
```
2023-08-08T12:24:05.835019 [SCHED   ] [D] [    64.1] Slot decisions cell=0 t=261us:
- DL PDCCH: rnti=0xffff type=si-rnti cs_id=0 ss_id=0 format=1_0 cce=0 al=4 
- SIB1 PDSCH: prb=[0..8) symb=[3..14) tbs=80 mcs=5 rv=0
2023-08-08T12:24:05.835051 [FAPI    ] [W] [     0.0] Unsupported PDSCH PDU detected. Skipping DL_TTI.request message in 64.1.
```
We will try to reproduce the problem and fix it.

herlesupreeth, 2023-09-06:
Hey @dconnollyapl, 

Band 5 is a FDD, which will result in SSB having 15Khz SCS. And, when you set `common_scs: 30`, this will result in PDCCH + PDSCH + PUSCH having 30Khz SCS and we dont support different SCS for SSB and others. So you should set `common_scs: 15` in order to see SIB1.

NOTE: Please use the latest latest srsRAN_Project code for testing

herlesupreeth, 2023-09-06:
Hey @dconnollyapl 

> Attempting to use the same setup on the latest srsRAN_Project code (as of 8/1/23), and modifying the config appropriately for new configuration options, the SRS software produces an error that DMRS Mapping type B is not supported.

Since you are setting common_scs: 30, the number of resource blocks in cell BW of 10Mhz are [0..24]. And, in order to be able to fit CORESET#0 and SSB without collision, the gNB is selecting CORESET 0 index as 5 to maximize CORESET#0 duration (from table 13-4 of TS 38.213) resulting in CORESET#0 duration of 3 symbols. Since we dont support DMRS type B for PDSCH, there are no entries in the Table 5.1.2.1.1-2 of TS 38.214 applicable for sending SIB1.

And, you see SIB1 when you increase the BW to 20 Mhz because the nof. resource blocks increases (i.e. [0..51]). This will result in able to select CORESET 0 index >= 10 (from table 13-4 of TS 38.213) where CORESET#0 duration is 2 symbols, which can be supported in DMRS type A.

herlesupreeth, 2023-09-06:
I was slightly wrong above. We are setting DMRS type A position as pos2 rather than pos3 when CORESET#0 duration was 3 OFDM symbols. Will fix this in next release. In the meantime in order to make it work with band 5, I would suggest either to increase the BW to 20 Mhz or set common_scs to 15.

ismagom, 2023-10-26:
Hey @frankist , @herlesupreeth , have this issue been fixed?

herlesupreeth, 2023-12-05:
This is fixed in the latest commits

closed 2023-12-05 by herlesupreeth