srsran/srsRAN_Project #158: Failed to build latest srsRAN_Project on Ubuntu 22.04

s5uishida, 2023-08-07:
## Issue Description ##
An error occurred when building latest srsRAN_Project (main branch) with ZMQ support on Ubuntu 22.04.
BTW, srsRAN Project 23.5 was able to build, but parse errors occurred when loading the `gNB config` linked below.

https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html#id7

## Setup Details ##
The build procedure is as follows.
```
apt update
apt upgrade
reboot
```
```
apt install cmake make gcc g++ pkg-config libfftw3-dev libmbedtls-dev libsctp-dev libyaml-cpp-dev libgtest-dev
apt install libzmq3-dev
```
```
git clone https://github.com/srsran/srsRAN_Project.git
cd srsRAN_Project
mkdir build
cd build
cmake ../ -DENABLE_EXPORT=ON -DENABLE_ZEROMQ=ON
make -j`nproc`
```
The error that occurred is as follows:
```
...
[ 14%] Built target srsran_radio
[ 14%] Building CXX object lib/asn1/CMakeFiles/e1ap_asn1.dir/e1ap/e1ap.cpp.o
In file included from /usr/lib/gcc/x86_64-linux-gnu/11/include/immintrin.h:99,
                 from /root/srsRAN_Project/lib/srsvec/simd.h:41,
                 from /root/srsRAN_Project/lib/srsvec/prod.cpp:26:
/usr/lib/gcc/x86_64-linux-gnu/11/include/fmaintrin.h: In function ‘srsran::simd_cf_t srsran::srsran_simd_cf_prod(srsran::simd_cf_t, srsran::simd_cf_t)’:
/usr/lib/gcc/x86_64-linux-gnu/11/include/fmaintrin.h:63:1: error: inlining failed in call to ‘always_inline’ ‘__m256 _mm256_fmadd_ps(__m256, __m256, __m256)’: target specific option mismatch
   63 | _mm256_fmadd_ps (__m256 __A, __m256 __B, __m256 __C)
      | ^~~~~~~~~~~~~~~
In file included from /root/srsRAN_Project/lib/srsvec/prod.cpp:26:
/root/srsRAN_Project/lib/srsvec/simd.h:881:38: note: called from here
  881 |   ret.im            = _mm256_fmadd_ps(a.re, b.im, _mm256_mul_ps(a.im, b.re));
      |                       ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/lib/gcc/x86_64-linux-gnu/11/include/immintrin.h:99,
                 from /root/srsRAN_Project/lib/srsvec/simd.h:41,
                 from /root/srsRAN_Project/lib/srsvec/prod.cpp:26:
/usr/lib/gcc/x86_64-linux-gnu/11/include/fmaintrin.h:111:1: error: inlining failed in call to ‘always_inline’ ‘__m256 _mm256_fmsub_ps(__m256, __m256, __m256)’: target specific option mismatch
  111 | _mm256_fmsub_ps (__m256 __A, __m256 __B, __m256 __C)
      | ^~~~~~~~~~~~~~~
In file included from /root/srsRAN_Project/lib/srsvec/prod.cpp:26:
/root/srsRAN_Project/lib/srsvec/simd.h:880:38: note: called from here
  880 |   ret.re            = _mm256_fmsub_ps(a.re, b.re, _mm256_mul_ps(a.im, b.im));
      |                       ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/lib/gcc/x86_64-linux-gnu/11/include/immintrin.h:99,
                 from /root/srsRAN_Project/lib/srsvec/simd.h:41,
                 from /root/srsRAN_Project/lib/srsvec/prod.cpp:26:
/usr/lib/gcc/x86_64-linux-gnu/11/include/fmaintrin.h:111:1: error: inlining failed in call to ‘always_inline’ ‘__m256 _mm256_fmsub_ps(__m256, __m256, __m256)’: target specific option mismatch
  111 | _mm256_fmsub_ps (__m256 __A, __m256 __B, __m256 __C)
      | ^~~~~~~~~~~~~~~
In file included from /root/srsRAN_Project/lib/srsvec/prod.cpp:26:
/root/srsRAN_Project/lib/srsvec/simd.h:880:38: note: called from here
  880 |   ret.re            = _mm256_fmsub_ps(a.re, b.re, _mm256_mul_ps(a.im, b.im));
      |                       ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/lib/gcc/x86_64-linux-gnu/11/include/immintrin.h:99,
                 from /root/srsRAN_Project/lib/srsvec/simd.h:41,
                 from /root/srsRAN_Project/lib/srsvec/prod.cpp:26:
/usr/lib/gcc/x86_64-linux-gnu/11/include/fmaintrin.h:63:1: error: inlining failed in call to ‘always_inline’ ‘__m256 _mm256_fmadd_ps(__m256, __m256, __m256)’: target specific option mismatch
   63 | _mm256_fmadd_ps (__m256 __A, __m256 __B, __m256 __C)
      | ^~~~~~~~~~~~~~~
In file included from /root/srsRAN_Project/lib/srsvec/prod.cpp:26:
/root/srsRAN_Project/lib/srsvec/simd.h:881:38: note: called from here
  881 |   ret.im            = _mm256_fmadd_ps(a.re, b.im, _mm256_mul_ps(a.im, b.re));
      |                       ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [lib/srsvec/CMakeFiles/srsvec.dir/build.make:216: lib/srsvec/CMakeFiles/srsvec.dir/prod.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:9326: lib/srsvec/CMakeFiles/srsvec.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 14%] Linking CXX static library libsrsran_pusch_processor.a
[ 14%] Built target srsran_pusch_processor
[ 14%] Linking CXX static library libsrsran_pdcch_processor.a
[ 14%] Built target srsran_pdcch_processor
[ 14%] Linking CXX static library libe1ap_asn1.a
[ 14%] Built target e1ap_asn1
make: *** [Makefile:146: all] Error 2
```

## Expected Behavior ##
I want the build to succeed without errors.

## Actual Behaviour ##

## Steps to reproduce the problem ##

## Additional Information ##
- Ubuntu 22.04 (Linux 5.15.0-78)
- gcc version 11.4.0 (Ubuntu 11.4.0-1ubuntu1~22.04)

herlesupreeth, 2023-08-07:
@xavierarteaga Can you take a look??

@s5uishida in the meantime can you please compile with following additional flag and see if the build passes?

`cmake ../ -DENABLE_EXPORT=ON -DENABLE_ZEROMQ=ON -DENABLE_AVX512=OFF`

s5uishida, 2023-08-07:
Hi @herlesupreeth 

Thank you very much for your reply!

When I tried from `git clone` again, the following error occurred.
```
cmake ../ -DENABLE_EXPORT=ON -DENABLE_ZEROMQ=ON -DENABLE_AVX512=OFF
make -j`nproc`
```
```
[ 10%] Building CXX object lib/srsvec/CMakeFiles/srsvec.dir/sc_prod.cpp.o
In file included from /usr/lib/gcc/x86_64-linux-gnu/11/include/immintrin.h:99,
                 from /root/srsRAN_Project/lib/srsvec/simd.h:41,
                 from /root/srsRAN_Project/lib/srsvec/prod.cpp:26:
/usr/lib/gcc/x86_64-linux-gnu/11/include/fmaintrin.h: In function ‘srsran::simd_cf_t srsran::srsran_simd_cf_prod(srsran::simd_cf_t, srsran::simd_cf_t)’:
/usr/lib/gcc/x86_64-linux-gnu/11/include/fmaintrin.h:63:1: error: inlining failed in call to ‘always_inline’ ‘__m256 _mm256_fmadd_ps(__m256, __m256, __m256)’: target specific option mismatch
   63 | _mm256_fmadd_ps (__m256 __A, __m256 __B, __m256 __C)
      | ^~~~~~~~~~~~~~~
In file included from /root/srsRAN_Project/lib/srsvec/prod.cpp:26:
/root/srsRAN_Project/lib/srsvec/simd.h:881:38: note: called from here
  881 |   ret.im            = _mm256_fmadd_ps(a.re, b.im, _mm256_mul_ps(a.im, b.re));
      |                       ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/lib/gcc/x86_64-linux-gnu/11/include/immintrin.h:99,
                 from /root/srsRAN_Project/lib/srsvec/simd.h:41,
                 from /root/srsRAN_Project/lib/srsvec/prod.cpp:26:
/usr/lib/gcc/x86_64-linux-gnu/11/include/fmaintrin.h:111:1: error: inlining failed in call to ‘always_inline’ ‘__m256 _mm256_fmsub_ps(__m256, __m256, __m256)’: target specific option mismatch
  111 | _mm256_fmsub_ps (__m256 __A, __m256 __B, __m256 __C)
      | ^~~~~~~~~~~~~~~
In file included from /root/srsRAN_Project/lib/srsvec/prod.cpp:26:
/root/srsRAN_Project/lib/srsvec/simd.h:880:38: note: called from here
  880 |   ret.re            = _mm256_fmsub_ps(a.re, b.re, _mm256_mul_ps(a.im, b.im));
      |                       ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/lib/gcc/x86_64-linux-gnu/11/include/immintrin.h:99,
                 from /root/srsRAN_Project/lib/srsvec/simd.h:41,
                 from /root/srsRAN_Project/lib/srsvec/prod.cpp:26:
/usr/lib/gcc/x86_64-linux-gnu/11/include/fmaintrin.h:111:1: error: inlining failed in call to ‘always_inline’ ‘__m256 _mm256_fmsub_ps(__m256, __m256, __m256)’: target specific option mismatch
  111 | _mm256_fmsub_ps (__m256 __A, __m256 __B, __m256 __C)
      | ^~~~~~~~~~~~~~~
In file included from /root/srsRAN_Project/lib/srsvec/prod.cpp:26:
/root/srsRAN_Project/lib/srsvec/simd.h:880:38: note: called from here
  880 |   ret.re            = _mm256_fmsub_ps(a.re, b.re, _mm256_mul_ps(a.im, b.im));
      |                       ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/lib/gcc/x86_64-linux-gnu/11/include/immintrin.h:99,
                 from /root/srsRAN_Project/lib/srsvec/simd.h:41,
                 from /root/srsRAN_Project/lib/srsvec/prod.cpp:26:
/usr/lib/gcc/x86_64-linux-gnu/11/include/fmaintrin.h:63:1: error: inlining failed in call to ‘always_inline’ ‘__m256 _mm256_fmadd_ps(__m256, __m256, __m256)’: target specific option mismatch
   63 | _mm256_fmadd_ps (__m256 __A, __m256 __B, __m256 __C)
      | ^~~~~~~~~~~~~~~
In file included from /root/srsRAN_Project/lib/srsvec/prod.cpp:26:
/root/srsRAN_Project/lib/srsvec/simd.h:881:38: note: called from here
  881 |   ret.im            = _mm256_fmadd_ps(a.re, b.im, _mm256_mul_ps(a.im, b.re));
      |                       ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [lib/srsvec/CMakeFiles/srsvec.dir/build.make:216: lib/srsvec/CMakeFiles/srsvec.dir/prod.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 10%] Building CXX object lib/ran/CMakeFiles/srsran_ran.dir/pdcch/pdcch_candidates.cpp.o
[ 11%] Building CXX object lib/fapi/pdu_validators/CMakeFiles/srsran_fapi_pdu_validators.dir/ul_pusch_pdu.cpp.o
[ 11%] Building CXX object lib/ran/CMakeFiles/srsran_ran.dir/pdcch/pdcch_type0_css_coreset_config.cpp.o
In file included from /usr/lib/gcc/x86_64-linux-gnu/11/include/immintrin.h:99,
                 from /root/srsRAN_Project/lib/srsvec/simd.h:41,
                 from /root/srsRAN_Project/lib/srsvec/sc_prod.cpp:25:
/usr/lib/gcc/x86_64-linux-gnu/11/include/fmaintrin.h: In function ‘srsran::simd_cf_t srsran::srsran_simd_cf_prod(srsran::simd_cf_t, srsran::simd_cf_t)’:
/usr/lib/gcc/x86_64-linux-gnu/11/include/fmaintrin.h:63:1: error: inlining failed in call to ‘always_inline’ ‘__m256 _mm256_fmadd_ps(__m256, __m256, __m256)’: target specific option mismatch
   63 | _mm256_fmadd_ps (__m256 __A, __m256 __B, __m256 __C)
      | ^~~~~~~~~~~~~~~
In file included from /root/srsRAN_Project/lib/srsvec/sc_prod.cpp:25:
/root/srsRAN_Project/lib/srsvec/simd.h:881:38: note: called from here
  881 |   ret.im            = _mm256_fmadd_ps(a.re, b.im, _mm256_mul_ps(a.im, b.re));
      |                       ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/lib/gcc/x86_64-linux-gnu/11/include/immintrin.h:99,
                 from /root/srsRAN_Project/lib/srsvec/simd.h:41,
[... 32 lines cut ...]
[ 11%] Building CXX object lib/ran/CMakeFiles/srsran_ran.dir/pdsch/dlsch_info.cpp.o
[ 12%] Building CXX object lib/ran/CMakeFiles/srsran_ran.dir/pdsch/pdsch_mcs.cpp.o
[ 12%] Building CXX object lib/ran/CMakeFiles/srsran_ran.dir/pdsch/pdsch_antenna_port_mapping.cpp.o
[... cut ...]
[ 12%] Building CXX object lib/ran/CMakeFiles/srsran_ran.dir/sch/sch_segmentation.cpp.o
[ 12%] Building CXX object lib/ran/CMakeFiles/srsran_ran.dir/ssb_freq_position_generator.cpp.o
[ 13%] Building CXX object lib/ran/CMakeFiles/srsran_ran.dir/ssb_gscn.cpp.o
[ 13%] Building CXX object lib/ran/CMakeFiles/srsran_ran.dir/ssb_mapping.cpp.o
[ 13%] Building CXX object lib/ran/CMakeFiles/srsran_ran.dir/tdd_ul_dl_config.cpp.o
[ 13%] Linking CXX static library libsrsran_ran.a
[ 13%] Built target srsran_ran
make: *** [Makefile:146: all] Error 2
```

herlesupreeth, 2023-08-07:
hmm..can you give us more information about your build environment?? What CPU you are running this one? Is it a VM or a host machine?? etc

s5uishida, 2023-08-07:
[Host OS]
- CPU: Intel Core-i5 (6 cores)
- Mem: 32GB
- Ubuntu 20.04

[VirtualBox Guest OS] <-- build OS
- CPU: 4 cores
- Mem: 4GB
- Ubuntu 22.04

herlesupreeth, 2023-08-07:
Ah, most likely the VM is not detecting the CPU instruction set correctly.

Can you try building with following cmake flag??

`cmake ../ -DENABLE_EXPORT=ON -DENABLE_ZEROMQ=ON -DAUTO_DETECT_ISA=OFF`

s5uishida, 2023-08-07:
Hi @herlesupreeth 

Thank you very much!

The cmake log is as follows. The build process seems fine in progress.
```
# cmake ../ -DENABLE_EXPORT=ON -DENABLE_ZEROMQ=ON -DAUTO_DETECT_ISA=OFF
-- The C compiler identification is GNU 11.4.0
-- The CXX compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build type not specified: defaulting to Release.
-- Performing Test HAS_MAYBE_UNINITIALIZED
-- Performing Test HAS_MAYBE_UNINITIALIZED - Success
-- Performing Test HAVE_NON_VIRTUAL_DTOR
-- Performing Test HAVE_NON_VIRTUAL_DTOR - Success
-- Performing Test HAVE_SUGGEST_OVERRIDE
-- Performing Test HAVE_SUGGEST_OVERRIDE - Success
-- Performing Test HAVE_SHADOW
-- Performing Test HAVE_SHADOW - Success
-- Assertion level set to NORMAL
-- Could NOT find libdw (missing: LIBDW_LIBRARY LIBDW_INCLUDE_DIR) 
-- Could NOT find libbfd (missing: LIBBFD_LIBRARY LIBBFD_INCLUDE_DIR) 
-- Could NOT find libdwarf (missing: LIBDWARF_LIBRARY LIBDWARF_INCLUDE_DIR LIBELF_LIBRARY LIBELF_INCLUDE_DIR) 
-- Found Backward: /root/srsRAN_Project/cmake/modules  
-- Backward-cpp found, but external libraries are missing.
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2") 
-- Checking for module 'mbedtls'
--   No package 'mbedtls' found
-- MBEDTLS LIBRARIES: /usr/lib/x86_64-linux-gnu/libmbedcrypto.so
-- MBEDTLS STATIC LIBRARIES: /usr/lib/x86_64-linux-gnu/libmbedcrypto.a
-- MBEDTLS INCLUDE DIRS: /usr/include
-- Found MbedTLS: /usr/lib/x86_64-linux-gnu/libmbedcrypto.so  
-- Checking for module 'fftw3f >= 3.0'
--   Found fftw3f , version 3.3.8
-- FFTW3F LIBRARIES: /usr/lib/x86_64-linux-gnu/libfftw3f.so
-- FFTW3F STATIC LIBRARIES: /usr/lib/x86_64-linux-gnu/libfftw3f.a
-- FFTW3F INCLUDE DIRS: /usr/include
-- Found FFTW3F: /usr/lib/x86_64-linux-gnu/libfftw3f.so  
-- Found GTest: /usr/lib/x86_64-linux-gnu/cmake/GTest/GTestConfig.cmake (found version "1.11.0")  
-- Found Threads: TRUE  
-- UHD LIBRARIES UHD_LIBRARIES-NOTFOUND
-- UHD INCLUDE DIRS UHD_INCLUDE_DIRS-NOTFOUND
-- Could NOT find UHD (missing: UHD_LIBRARIES UHD_INCLUDE_DIRS) 
-- Checking for module 'yaml-cpp'
--   Found yaml-cpp, version 0.7.0
-- YAMLCPP LIBRARIES: /usr/lib/x86_64-linux-gnu/libyaml-cpp.so
-- YAMLCPP INCLUDE DIRS: /usr/include
-- Found YAMLCPP: /usr/lib/x86_64-linux-gnu/libyaml-cpp.so  
-- FINDING ZEROMQ.
-- Checking for module 'ZeroMQ'
--   No package 'ZeroMQ' found
-- Found libZEROMQ: /usr/include, /usr/lib/x86_64-linux-gnu/libzmq.so
-- Could NOT find DPDK (missing: DPDK_INCLUDE_DIRS CHECK_LIBRARIES) (Required is at least version "22.11")
-- DPDK LIBRARIES: 
-- DPDK INCLUDE DIRS: DPDK_INCLUDE_DIRS-NOTFOUND
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- Checking for module 'sctp'
--   No package 'sctp' found
-- SCTP LIBRARIES: /usr/lib/x86_64-linux-gnu/libsctp.so
-- SCTP INCLUDE DIRS: /usr/include
-- Found SCTP: /usr/lib/x86_64-linux-gnu/libsctp.so  
-- FFT_LIBRARIES: /usr/lib/x86_64-linux-gnu/libfftw3f.so
-- Building srsRAN version 23.5.0
-- Configuring done
-- Generating done
-- Build files have been written to: /root/srsRAN_Project/build
```

s5uishida, 2023-08-07:
Build succeeded!

Thank you very much!!

s5uishida, 2023-08-07:
A little information.

At the end of the build, 4GB memory was not enough and changed it to 8GB.

xavierarteaga, 2023-08-07:
This problem was seen in [this discussion](https://github.com/srsran/srsRAN_Project/discussions/151). A patch is mentioned in the last comment and it will be included in the next release.