srsran/srsRAN_Project #162: Make error with the latest srsRAN drop.

asheeshtripathi, 2023-08-08:
## Issue Description ##
With git pull, tried the latest srsRAN project install but got errors with make -j $(nproc):

`In file included from /home/cciadmin/Github/srsRAN_Project/lib/hal/dpdk/dpdk_eal.cpp:24:
/home/cciadmin/Github/srsRAN_Project/lib/hal/dpdk/dpdk.h:28:10: fatal error: rte_eal.h: No such file or directory
   28 | #include "rte_eal.h"
      |          ^~~~~~~~~~~
compilation terminated.
make[2]: *** [lib/hal/dpdk/CMakeFiles/hal_dpdk.dir/build.make:63: lib/hal/dpdk/CMakeFiles/hal_dpdk.dir/dpdk_eal.cpp.o] Error 1
In file included from /home/cciadmin/Github/srsRAN_Project/lib/hal/dpdk/dpdk_eal_factory.cpp:24:
/home/cciadmin/Github/srsRAN_Project/lib/hal/dpdk/dpdk.h:28:10: fatal error: rte_eal.h: No such file or directory
   28 | #include "rte_eal.h"
      |          ^~~~~~~~~~~
`
 **Similarly, there are other errors in other header file path:**

`In file included from /home/cciadmin/Github/srsRAN_Project/lib/hal/dpdk/mbuf_pool.cpp:23:
/home/cciadmin/Github/srsRAN_Project/include/srsran/hal/dpdk/mbuf_pool.h:25:10: fatal error: rte_mempool.h: No such file or directory
   25 | #include "rte_mempool.h"

In file included from /home/cciadmin/Github/srsRAN_Project/lib/hal/dpdk/dpdk_eal.cpp:24:
/home/cciadmin/Github/srsRAN_Project/lib/hal/dpdk/dpdk.h:28:10: fatal error: rte_eal.h: No such file or directory
   28 | #include "rte_eal.h"
      |          ^~~~~~~~~~~
compilation terminated.
make[2]: *** [lib/hal/dpdk/CMakeFiles/hal_dpdk.dir/build.make:63: lib/hal/dpdk/CMakeFiles/hal_dpdk.dir/dpdk_eal.cpp.o] Error 1
In file included from /home/cciadmin/Github/srsRAN_Project/lib/hal/dpdk/dpdk_eal_factory.cpp:24:
/home/cciadmin/Github/srsRAN_Project/lib/hal/dpdk/dpdk.h:28:10: fatal error: rte_eal.h: No such file or directory
   28 | #include "rte_eal.h"
      |          ^~~~~~~~~~~
compilation terminated.
make[2]: *** [lib/hal/dpdk/CMakeFiles/hal_dpdk.dir/build.make:76: lib/hal/dpdk/CMakeFiles/hal_dpdk.dir/dpdk_eal_factory.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:7851: lib/hal/dpdk/CMakeFiles/hal_dpdk.dir/all] Error 2
`

## Setup Details ##
This setup already had the previous srsRAN project drop running.

## Expected Behavior ##
make is successful.

## Actual Behaviour ##
Errors in make.

## Steps to reproduce the problem ##
git pull or git clone the latest srsRAN project and following steps:
cd build
cmake ../
make -j $(nproc)
make test -j $(nproc)

## Additional Information ##
I see some issue in locating path for multiple header files.

ismagom, 2023-08-08:
Thanks for reporting. We'll fix the issue soon. In the meantime, you can compile passing `-DENABLE_DPDK=OFF` to cmake

asheeshtripathi, 2023-08-08:
Thanks. I will try with `-DENABLE_DPDK=OFF`

asheeshtripathi, 2023-08-09:
Update: I completed the installation with `-DENABLE_DPDK=OFF`.

andrepuschmann, 2023-08-30:
This is fixed in the latest drop.

closed 2023-08-30 by andrepuschmann