srsran/srsRAN_Project #165: Adding a slice section to ue_zmq.conf causes attach to Open5GS  to fail

s5uishida, 2023-08-11:
## Issue Description ##

Adding a slice section to `ue_zmq.conf` causes `InitialUEMessage, Registration request` to Open5GS to fail.

## Setup Details ##

I added a slice section to `ue_zmq.conf` as below.
```
[slicing]
enable = true
nssai-sst = 1
nssai-sd = 1
```
Also I added a slice section to `gnb_zmq.yaml`.
```
slicing:
  - sst: 1
    sd: 1
#  - sst: 1
#    sd: 2
```
Open5GS log is as follows.
```
08/11 13:41:18.787: [amf] INFO: InitialUEMessage (../src/amf/ngap-handler.c:372)
08/11 13:41:18.787: [amf] INFO: [Added] Number of gNB-UEs is now 1 (../src/amf/context.c:2510)
08/11 13:41:18.787: [amf] INFO:     RAN_UE_NGAP_ID[0] AMF_UE_NGAP_ID[5] TAC[1] CellID[0x19b0] (../src/amf/ngap-handler.c:533)
08/11 13:41:18.787: [amf] INFO: [suci-0-001-01-0000-0-0-0000000000] Unknown UE by SUCI (../src/amf/context.c:1776)
08/11 13:41:18.787: [amf] INFO: [Added] Number of AMF-UEs is now 1 (../src/amf/context.c:1563)
08/11 13:41:18.787: [gmm] INFO: Registration request (../src/amf/gmm-sm.c:1061)
08/11 13:41:18.787: [gmm] ERROR: Non cleartext IEs is included [0xc] (../src/amf/gmm-handler.c:116)
08/11 13:41:18.787: [gmm] ERROR: Unknown reg_type[0] (../src/amf/gmm-sm.c:1080)
08/11 13:41:18.787: [gmm] ERROR: gmm_handle_registration_request() failed [95] (../src/amf/gmm-sm.c:1084)
08/11 13:41:18.787: [amf] ERROR: Unknown reg_type[0] (../src/amf/nas-path.c:215)
08/11 13:41:18.787: [amf] WARNING: [(null)] Registration reject [95] (../src/amf/nas-path.c:219)
08/11 13:41:18.789: [amf] INFO: UE Context Release [Action:3] (../src/amf/ngap-handler.c:1632)
08/11 13:41:18.789: [amf] INFO:     RAN_UE_NGAP_ID[0] AMF_UE_NGAP_ID[5] (../src/amf/ngap-handler.c:1633)
08/11 13:41:18.789: [amf] INFO:     SUCI[Unknown] (../src/amf/ngap-handler.c:1636)
08/11 13:41:18.789: [amf] INFO: [Removed] Number of gNB-UEs is now 0 (../src/amf/context.c:2517)
08/11 13:41:18.789: [amf] INFO: [Removed] Number of AMF-UEs is now 0 (../src/amf/context.c:1656)
```
The pcap log at this time is as follows.
[slice_section_exist.zip](https://github.com/srsran/srsRAN_Project/files/12318722/slice_section_exist.zip)
```
Frame 113: 112 bytes on wire (896 bits), 112 bytes captured (896 bits)
Linux cooked capture v2
Internet Protocol Version 4, Src: 192.168.0.111, Dst: 192.168.0.121
Stream Control Transmission Protocol, Src Port: 38412 (38412), Dst Port: 37003 (37003)
NG Application Protocol (DownlinkNASTransport)
    NGAP-PDU: initiatingMessage (0)
        initiatingMessage
            procedureCode: id-DownlinkNASTransport (4)
            criticality: ignore (1)
            value
                DownlinkNASTransport
                    protocolIEs: 3 items
                        Item 0: id-AMF-UE-NGAP-ID
                        Item 1: id-RAN-UE-NGAP-ID
                        Item 2: id-NAS-PDU
                            ProtocolIE-Field
                                id: id-NAS-PDU (38)
                                criticality: reject (0)
                                value
                                    NAS-PDU: 7e00445f
                                        Non-Access-Stratum 5GS (NAS)PDU
                                            Plain NAS 5GS Message
                                                Extended protocol discriminator: 5G mobility management messages (126)
                                                0000 .... = Spare Half Octet: 0
                                                .... 0000 = Security header type: Plain NAS message, not security protected (0)
                                                Message type: Registration reject (0x44)
                                                5GMM cause
                                                    5GMM cause: Semantically incorrect message (95)
```
If make `slicing.enable = false` in `ue_zmq.conf`, it works fine.

## Expected Behavior ##

I want to specify slice and attach UE with ZMQ to Open5GS.

## Actual Behaviour ##

## Steps to reproduce the problem ##

## Additional Information ##

The environment of gNodeB with ZMQ (srsRAN_Project) and UE with ZMQ (srsRAN_4G) are as follows.

- Virtualbox VM
- Ubuntu 22.04 (Linux 5.15.0-78)
- gcc version 11.4.0 (Ubuntu 11.4.0-1ubuntu1~22.04)

s5uishida, 2023-08-13:
For free5GC, UE was able to connect to free5GC 5GC using the `ue_zmq.conf` with the `slicing` section enabled.

s5uishida, 2023-08-13:
This issue has been clear a little.

When added the `slicing` section to `ue_zmq.yaml`, `srsue` created an `InitialUEMessage` appended `NSSAI` to the `Plain NAS 5GS Message`, and sent it to AMF.

On the other hand, according to the specification below, `NSSAI` is not specified to be included in cleartext IE.

`3GPP TS 24.501 - 4.4.6 Protection of initial NAS signalling messages`
```
When the initial NAS message is a REGISTRATION REQUEST message, the cleartext IEs are:

- Extended protocol discriminator;
- Security header type;
- Spare half octet;
- Registration request message identity;
- 5GS registration type;
- ngKSI;
- 5GS mobile identity;
- UE security capability;
- Additional GUTI;
- UE status; and
- EPS NAS message container.
```
Open5GS checked this and returned an error.

So I think that `srsue` needs to follow the above specification.

I'm sorry if I misunderstood.

Best regards,

--Shigeru

andrepuschmann, 2023-08-30:
Hey @s5uishida - thanks for digging into this issue yourself. Your conclusion seems correct - NSSAI IE is not allowed to go over the air in cleartext. Please feel free to open a ticket in [srsRAN_4G](https://github.com/srsran/srsRAN_4G/issues) for it. Closing this one here.
Thanks