srsran/srsRAN_Project #172: Connection issues between gnb hosts and core network hosts

baolindong, 2023-08-16:
## Issue Description ##
[Describe the issue in detail]
Hello!
I have deployed the Open5GS core network and SRS gnb according to the tutorial on the official website. When the gnb and core network are deployed on the same server, they can be used normally; When the core network and gnb are deployed on different servers, there may be problems in transmitting data between the base station and the core network. Here are my configuration files for amf, upf, and gnb, as well as the log files for the runtime gnb. 
## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions
 amf.yaml 
`amf:
    sbi:
      - addr: 127.0.0.5
        port: 7777
    ngap:
      - addr: 10.128.181.23 #127.0.0.5
    metrics:
      - addr: 127.0.0.5
        port: 9090
    guami:
      - plmn_id:
          mcc: 001 #999
          mnc: 01 #70
        amf_id:
          region: 2
          set: 1
    tai:
      - plmn_id:
          mcc: 001 #999
          mnc: 01 #70
        tac: 7
    plmn_support:
      - plmn_id:
          mcc: 001 #999
          mnc: 01 #70
        s_nssai:
          - sst: 1
    security:
        integrity_order : [ NIA2, NIA1, NIA0 ]
        ciphering_order : [ NEA0, NEA1, NEA2 ]
    network_name:
        full: Open5GS
    amf_name: open5gs-amf0
`
upf.yaml
`upf:
    pfcp:
      - addr: 127.0.0.7
    gtpu:
      - addr: 10.128.181.23 #127.0.0.7
    subnet:
      - addr: 10.45.0.1/16
      - addr: 2001:db8:cafe::1/48
    metrics:
      - addr: 127.0.0.7
        port: 9090
`
gnb_rf_b200_tdd_n78_20mhz.yml
`amf:
  addr: 10.128.181.23 # 10.128.191.80  #10.128.181.23 #127.0.1.100                                     # The address or hostname of the AMF.
  bind_addr: 127.0.0.1                                            # A local IP that the gNB binds to for traffic from the AMF.

ru_sdr:
  device_driver: uhd                                              # The RF driver name.
  device_args: type=b200,num_recv_frames=64,num_send_frames=64    # Optionally pass arguments to the selected RF driver.
  srate: 23.04                                                    # RF sample rate might need to be adjusted according to selected bandwidth.
  otw_format: sc12
  tx_gain: 76 #50                                                     # Transmit gain of the RF might need to adjusted to the given situation.
  rx_gain: 76 #60                                                     # Receive gain of the RF might need to adjusted to the given situation.

cell_cfg:
  dl_arfcn: 632628                                                # ARFCN of the downlink carrier (center frequency).
  band: 78                                                        # The NR band.
  channel_bandwidth_MHz: 20                                       # Bandwith in MHz. Number of PRBs will be automatically derived.
  common_scs: 30                                                  # Subcarrier spacing in kHz used for data.
  plmn: "00101"                                                   # PLMN broadcasted by the gNB.
  tac: 7                                                          # Tracking area code (needs to match the core configuration).
  pci: 1                                                          # Physical cell ID.

log:
  filename: /tmp/gnb.log                                          # Path of the log file.
  all_level: info #debug #warning                                                 # Logging level applied to all layers.

pcap:
  mac_enable: false                                               # Set to true to enable MAC-layer PCAPs.
  mac_filename: /tmp/gnb_mac.pcap                                 # Path where the MAC PCAP is stored.
  ngap_enable: true #false                                              # Set to true to enable NGAP PCAPs.
  ngap_filename: /tmp/gnb_ngap.pcap        
 `

gnb.log
`2023-08-16T02:30:24.041108 [SCHED   ] [I] [   429.5] Slot decisions cell=0 t=25us: DL: ue=0 c-rnti=0x4601 rb=[0..1) h_id=0 ss_id=2 k1=4 rv=0 tbs=76
2023-08-16T02:30:24.041131 [RLC     ] [I] ue=0 DRB1 DL: TX status PDU. pdu_len=3 grant_len=3
2023-08-16T02:30:24.041134 [MAC     ] [I] [   429.5] DL PDU: ue=0 rnti=0x4601 size=76: SDU: lcid=4 size=5
2023-08-16T02:30:24.041232 [DL-PHY9 ] [I] [   429.5] PDCCH: rnti=0x4601 ss_id=2 format=1_1 cce=0 al=4 t=96.3us
2023-08-16T02:30:24.041342 [DL-PHY9 ] [I] [   429.5] PDSCH: rnti=0x4601 h_id=0 k1=4 prb=[0, 1) symb=[2, 14) mod=64QAM rv=0 tbs=76 t=103.6us
2023-08-16T02:30:24.046248 [UL-PHY4 ] [I] [   429.9] PUCCH: rnti=0x4601 format=1 prb1=0 prb2=na symb=[0, 14) cs=0 occ=0 ack=0 t=100.9us
2023-08-16T02:30:24.048673 [SCHED   ] [I] [   430.0] Slot decisions cell=0 t=14us: DL: ue=0 c-rnti=0x4601 rb=[0..1) h_id=0 ss_id=2 k1=7 rv=2 tbs=76
2023-08-16T02:30:24.048961 [DL-PHY8 ] [I] [   430.0] PDCCH: rnti=0x4601 ss_id=2 format=1_1 cce=8 al=4 t=92.6us
2023-08-16T02:30:24.049066 [DL-PHY8 ] [I] [   430.0] PDSCH: rnti=0x4601 h_id=0 k1=7 prb=[0, 1) symb=[2, 14) mod=64QAM rv=2 tbs=76 t=99.0us
2023-08-16T02:30:24.055206 [UL-PHY1 ] [I] [   430.7] PUCCH: rnti=0x4601 format=1 prb1=0 prb2=na symb=[0, 14) cs=0 occ=0 ack=1 t=78.3us
2023-08-16T02:30:24.070641 [UL-PHY2 ] [I] [  431.18] PUCCH: rnti=0x4601 format=2 prb=[3, 4) prb2=na symb=[12, 14) csi1=1101 t=66.6us
2023-08-16T02:30:24.071196 [UL-PHY3 ] [I] [  431.19] PUCCH: rnti=0x4601 format=1 prb1=49 prb2=na symb=[0, 14) cs=0 occ=0 sr=yes t=57.2us
2023-08-16T02:30:24.071299 [DL-PHY5 ] [I] [   432.5] PDCCH: rnti=0x4601 ss_id=2 format=0_1 cce=0 al=4 t=30.8us
2023-08-16T02:30:24.073072 [SCHED   ] [I] [   432.9] Slot decisions cell=0 t=3us: UL: ue=0 rnti=0x4601 h_id=0 ss_id=2 rb=[4..10) k2=4 rv=0 tbs=544
2023-08-16T02:30:24.076430 [UL-PHY4 ] [I] [   432.9] PUSCH: rnti=0x4601 h_id=0 prb=[4, 10) symb=[0, 14) mod=64QAM rv=0 tbs=544 crc=OK iter=2.0 snr=23.6dB t=312.2us
2023-08-16T02:30:24.076444 [MAC     ] [I] [  432.15] UL PDU rnti=0x4601 ue=0 subPDUs: [lcid=4: len=88, SBSR: lcg=0 bs=0, PAD: len=451]
2023-08-16T02:30:24.076451 [RLC     ] [I] ue=0 DRB1 UL: RX PDU. pdu_len=88 dc=data p=1 si=full sn=26 so=0
2023-08-16T02:30:24.076456 [RLC     ] [I] ue=0 DRB1 UL: RX SDU. sn=26 sdu_len=86
2023-08-16T02:30:24.076464 [PDCP    ] [I] ue=0 DRB1 UL: RX PDU. type=data pdu_len=86 sn=26 count=26
2023-08-16T02:30:24.076468 [PDCP    ] [I] ue=0 DRB1 UL: RX SDU. count=26
2023-08-16T02:30:24.076474 [GTPU    ] [I] ue=0 UL teid=0x7eb4: TX PDU. pdu_len=92 teid=0x7eb4
2023-08-16T02:30:24.076490 [UDP-GW  ] **[E] Couldn't send 92 B of data on UDP socket: local_ip=127.0.0.1 dest_ip=10.128.181.23 error="Invalid argument"**
`
## Expected Behavior ##
[What you expect to happen]
When the gnb and core network are deployed on different servers, the cost UE can connect to the internet after connecting.
## Actual Behaviour ##
[What happens instead e.g. error message]
When the gnb and core network are deployed on a server, the cost UE can be connected to the internet; When deployed on different servers, cost UE cannot connect to the internet due to data transmission issues.
## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

ngovanmao, 2023-08-17:
I faced the same issue a few days ago. Do you have any updates on this? 

I observed the SCTP connection after first Established, the heartbeat between Open5GS core and SRS-5G gNB could not maintain after that. Eventually, the SCTP connection between gNB and AMF is closed.

pgawlowicz, 2023-08-17:
Hi, could you try with this patch: [0001-sctp-create-socket-only-once.txt](https://github.com/srsran/srsRAN_Project/files/12367386/0001-sctp-create-socket-only-once.txt)

```
cd ./srsRAN_Project
git apply 0001-sctp-create-socket-only-once.txt
```

s5uishida, 2023-08-17:
Hi @pgawlowicz 

Your patch seems to solve [this](https://github.com/srsran/srsRAN_Project/discussions/107#discussioncomment-6706349) as well.

Thank you very much!

baolindong, 2023-08-17:
Hi @pgawlowicz 
Thank you very much for your answer！Unfortunately, the issue was not resolved after I applied the patch. I see that this patch is modifying the creation of SCTP socket, but the error in gnb.log is related to UDP socket.

s5uishida, 2023-08-17:
Hi @baolindong 

Isn't it necessary to change the `bind_addr` as follows to the binding address of the OS that `gnb` is running instead of the loopback address?

`gnb_rf_b200_tdd_n78_20mhz.yml`
```
amf:
  addr: 10.128.181.23 # 10.128.191.80  #10.128.181.23 #127.0.1.100                                               # The address or hostname of the AMF.
  bind_addr: 127.0.0.1 <--
...
```

baolindong, 2023-08-17:
Hi @s5uishida 
Thank you very much for your answer！After change the `bind_addr` to the IP address of gnb host, the problem was resolved.

pgawlowicz, 2023-08-17:
nice, could you try to remove the patch and see whether the problem was only related to the `bind_addr` ?

ngovanmao, 2023-08-18:
Thanks @pgawlowicz . Patch works well.

baolindong, 2023-08-18:
Hi @pgawlowicz 
After change the `bind_addr`, I have try to use and remove patches, and cost UE can be used normally.

s5uishida, 2023-08-18:
Hi @pgawlowicz 

I'm looking forward to your patch being merged into the srsRAN_Project.

Thanks again!

Steve-George, 2023-08-29:
This patch is perfect. Thanks.

andrepuschmann, 2023-08-30:
This issue seems to have been a configuration problem afterall. Closing this therefore.

closed 2023-08-30 by andrepuschmann