srsran/srsRAN_Project #181: No Radio device available. Unable to create radio session

Sabbir89, 2023-08-23:
Hi,
First time going to use srsRAN following the tutorials on : https://docs.srsran.com/projects/project/en/latest/tutorials/source/cotsUE/source/index.html and the installation guide: https://docs.srsran.com/projects/project/en/latest/user_manuals/source/installation.html
However, not getting success as the radio session is not getting established. 

I have connected the usrp B210 radio with the gNB pc, but still getting the issue. 
I have attached the snap of the output after using the command: sudo ./gnb -c gnb_b210_20MHz_oneplus_8t.yml
[image]

Any hint will be highly appreciated. Thanks

Best Regards,
Sabbir

robertfalkenberg, 2023-08-24:
To provide some help, please repeat the following [steps](https://docs.srsran.com/projects/project/en/latest/user_manuals/source/installation.html#clone-and-build) and share the terminal outputs (**as attached text file**):

```
cd srsRAN_Project
mkdir build
cd build
cmake ../
make -j $(nproc)
make test -j $(nproc)
```

Sabbir89, 2023-08-24:
Dear Robert,

Thank you so much for the great tips. It started immediately right after I repeated the steps.
However, I am getting new issue with Connection refused from the AMF log. the gNB connects to the core then immediately drops the connection. Seems no SCTP stable connection is getting established between the core and the gNB.

Any further help?

Best regards,
Sabbir

s5uishida, 2023-08-25:
Hi @Sabbir89 

Is [this](https://github.com/srsran/srsRAN_Project/issues/172#issuecomment-1681908406) helpful?

asheeshtripathi, 2023-08-25:
Hi @Sabbir89, for the "no radio found" issue, use `uhd_usrp_probe` and `uhd_find_devices` to check if USRP is reachable and configured properly. For the SCTP connection, you need to check the AMF addr in gNB config file and in Open5GS config as @s5uishida mentioned.

Sabbir89, 2023-08-25:
Hi Shigeru and Asheesh,

Thanks for your prompt responses.

As mentioned before, the first issue solved and the gNB is getting started every time I am checking. However, the connection between the core and the gNB is not establishing (Connection refuse). 
The config parameters are similar in the Core and in the gNB files.
What I have done so far:
1. Changed the bind_addr to the IP of the gNB host machine
2. Tried this patch: try with this patch: [0001-sctp-create-socket-only-once.txt](https://github.com/srsran/srsRAN_Project/files/12367386/0001-sctp-create-socket-only-once.txt) 

Could you please tell me what should I try next?

Best Regards,
Sabbir

asheeshtripathi, 2023-08-25:
Please see the following issues; they are similar to your issue:

https://github.com/srsran/srsRAN_Project/discussions/159
https://github.com/srsran/srsRAN_Project/discussions/107
https://github.com/srsran/srsRAN_Project/discussions/95
https://github.com/srsran/srsRAN_Project/issues/104

Debugging will be based on your deployment: Where you have installed srsRAN gNB and Open5GS on the same machine or a different machine.

Sabbir89, 2023-08-25:
Hi Asheesh,
Thanks for the links but no luck yet.
In my case, core and gNB are in two different physical machines (Lenovo ThinkStation and ThinkCentre with Ubuntu OS).
I have already tested my core with OAI-gNB successfully. But, in this case, having connection refuse every time in the AMF log. It connects and drops immediately. I am a bit concerned about SCTP signalling. Maybe something is wrong in my gNB machine with SCTP. 
Will be continuing debugging but any further help from anyone will be appreciated.

Best Regards,
Sabbir

andrepuschmann, 2023-08-30:
Can you please share full logs and NGAP PCAPs for your setup and also explain it a bit more. The "Unable to create radio session" is orthogonal to the AMF connection issue.

Sabbir89, 2023-09-06:
Hi,

This to confirm you that the problem has been solved. It was an issue connecting two gNBs to a single core. One was always been refused. When I try standalone in a new setup, it works. Now troubling with UE connection. However, I will make it happen I think. I also changed the band, srate, etc. according to my setup.

Thanks everyone for your help.

Best Regards,
Sabbir

jackmygreat, 2023-09-16:
@Sabbir89 I alos meet this problem. How do you solve it ?

closed 2023-10-26 by ismagom

connor01M, 2023-12-06:
> Hi,
>
> This to confirm you that the problem has been solved. It was an issue connecting two gNBs to a single core. One was always been refused. When I try standalone in a new setup, it works. Now [...]

Hey Sabbir, what did you do to fix this?