srsran/srsRAN_Project #186: Unable to set appropriate sample rate for gNB

ncmatson, 2023-08-28:
## Issue Description ##
This is my first time attempting to set up the system.  I'm having trouble setting the sample rate for an N310.  When I use the default ``configs/gnb_rf_n310_fdd_n3_20mhz.yml`` file I get many over/under runs.  The over/under runs occur when I use the same sample rate (30.72 MS/s) with ``uhd/examples/benchmark_rate`` so I thought I would just reduce `srate` in the config file to get things working.

The first issue is that it complains 

```
The sampling rate must be multiple of 1.92 MHz.
srsGNB ERROR: Invalid configuration detected.
```

I'm not sure who (srs or uhd) is enforcing this requirement.

The second issue is that if I do in fact set a multiple of 1.92, i.e. 26.88, it gives the following error 

``Error: setting Tx sampling rate. Tx Rate 26.88 MHz is invalid. The nearest valid value is 24.576.``

and then aborts.

Normally when giving a non integer decimation of the master clock rate, UHD will just set it to the nearest value and warn the user i.e.

``
[WARNING] [MULTI_USRP] The hardware does not support the requested TX sample rate:                                          
Target sample rate: 26.880000 MSps                                                                                          
Actual sample rate: 24.576000 MSps           
``

which is what I get when I run the command ``uhd/examples/benchmark_rate --tx_rate 26.88e6 --rx_rate 26.88e6 --args=master_clock_rate=122.88e6,recv_frame_size=8000,send_frame_size=8000``

## Setup Details ##
Using an N310 and UHD version 3.15, the srsRan_Project is at commit 1afd7240f.

andrepuschmann, 2023-08-30:
Hey @ncmatson - you can't just arbitrarily reduce the sample rate. The sample rate of the radio and the configured cell bandwidth need to match. In any case if you see issue with just running `uhd/examples/benchmark_rate` at 30.72msps on your PC that is a strong indicator that something in your setup is wrong. You need that to work with issues. It doesn't make sense to run the gNB otherwise. For the N310 we also suggest to use a more recent UHD version. 3.15 works well for the B210 for example but has issues with the N-series.

ncmatson, 2023-08-30:
Ah.  Learned something new.  Didn't realize the sampling rate and bandwidth were paired in that way.  Definitely working on fixing the networking issue.  Thanks for your help and the tip about uhd version.

closed 2023-08-30 by ncmatson