srsran/srsRAN_Project #198: O-RAN NearRT-RIC  Open5GS docker Error

aldalahmeh, 2023-09-05:
## Issue Description ##
[Describe the issue in detail]
When isntalling the Open5GS docker image I get the following error

Step 18/20 : COPY subscriber_db.cs[v] .
COPY failed: no source files were specified
ERROR: Service '5gc' failed to build : Build failed

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions

## Expected Behavior ##
[What you expect to happen]

## Actual Behaviour ##
[What happens instead e.g. error message]

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

pgawlowicz, 2023-09-06:
Do you build the docker image as follows?

```
cd ./srsRAN_Project/docker
docker-compose up --build 5gc
```

joshuamoorexyz, 2023-09-08:
I also have this issue it seems when running the command 'docker-compose up --build 5gc'    

Output: joshuamoorexyz@joshuamoorexyz-B550-UD-AC:~/Github/srsRAN_Project/docker$ docker-compose up --bu
ild 5gc                                                                                        
WARNING: Some services (gnb) use the 'configs' key, which will be ignored. Compose does not sup
port 'configs' configuration - use `docker stack deploy` to deploy to a swarm.                 
Traceback (most recent call last):                                                             
  File "urllib3/connectionpool.py", line 677, in urlopen                                       
  File "urllib3/connectionpool.py", line 392, in _make_request                                 
  File "http/client.py", line 1277, in request                                                 
  File "http/client.py", line 1323, in _send_request                                           
  File "http/client.py", line 1272, in endheaders                                              
  File "http/client.py", line 1032, in _send_output                                            
  File "http/client.py", line 972, in send
  File "docker/transport/unixconn.py", line 43, in connect
FileNotFoundError: [Errno 2] No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "requests/adapters.py", line 449, in send
  File "urllib3/connectionpool.py", line 727, in urlopen
  File "urllib3/util/retry.py", line 410, in increment
  File "urllib3/packages/six.py", line 734, in reraise
  File "urllib3/connectionpool.py", line 677, in urlopen
  File "urllib3/connectionpool.py", line 392, in _make_request
  File "http/client.py", line 1277, in request
  File "http/client.py", line 1323, in _send_request
  File "http/client.py", line 1272, in endheaders
  File "http/client.py", line 1032, in _send_output
  File "http/client.py", line 972, in send
  File "docker/transport/unixconn.py", line 43, in connect
urllib3.exceptions.ProtocolError: ('Connection aborted.', FileNotFoundError(2, 'No such file or directory'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "docker/api/client.py", line 214, in _retrieve_server_version
  File "docker/api/daemon.py", line 181, in version
  File "docker/utils/decorators.py", line 46, in inner
  File "docker/api/client.py", line 237, in _get
  File "requests/sessions.py", line 543, in get 
  File "requests/sessions.py", line 530, in request
  File "requests/sessions.py", line 643, in send
  File "requests/adapters.py", line 498, in send
requests.exceptions.ConnectionError: ('Connection aborted.', FileNotFoundError(2, 'No such file
 or directory'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "docker-compose", line 3, in <module>
  File "compose/cli/main.py", line 81, in main
  File "compose/cli/main.py", line 200, in perform_command
  File "compose/cli/command.py", line 70, in project_from_options
  File "compose/cli/command.py", line 153, in get_project
  File "compose/cli/docker_client.py", line 43, in get_client
  File "compose/cli/docker_client.py", line 170, in docker_client
  File "docker/api/client.py", line 197, in __init__
  File "docker/api/client.py", line 222, in _retrieve_server_version
docker.errors.DockerException: Error while fetching server API version: ('Connection aborted.',
 FileNotFoundError(2, 'No such file or directory'))
[54435] Failed to execute script docker-compose 

I have tried cloning down the srsran_project again with no luck.

aldalahmeh, 2023-09-10:
Yes, exactly

pgawlowicz, 2023-09-11:
OK, seems that there was a bug, it should be fixed in the next release.

FabianEckermann, 2023-09-12:
Could you provide some more info on your system, like the OS you are using and the version of docker?

aldalahmeh, 2023-09-12:
The OS is ubuntu 22.0402 LTS
docker version 20.10.24

abelcttc, 2023-09-12:
Morning,
Same issue but with docker version 24.0.5
Thanks!

FabianEckermann, 2023-09-12:
I'm not able to reproduce this issue locally. Did you change anything in the docker folder?

abelcttc, 2023-09-12:
By commenting the command at the DOCKERFILE ""COPY subscriver_db.cs[v] ." (line 79) it works

FabianEckermann, 2023-09-12:
Does it work if you change line 77 at the Dockerfile from `COPY setup_tun.py .` to `COPY setup_tun.py subscriber_db.cs[v] ./`?

abelcttc, 2023-09-12:
Yes, solved!
Now I have a problem at the time of executing the ue: 
RRC Connected
Failed to setup/configure GW interface
RRC NR reconfiguration successful.

Attach are the logs from ue and gnb
[gnb.log](https://github.com/srsran/srsRAN_Project/files/12587450/gnb.log)
[ue.log](https://github.com/srsran/srsRAN_Project/files/12587451/ue.log)

FabianEckermann, 2023-09-12:
Please create a separate issue for this. I'll add a fix for the docker error that will probably be available in the next release

closed 2023-11-14 by yagoda