srsran/srsRAN_Project #199: srsUE Retransmission Failed, Authentication Reject.

dhiaboujebha, 2023-09-06:
## Issue Description ##
I am building a setup using the Tutorial [srsRAN gNB with srsUE](https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html) and i managed to get the gnb of srsRAN_Project connected to the core of open5gs, but the srsUE of srsRAN4G tries to connect to the core but it gets rejected after 30 seconds: Retransmission Failed. Stop Retransmission. Authentication Reject. 

## Setup Details ##
Setup:
3 PCs Ubuntu 20.04
USRP X310 (gNB)
USRP B200mini (UE)
UHD 4.4.0.0

[gnb_files.zip](https://github.com/srsran/srsRAN_Project/files/12538876/gnb_files.zip)

[UE_files.zip](https://github.com/srsran/srsRAN_Project/files/12538887/UE_files.zip)

[open5gs_core.zip](https://github.com/srsran/srsRAN_Project/files/12538894/open5gs_core.zip)

## Expected Behavior ##
[image]

## Actual Behaviour ##
[image]

[image]

[image]

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]

## Additional Information ##
I have tried to connect UERANSIM gnb and UE to the core and it worked without problems. I used the same USIM, that i already registred in the WEBUI. I changed several times the TX_Gain and RX_Gain but the best result that i've got is this. 

I hope that you help me solving this.

yagoda, 2023-09-18:
are you sure you have attached the correct gnb.log, i dont see any sign of a UE attaching in the gnb.log or in the PCAP.

dhiaboujebha, 2023-09-20:
@yagoda Sorry, i think, i have uploaded the wrong logs. Here are new gnb and ue logs. Thanks 
[UE1.zip](https://github.com/srsran/srsRAN_Project/files/12671886/UE1.zip)
[gNB1.zip](https://github.com/srsran/srsRAN_Project/files/12671888/gNB1.zip)

andrepuschmann, 2023-09-21:
Please share also PCAP and 5GC logs/PCAP

dhiaboujebha, 2023-09-22:
@andrepuschmann there are no PCAP for the UE. New logs are attached. Thanks
[ue.zip](https://github.com/srsran/srsRAN_Pr
[gNB.zip](https://github.com/srsran/srsRAN_Project/files/12699530/gNB.zip)
oject/files/12699526/ue.zip)
[5gc.zip](https://github.com/srsran/srsRAN_Project/files/12699533/5gc.zip)

ismagom, 2023-11-08:
Hi @dhiaboujebha , any progress on this issue?

dhiaboujebha, 2024-01-18:
We didn't manage to fix this issue on this setup. So we switched to another setup, which worked #269 and now we are facing a new issue #412.

closed 2024-01-18 by dhiaboujebha