srsran/srsRAN_Project #204: COTS UE cant reach the core network

ushasigh, 2023-09-07:
## COTS UE cant reach the core network ##
The amf logs tag the UE as Unknown ID

## Setup Details ##
I am using a oneplus 8t phone to test the setup according to app note "srsRAN gNB with COTS UEs". I am using the sim cards provided by the open cells project and used uicc code to program my sim card:
Here is the command I used: 
sudo ./program_uicc --adm 12345678 --imsi 001010123456789 --isdn 00000001 --acc 0001 --key 41B7157E3337F0ADD8DA89210D89E17F --opc 1CD638FC96E02EBD35AA0D41EB6F812F -spn "test_sim" --authenticate --iccid 89882110000000689615 --adm 12345678

I have checked the apn names and verified the open5gs webui database to ensure all fields match between the sim card and the phone. The phone can connect to the gnb but the core network is showing as "Unknown ID". The amf log is attached in the section on actual behaviour.

## Expected Behavior ##
04/03 10:36:52.012: [sctp] INFO: AMF initialize...done (../src/amf/app.c:33)
04/03 10:36:52.049: [sbi] INFO: [aea4db10-d1fa-41ed-916b-e56218b693e5] (NRF-notify) NF registered (../lib/sbi/nnrf-handler.c:632)
04/03 10:36:52.049: [sbi] INFO: [aea4db10-d1fa-41ed-916b-e56218b693e5] (NRF-notify) NF Profile updated (../lib/sbi/nnrf-handler.c:642)
04/03 10:36:52.049: [sbi] WARNING: [aea4db10-d1fa-41ed-916b-e56218b693e5] (NRF-notify) NF has already been added (../lib/sbi/nnrf-handler.c:636)
04/03 10:36:52.049: [sbi] INFO: [aea4db10-d1fa-41ed-916b-e56218b693e5] (NRF-notify) NF Profile updated (../lib/sbi/nnrf-handler.c:642)
04/03 10:36:52.049: [sbi] WARNING: NF EndPoint updated [127.0.0.12:80] (../lib/sbi/context.c:1618)
04/03 10:36:52.049: [sbi] WARNING: NF EndPoint updated [127.0.0.12:7777] (../lib/sbi/context.c:1527)
04/03 10:36:52.238: [app] INFO: SIGHUP received (../src/main.c:58)
04/03 10:36:52.350: [sbi] INFO: [aea6bae8-d1fa-41ed-904f-f78f7a58f5f3] (NRF-notify) NF registered (../lib/sbi/nnrf-handler.c:632)
04/03 10:36:52.350: [sbi] INFO: [aea6bae8-d1fa-41ed-904f-f78f7a58f5f3] (NRF-notify) NF Profile updated (../lib/sbi/nnrf-handler.c:642)

## Actual Behaviour ##
09/06 17:18:36.176: [amf] INFO: [Added] Number of gNB-UEs is now 1 (../src/amf/context.c:2502)
09/06 17:18:36.176: [amf] INFO:     RAN_UE_NGAP_ID[0] AMF_UE_NGAP_ID[3] TAC[7] CellID[0x19b0] (../src/amf/ngap-handler.c:533)
09/06 17:18:36.176: [amf] INFO: Unknown UE by 5G-S_TMSI[AMF_ID:0x20040,M_TMSI:0xc0000205] (../src/amf/context.c:1794)
09/06 17:18:36.176: [amf] INFO: [Added] Number of AMF-UEs is now 1 (../src/amf/context.c:1563)
09/06 17:18:36.176: [gmm] INFO: Registration request (../src/amf/gmm-sm.c:985)
09/06 17:18:36.176: [gmm] INFO: [Unknown ID]    5G-S_GUTI[AMF_ID:0x20040,M_TMSI:0xc0000205] (../src/amf/gmm-handler.c:165)
09/06 17:18:36.215: [gmm] INFO: Identity response (../src/amf/gmm-sm.c:1154)
09/06 17:18:36.215: [gmm] INFO: [suci-0-001-01-0-0-0-0123456789]    SUCI (../src/amf/gmm-handler.c:916)
09/06 17:18:36.741: [gmm] INFO: [imsi-001010123456789] Registration complete (../src/amf/gmm-sm.c:1917)
09/06 17:18:36.741: [amf] INFO: [imsi-001010123456789] Configuration update command (../src/amf/nas-path.c:612)
09/06 17:18:36.741: [gmm] INFO:     UTC [2023-09-07T00:18:36] Timezone[0]/DST[0] (../src/amf/gmm-build.c:546)
09/06 17:18:36.741: [gmm] INFO:     LOCAL [2023-09-06T17:18:36] Timezone[-25200]/DST[1] (../src/amf/gmm-build.c:551)

robertfalkenberg, 2023-09-07:
Could you please clarify the actual issue with the gNB?

ushasigh, 2023-09-07:
Hi, there was no issue with the gnb, since I was using the uicc program for my SIM card, I had to update the iccid and adm keys correctly, that helped to get the connection through, thanks a lot!

closed 2023-09-07 by ushasigh