srsran/srsRAN_Project #208: Connect WAN on the example of O-RAN NearRT-RIC and xApp

lgs96, 2023-09-13:
## Issue Description ##
[Describe the issue in detail]
I want to connect ZMQ device to the WAN through Open5GC docker, but it fails.

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions
Details are followed in the example of https://docs.srsran.com/projects/project/en/latest/tutorials/source/flexric/source/index.html

## Expected Behavior ##
[What you expect to happen]
ZMQ device can ping to google.com

## Actual Behaviour ##
[What happens instead e.g. error message]
ZMQ device cannot ping to WAN

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]
Follow the example in https://docs.srsran.com/projects/project/en/latest/tutorials/source/flexric/source/index.html

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]
If I try to connect ZMQ device with non-docker Open5GC, still it fails to connect to AMF. I do not know the reason of it.

lgs96, 2023-09-13:
As an addtional question, is there example to connect srsUE to the internet with srsRAN project? If it is, please let me know. 

Best regards.

pgawlowicz, 2023-11-27:
Please check the updated tutorial. It contains a new part describing how to configure routing for the UE and the host running the core network.

closed 2023-11-27 by pgawlowicz