srsran/srsRAN_Project #209: DCI formats 0_1 and 1_1 do not work with srsUE

dcdelia, 2023-09-14:
Hi! I'm experimenting with srsRAN for NTN-related research. I'd like to make the gNB-UE communication use DCI formats 0_1 and 1_1 so I can be able to set and use DL-DataToUL-ACK values for k1. The need for this is that, unless I tweak the handling logic and encoding of the PDSCH-to-HARQ Feedback Timing Indicator field, with DCI 0_0 and 1_0 my options for k1 are limited to {4, ..., 8}.

Am I doing something wrong or the implementation (for either the gNB or the UE) is currently partial?

## Issue Description ##
The SRSran 5G gNB and the 4G UE can no longer establish a connection if DCI 0_1 and 1_1 formats are chosen for the PDCCH dedicated setup for the UE (`dci_format_0_1_and_1_1` set to `true` in the gNB config file).

## Setup Details ##
Ubuntu 20.04 with srsRAN gNB commit `1afd7240`, srsUE release 23.04, ZMQ-based communication.

## Expected Behavior ##
The SRSran 5G gNB and the 4G UE should be able to communicate just as they do when using only DCI 0_1 and 1_0.

## Actual Behaviour ##
gNB and UE can no longer establish a connection.

I'm attaching the logs obtained at `info` level are attached (please let me know if `debug` would be better.) On the gNB side, I get plenty of `PDSCH allocation skipped. Cause: No available HARQs`.

[gnb.log](https://github.com/srsran/srsRAN_Project/files/12607667/gnb.log)
[ue.log](https://github.com/srsran/srsRAN_Project/files/12607668/ue.log)

Output on stderr for the UE upon killing the process:
```
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=334
Random Access Complete.     c-rnti=0x4601, ta=0
RRC Connected
Stopping .
```

If I modify the gNB configuration below to use `false` for `dci_format_0_1_and_1_1`, after `RRC Connected` the UE correctly shows `PDU Session Establishment Successful. IP 10.45.0.64` and `RRC NR Reconfiguration Successful`.

I noticed that several recent commits involved DCI *_1-related functionality, such as adding a UE fallback mechanism (`e512dd5` and related commits). I do obtain the same issue if I tweak the gNB release 23.04 from May to use DCI 0_1 and 1_1 by setting in the `make_default_ue_search_space_config()` helper of `lib/scheduler/config/serving_cell_config_factory.cpp`:

`cfg.ue_specific = search_space_configuration::ue_specific_dci_format::f0_1_and_1_1;`

in place of `ue_specific_dci_format::f0_0_and_f1_0` used there before such commits (and the related refactoring). This made me think the functionality was not fully supported at the time, but I may be wrong due to my limited knowledge of srsRAN.

## Steps to reproduce the problem ##

gNB configuration:
```
amf:
  addr: 127.0.0.5                  # The address or hostname of the AMF.
  bind_addr: 127.0.0.10            # A local IP that the gNB binds to for traffic from the AMF.

ru_sdr:
  device_driver: zmq                # The RF driver name.
  device_args: tx_port=tcp://127.0.0.1:2000,rx_port=tcp://127.0.0.1:2001,base_srate=11.52e6            # Optionally pass arguments to the selected RF driver.
  srate: 11.52
  tx_gain: 75                       # Transmit gain of the RF might need to adjusted to the given situation.
  rx_gain: 75                       # Receive gain of the RF might need to adjusted to the given situation.

cell_cfg:
  dl_arfcn: 368500                  # ARFCN of the downlink carrier (center frequency).
  band: 3                           # The NR band.
  channel_bandwidth_MHz: 10         # Bandwith in MHz. Number of PRBs will be automatically derived.
  common_scs: 15                    # Subcarrier spacing in kHz used for data.
  plmn: "00101"                     # PLMN broadcasted by the gNB.
  tac: 7                            # Tracking area code (needs to match the core configuration).
  pdcch:
    dedicated:
      ss2_type: ue_dedicated
      dci_format_0_1_and_1_1: true

log:
  filename: /tmp/gnb.log            # Path of the log file.
  all_level: info                   # Logging level applied to all layers.
  hex_max_size: 0
```

UE configuration:
```
[rf]
freq_offset = 0
tx_gain = 50
rx_gain = 40
srate = 11.52e6
nof_antennas = 1

device_name = zmq
device_args = tx_port=tcp://127.0.0.1:2001,rx_port=tcp://127.0.0.1:2000,base_srate=11.52e6

[rat.eutra]
dl_earfcn = 2850
nof_carriers = 0

[rat.nr]
bands = 3
nof_carriers = 1

[log]
all_level = info
phy_lib_level = none
all_hex_limit = 32
filename = /tmp/ue.log
file_max_size = -1

[usim]
mode = soft
algo = milenage
opc  = 63BFA50EE6523365FF14C1F45F88737D
k    = 00112233445566778899aabbccddeeff
imsi = 001010123456780
imei = 353490069873319

[rrc]
release = 15
ue_category = 4

[nas]
apn = srsapn
apn_protocol = ipv4

[gw]
netns = ue1
ip_devname = tun_srsue
ip_netmask = 255.255.255.0

[gui]
enable = false
```

herlesupreeth, 2023-09-14:
Hey @dcdelia , srsUE does not support DCI format 1_1/0_1, thats why it fails after RRC setup

dcdelia, 2023-09-14:
Hi @herlesupreeth thanks for the quick reply! Ah, I see now. I was swayed by the presence of `dci_nr_format_1_1`-related methods in `lib/src/phy/phch/dci_nr.c`.

closed 2023-09-21 by andrepuschmann