srsran/srsRAN_Project #214: gnb amf --addr command line parameter inneffective

dziugas-wg2, 2023-09-19:
## Issue Description ##
Passing `amf --addr` command line parameter to the `gnb` binary has no effect. 

[sctp_network_gateway_impl::create_and_bind()](https://github.com/srsran/srsRAN_Project/blob/1afd7240f2b5e2061ab4158e8fcdacb15961813a/lib/gateways/sctp_network_gateway_impl.cpp#L233C6-L233C53) function creates an SCTP socket and binds an IP address provided in the `amf --addr` command line parameter to it.  However, after calling this function, `sctp_network_gateway_impl::create_and_connect()` function overrides the  `sock_fd` socket with a newly created socket that used in `connect()`:

https://github.com/srsran/srsRAN_Project/blob/1afd7240f2b5e2061ab4158e8fcdacb15961813a/lib/gateways/sctp_network_gateway_impl.cpp#L271

## Setup Details ##

- srsRAN_Project: 1afd7240f2b5e2061ab4158e8fcdacb15961813a (main)
- (https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html) ("The following config files were modified to use ZMQ-based RF driver") together with [qos.yaml](https://github.com/srsran/srsRAN_Project/blob/main/configs/qos.yml)
- Ubuntu 22.04.1 on x86_64

## Expected Behavior ##
Outbound SCTP socket should be bound to the IP address provided in the `amf --addr` command line parameter.

## Actual Behaviour ##
While the call to `sctp_network_gateway_impl::create_and_bind()` will ensure that an SCTP socket can be bound to the provided IP address, the `bind()` is not be called on the final SCTP socket.

## Steps to reproduce the problem ##
- Create an additional network interface or set a secondary IP address on an existing one
- Start gnb with `amf --addr` command line parameter
- Observe Local address in the output of `netstat -an | grep sctp` of the respective SCTP socket

wmwwimor, 2023-10-21:
[image]
[image]
I have the same problem

FabianEckermann, 2023-10-26:
Hey,
the `amf --addr` is the IP address of the AMF (the address to connect to), the `amf --bind_addr` is the local IP address to bind for AMF connection.
You need to set the `-bind_addr` to the secondary IP address you created to achieve your expected behavior.

Best regards
Fabian

closed 2023-11-06 by FabianEckermann